MarlinFirmware/Marlin #13926: [BUG][bugfix-2.0.x] Z-MIN Endstop not work IF ...

kpter, 2019-05-06:
### Description
IF all endstops ON and 
Z-probe not use Z-MIN endstop  and
Z_HOME_DIR 1 (homing to Z-MAX endstop)
 =>  
Z-MIN - not stopping if move to it
It is config for CNC 
Similar issue  #12841

### Steps to Reproduce

1. Latest bugfix-2.0.x
2.  DIFF with default  Configuration.h
>   #define MOTHERBOARD BOARD_RAMPS_14_SF
> #define USE_XMAX_PLUG
> #define USE_YMAX_PLUG
> #define USE_ZMAX_PLUG
> #define ENDSTOP_INTERRUPTS_FEATURE
> // #define Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN
> #define Z_MIN_PROBE_PIN 32 // Pin 32 is the RAMPS default
> #define Z_HOME_DIR 1

3. DIFF with default  Configuration_adv.h
> #define ENDSTOPS_ALWAYS_ON_DEFAULT
>   #define ABORT_ON_ENDSTOP_HIT_FEATURE_ENABLED

**Expected behavior:** Upon reaching the Z-MIN ENDSTOP - this will stop the Z axis

**Actual behavior:** Upon reaching the Z-MIN ENDSTOP - this will not stop Z axis

#### I solve a problem in **endstops.cpp**
to the end of `    #elif Z_HOME_DIR < 0`
add 
` || Z_HOME_DIR > 0 && HAS_Z_MAX && DISABLED(Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN)` 

code with this edit:
  #if HAS_Z_MIN
    #if Z_MULTI_ENDSTOPS
    ***
    ***
    #elif Z_HOME_DIR < 0 || Z_HOME_DIR > 0 && HAS_Z_MAX && DISABLED(Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN)
      UPDATE_ENDSTOP_BIT(Z, MIN);
    #endif
  #endif

kpter, 2019-05-07:
[endstops.zip](https://github.com/MarlinFirmware/Marlin/files/3150980/endstops.zip)

robbycandra, 2019-05-09:
What about define USE_ZMIN_PLUG ?

kpter, 2019-05-09:
This is enabled by default, I wrote only the difference between the configurations. 
ALL of 6 endstops are enabled and hardware works fine. 
And if I'll made a change to '-1' of the parameter Z_HOME_DIR it (Z_MIN_Endstop) works fine. Bug in this behavior.
But this must be exactly Z_HOME_DIR 1 (and it is not work) in the CNC configuration, because the Z-probe measures the length of tool periodically, and the Z-axis can be moved down to the end of the Z-Min in some situations with tools that dive deep down parts.

kpter, 2019-05-10:
I think that my small correction can be added to the main code, if this does not contradict the logic of the 3D printer.
Especially, if such a combination of parameters for a 3D printer is not used at all.

boelle, 2019-07-21:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

kpter, 2019-07-27:
I solve a problem in 
Marlin/src/module/endstops.cpp
```
593c593
<     #elif Z_HOME_DIR < 0
---
>     #elif Z_HOME_DIR < 0 || Z_HOME_DIR > 0 && HAS_Z_MAX && DISABLED(Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN)
```

thinkyhead, 2019-10-13:
Please give us your configurations so we can compile and take a look at whether this is a user error or a code error.

thinkyhead, 2019-10-13:
Note that Marlin only supports one endstop per axis!

kpter, 2019-10-14:
> Please give us your configurations so we can compile and take a look at whether this is a user error or a code error.
Attached, includes configs and diffs.
[last_configs.zip](https://github.com/MarlinFirmware/Marlin/files/3723225/last_configs.zip)

kpter, 2019-10-15:
>Ok, so on re-re-reading your description, it now seems that what you want is to have a Z min endstop that is functional for the ABORT feature, but otherwise it is not used. That is an interesting case.

Yes you are right. This is what I want. This applies to the CNC.

closed 2019-10-15 by kpter