MarlinFirmware/Marlin #13928: [BUG] fan number lcd

el-quique, 2019-05-07:
Hi, configure the Marlin 2.0bugfix in a delta Geeetech Rostock with:
3 Extruders and SingleNozzle on a Board Rumba+ .
Compile and it works very well. But in the Temperature Menu show 3 fans and only control the fan 0 which is the active on the board.
The BUG is that more fans appear than those configured, In case someone knows how to repair it. Thank you very much for your time

[config_and_configadv.zip](https://github.com/MarlinFirmware/Marlin/files/3150632/config_and_configadv.zip)

el-quique, 2019-05-08:
Hi. The solution I found was to modify the file:
Marlin/src/lcd/menu/menu_temperature.cpp

#Original:
//
  // Fan Speed:
  //
  #if FAN_COUNT > 0
    #if HAS_FAN0
      MENU_MULTIPLIER_ITEM_EDIT_CALLBACK(percent, MSG_FAN_SPEED FAN_SPEED_1_SUFFIX, &thermalManager.lcd_tmpfan_speed[0], 0, 255, thermalManager.lcd_setFanSpeed0);
      #if ENABLED(EXTRA_FAN_SPEED)
        MENU_MULTIPLIER_ITEM_EDIT(percent, MSG_EXTRA_FAN_SPEED FAN_SPEED_1_SUFFIX, &thermalManager.new_fan_speed[0], 3, 255);
      #endif
    #endif
    #if HAS_FAN1 || (ENABLED(SINGLENOZZLE) && EXTRUDERS > 1)
      MENU_MULTIPLIER_ITEM_EDIT_CALLBACK(percent, MSG_FAN_SPEED " 2", &thermalManager.lcd_tmpfan_speed[1], 0, 255, thermalManager.lcd_setFanSpeed1);
      #if ENABLED(EXTRA_FAN_SPEED)
        MENU_MULTIPLIER_ITEM_EDIT(percent, MSG_EXTRA_FAN_SPEED " 2", &thermalManager.new_fan_speed[1], 3, 255);
      #endif
    #endif
    #if HAS_FAN2 || (ENABLED(SINGLENOZZLE) && EXTRUDERS > 2)
      MENU_MULTIPLIER_ITEM_EDIT_CALLBACK(percent, MSG_FAN_SPEED " 3", &thermalManager.lcd_tmpfan_speed[2], 0, 255, thermalManager.lcd_setFanSpeed2);
      #if ENABLED(EXTRA_FAN_SPEED)
        MENU_MULTIPLIER_ITEM_EDIT(percent, MSG_EXTRA_FAN_SPEED " 3", &thermalManager.new_fan_speed[2], 3, 255);
      #endif
    #endif
  #endif // FAN_COUNT > 0

#Modified:
//
  // Fan Speed:
  //
  #if FAN_COUNT > 0
    #if HAS_FAN0
      MENU_MULTIPLIER_ITEM_EDIT_CALLBACK(percent, MSG_FAN_SPEED FAN_SPEED_1_SUFFIX, &thermalManager.lcd_tmpfan_speed[0], 0, 255, thermalManager.lcd_setFanSpeed0);
      #if ENABLED(EXTRA_FAN_SPEED)
        MENU_MULTIPLIER_ITEM_EDIT(percent, MSG_EXTRA_FAN_SPEED FAN_SPEED_1_SUFFIX, &thermalManager.new_fan_speed[0], 3, 255);
      #endif
    #endif
   #if DISABLED(SINGLENOZZLE) && DISABLED(MIXING_EXTRUDER) // line add
    #if HAS_FAN1 && EXTRUDERS > 1 //line modified
      MENU_MULTIPLIER_ITEM_EDIT_CALLBACK(percent, MSG_FAN_SPEED " 2", &thermalManager.lcd_tmpfan_speed[1], 0, 255, thermalManager.lcd_setFanSpeed1);
      #if ENABLED(EXTRA_FAN_SPEED)
        MENU_MULTIPLIER_ITEM_EDIT(percent, MSG_EXTRA_FAN_SPEED " 2", &thermalManager.new_fan_speed[1], 3, 255);
      #endif
    #endif
    #if HAS_FAN2 && EXTRUDERS > 2
      MENU_MULTIPLIER_ITEM_EDIT_CALLBACK(percent, MSG_FAN_SPEED " 3", &thermalManager.lcd_tmpfan_speed[2], 0, 255, thermalManager.lcd_setFanSpeed2);
      #if ENABLED(EXTRA_FAN_SPEED)
        MENU_MULTIPLIER_ITEM_EDIT(percent, MSG_EXTRA_FAN_SPEED " 3", &thermalManager.new_fan_speed[2], 3, 255);
      #endif
    #endif
   #endif // line add
  #endif // FAN_COUNT > 0

InsanityAutomation, 2019-05-08:
The reason is standby fan speeds are stored, so when you change active tools it will set the fan to the corresponding stored value. Maybe not clear enough verbage, but intended.

el-quique, 2019-05-10:
Hi. In my case, the patch worked. But due to printing failures I decided to return to Marlin 1.1.9. The circles come out very oval and I have to print several important pieces. As soon as I can I will use the Marlin 2.0 again. Thx

el-quique, 2019-05-12:
> The reason is standby fan speeds are stored, so when you change active tools it will set the fan to the corresponding stored value. Maybe not clear enough verbage, but intended.
Hi. Effectively works as you say. But I think it would be more comfortable if only the number of fans supported or configured was seen. Because there were 5 or 6 extruders like the one with the diamond hotend, only 3 fans would be shown?

el-quique, 2019-05-12:
This is for Fan_Layer or Fan_Extruders?

InsanityAutomation, 2019-05-13:
part cooling fans, so i think as youre referring layer fans

boelle, 2019-07-21:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

boelle, 2019-09-24:
@el-quique is the issue still there?

closed 2019-10-12 by boelle