MarlinFirmware/Marlin #13932: [BUG] Extruder moves lock firmware (SKR v1.3, A4988 and TMC2208)

PeteBurgess, 2019-05-07:
### Description

No extrusion and firmware lockup on manual extruder move. 

Board is an SKR v1.3 with TMC2208s on X and Y, and A4988 on E and Z. I've seen this before but it's when I've made a complete hash of the config. I usually just create a new clean one and go from there, but this time I'm stumped.

### Steps to Reproduce

- Bring nozzle to temperature
- Move axis menu
- Attempt to extrude any amount of filament

**Expected behavior:** 
Filament extruded

**Actual behavior:** [What actually happens]
No extrusion and firmware locks up (can't make navigate menus etc.)

[configuration.zip](https://github.com/MarlinFirmware/Marlin/files/3152888/configuration.zip)

reloxx13, 2019-05-07:
sounds like this #13213

try disabling ADAPTIVE_STEP_SMOOTHING and try again.

but the skr should have enough resources.

PeteBurgess, 2019-05-07:
Potentially dumb question but I really don't understand the internals. Could buffer size affect this in any way ? The SKR's got plenty of resources and in order to try and troubleshoot I've removed Linear Advance and filament change options to free up memory.

I did fiddle with the buffer size though. Come to think of it, I've got onboard SD enabled too, which I didn't have turned on before.

LastDragon-ru, 2019-05-08:
Extruder doesn't work in StealthChop (there are several bugs about this) :( The good news it will work in SpreadCycle (but #13092) and/or Hybrid mode (`#define HYBRID_THRESHOLD`). In any case, also required:

```
#define MINIMUM_STEPPER_PULSE 1 // or 2
```

LastDragon-ru, 2019-05-08:
> Could buffer size affect this in any way ?

yup (eg my delta on msk sbase doesn't work with value higher `BLOCK_BUFFER_SIZE 32`)

PeteBurgess, 2019-05-08:
Extruder's on an A4988 so don't have to worry about hybrid mode. 

@reloxx13 Adaptive Step Smoothing didn't make any difference. Thanks for the suggestion though.

boelle, 2019-07-21:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

closed 2019-07-21 by boelle