MarlinFirmware/Marlin #13934: [FR] Progress for G29

guestisp, 2019-05-07:
During a G29, would be usefull to show, on LCD (and report via USB, for octoprint or similiar) the completed probes and the remaining ones, like:

`Calibrating 1/10`

Something similiar to the Original Prusa autocalibration.
Very useful with huge meshes and slow probe speed. My mesh is built with 10 (thinking to increase to 12) points, 3 probes and takes more than 1 hour. :)

Currently a generic "Doing G29" is shown, something more usefull like a progress indicator would be nice and useful

InsanityAutomation, 2019-05-16:
This has actually been on my list for awhile but low priority. Itll get done.

guestisp, 2019-05-16:
Thank you. Please also send the progress via USB, so that could be catched by octoprint or similiar

InsanityAutomation, 2019-05-16:
Good thought, would need a plugin to display outside the terminal window though. If it's there I'm sure someone will do it!

guestisp, 2019-05-16:
Exactly. I've tried to see the best place to add the progress. Probably in `unified_bed_leveling::probe_entire_mesh` by using `count` as the total number of points to probe

guestisp, 2019-05-16:
I did a small progress on this, i'm able to update the terminal, but i'm having trouble in writing a string LCD.
Any advice ?

InsanityAutomation, 2019-05-16:
Call lcd_setstatus(STRING)

Needs to handle bilinear and mesh leveling as well, so itll be more than just ubl. Probably tag the same lines I added onMeshUpdate for extui not long ago.

guestisp, 2019-05-16:
I've made a PR. This works for new mesh build, 3-points mesh tilting and multiple points mesh tilt (G29 J)

Isn't the bilinear dismissed by UBL ?

guestisp, 2019-05-16:
> No... Bi-Linear results in much less program memory being consumed. So for some people, that is a better solution. Also, Bi-Linear does the actual Z-Height correction calculations differently. So some people might prefer its approach.

Understood. So I can apply my PR to handle the bilinear as well, but ..... can you point me to the code :D ? in /feature/bedlevel I don't see the bilinear but only ubl/mbl/abl

Roxy-3D, 2019-05-16:
The Bi-Linear correction is done in the Planner Scheduling Code.   Roughly 1400 lines into the Planner.cpp module.

guestisp, 2019-05-18:
@thinkyhead thank you for merging my PR
As there is a suggestion to not use more than 15 points per axis, won't be better to use uint8 instead uint16 for grid size variable? The maximum storable number would be 225 (15*15) that would fit in a 8bit int

InsanityAutomation, 2019-10-15:
@boelle @shitcreek The initial PR was done and I finished extending it for bilinear and linear as well. This can be closed.

PR #14036 (2019-05-18): Add g29 progress update
Add progress updates for G29 meshes via terminal and via LCD 
Fix #13934
closed 2019-10-15 by boelle