MarlinFirmware/Marlin #13936: tabs and marlin size

fiegener, 2019-05-07:
Hi, I am a big fan of Marlin but I  have a question what's need to run marlin. Are all of the tabs in marlin necessary to keep. If just configuration h, and configuration adv h is all that I am playing with. Can I make just a Marlin folder with just these two tabs? Do I need all the languages. This also goes for the text, can I remove everything in Configuration H not pertaining to Code or #define ? I am just a fan of seeing everything on one page or one screen. Sorry if this a dumb question in the wrong place.

tcm0116, 2019-05-07:
Marlin 2.0 moved to a folder hierarchy, so you might consider giving that a try.

fiegener, 2019-05-07:
Thats in beta, with a clear sign that says "proceed with caution." I'll look at it.

tcm0116, 2019-05-07:
2.0 is very stable for several platforms and 1.1.x is not receiving updates.

fiegener, 2019-05-07:
TCM0116, I am looking for some support on Marlin. Not sure if this is the place for this, but there is no means to post directly to a member. are you a marlin expert where you may be able to work on a project. It's a paid gig.

tcm0116, 2019-05-07:
You can join us on Discord at:

https://discord.gg/n5NJ59y

You should be able to DM me, and we can discuss what you're trying to do.

closed 2019-05-10 by thinkyhead