MarlinFirmware/Marlin #13940: [BUG Marlin 1.1.9] LCD cycles in and out of different menus, when heat bed is turned on using LCD/ pronterface

Fr33man85, 2019-05-07:
Like this guy i have excatly the same bug. https://github.com/MarlinFirmware/Marlin/issues/12712

I own an Creality Ender 5 Printer and tryed some different Firmwares like TH3D, Stock Creality, Stock Marlin 1.1.9 and all ended up with the same Bug:
When Bed-Heating is on, the LCD jump´s into the Menu, and back to Info, back in Menu, back to Info,.... More or less fast, depending on - i think - the PID-Value calculated for the Cycle-time of the heating-element.

Only the Firmware, what was delivered, dont show this behavior. But this firmware is gone after the Update of the offered Stock Creality FW

Any ideas, why this happend?

greetings from Germany

Manuel

mhsuhail, 2019-05-08:
The main thread was closed without any solution.

Fr33man85, 2019-05-08:
Thanks for your reply.
i go ahead in this thread.

i try some other things, maybe i ask the support from Creality directly because it is also happening when i use the provided Firmware from they homepage (the Source files, not the Hex, because i flashed the bootloader)

but maybe thats the Problem?

Do you use a bootloader on your board? 
Which board your Prusa i3 clone use?

Fr33man85, 2019-05-08:
ah ok, 
but not so well documented in the general web :D

Do you have some links where i can read about it?

If this is a noise problem, maybe it can be solved to add some Pullup/down resistors to the corresponding Pins.

but anyway, i try to shield the ribbon cable instead

Fr33man85, 2019-05-08:
o.O

Or i first try to chance the cable complete

[image]

InsanityAutomation, 2019-05-08:
Looks like something found that cable a bit tasty to gnaw on! Its mostly on the Creality FB groups, but at least in your case hopefully its the obvious!

mhsuhail, 2019-05-08:
Thanks for the lead, InsanityAutomation, will give it a try when I get some time. Manuel I will appreciate if you can update me as well, if this works out for you. Thanks

Fr33man85, 2019-05-08:
So, a few frustrating hours later:

First, I renewed the cable. But without success.

[image]

Afterwards I thought about the statement of InsanityAutomation. This noise ... So I went in search of appropriate documentation on the pinout of the LCD.

But I did not find much, so I took the whole thing apart and measured myself.

The button does nothing else as pull a signal from a dedicated pin to ground. So the easiest thing to do was to put in a pullup resistor.

[image]

In the first attempt, I took a 10k.
Without putting everything together again, it looked good at first. The behavior did not recur.
I was already happy and put everything back together.
After that - oh, miracles - it happened again. This time, however, not nearly so bad.

So dismantled and changed the 10k to 4,7k.
Also tested in disassembled condition; looked good.
But here too: after assembly, it happened again.. grrrr

[image]

I think there might be an earth loop problem somewhere or something. Next, I will probably try to improve the grounding as a whole.

Fr33man85, 2019-05-10:
So,
I've added a few ground wires and I suggest everyone do the same or at least check it!
The entire frame (metal, conductive) is NOT grounded. The output of the power supply is also NOT connected to ground.

I'm not an expert, but I think that should not be so.

I also stripped a CAT5e Ethernet cable to collect the shield and put the recycled shield on the ribbon cable.

since 3h printing time the "auto-menu-click" has not happened again.

[image]
[image]
[image]
[image]

Fr33man85, 2019-05-13:
I close this because
1. i fixed it
2. it was not a Marlin issue

but thanks to all.
of course for the great firmware "Marlin"

closed 2019-05-13 by Fr33man85