MarlinFirmware/Marlin #13942: M122 stepper motor temp

cat007mine, 2019-05-08:
Hi,
Running Marlin 1.1.9 on an MKS Gen L board for 3D printing.

I have my stepper motor driver currents as 760 for XYZ and 900 for extruder.
M122 output below.
I'm trying to work out what the stepper motors temperatures are. Are the 4 numbers towards the end, 157C, 150C, 143C and 120C the temperature of the stepper motors? I would've thought that is way too hot for them?

Recv: 		X	Y	Z	E0
Recv: Enabled		false	false	false	false
Recv: Set current	760	760	760	900
Recv: RMS current	734	734	734	887
Recv: MAX current	1035	1035	1035	1251
Recv: Run current	23/31	23/31	23/31	28/31
Recv: Hold current	11/31	11/31	11/31	14/31
Recv: CS actual		0/31	0/31	11/31	28/31
Recv: PWM scale		76	95	39	0
Recv: vsense		1=.18	1=.18	1=.18	1=.18
Recv: stealthChop	false	true	true	true
Recv: msteps		16	16	16	16
Recv: tstep		0	0	1048575	5251
Recv: pwm
Recv: threshold		0	0	0	0
Recv: [mm/s]		-	-	-	-
Recv: OT prewarn	false	false	false	false
Recv: OT prewarn has
Recv: been triggered	false	false	false	false
Recv: off time		5	5	5	5
Recv: blank time	24	24	24	24
Recv: hysteresis
Recv: -end		2	2	2	2
Recv: -start		3	3	3	3
Recv: Stallguard thrs
Recv: DRVSTATUS	X	Y	Z	E0
Recv: stst				X
Recv: olb
Recv: ola
Recv: s2gb
Recv: s2ga
Recv: otpw
Recv: ot
Recv: 157C
Recv: 150C
Recv: 143C
Recv: 120C
Recv: s2vsa
Recv: s2vsb
Recv: Driver registers:	X = 0x40:17:00:00
Recv: 	Y = 0x40:17:00:00
Recv: 	Z = 0xC0:0B:00:00
Recv: 	E0 = 0x40:1C:00:00

teemuatlut, 2019-05-08:
No, they are the prewarn temperature thresholds for the TMC2208.

cat007mine, 2019-05-08:
Ah ok. Thanks. They are too hot to touch when running, well X and Y are. Z is cool and E is warm.
is there a way to get the temp from the motors with a gcode command? I can't find one.

teemuatlut, 2019-05-08:
No, use a thermal probe instead.
Remember that your hand is a very poor measurement device. You can mostly tell the difference between hot and cold but not how hot something is. The insulation material in motors is rated to 130C (a sample of one).

cat007mine, 2019-05-09:
Oh - ok so those pre-warn temps are for the drivers, not the motors? Ok I thought the motors had temp sensors in them.

teemuatlut, 2019-05-10:
Yes.

boelle, 2019-07-21:
will close this one as its more a misunderstanding than a bug

closed 2019-07-21 by boelle