MarlinFirmware/Marlin #13945: [BUG] (height error 327.67)

Goodfeat, 2019-05-08:
Hello, I found a bug in the marlin 2.0 firmware
 My printer - Delta, arduino DUE

The height in the UI interface does not match the real value.
The maximum height of my printer is 550mm.
Path errors: Ui menu-> configuration -> delta calibration -> delta setting-> height (+550) -> Height: 327.67 ???
I use REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER

In the Ui interface, when adjusting a value above 327.67, the firmware does not allow installation.
Most likely this is due to the restriction of the int_16 data type. The values ​​of the limit height coincides with the maximum limit for int16 - 32,768

[image]

InsanityAutomation, 2019-05-12:
Can you test with todays bugfix branch? A PR was just merged that may fix this. If not we may need to take a closer look. If this is now working, please go ahead and close the issue. Thanks.

Goodfeat, 2019-05-13:
Fixed!

closed 2019-05-13 by Goodfeat