MarlinFirmware/Marlin #13949: Fade Height Z can't be set

MupetS, 2019-05-08:
HI

I can't change value of Z offset , it's enabled in code ,M420 and option in menu doing nothing.I can press up and down and its still 0.

SENDING:G29
Bilinear Leveling Grid:
      0      1      2      3
 0 +1.061 +0.510 -0.005 -0.380
 1 +0.828 +0.265 -0.287 -0.670
 2 +0.609 +0.087 -0.498 -0.850
>>> M420 Z-0.8
SENDING:M420 Z-0.8
echo:Bed Leveling On.
echo:Fade Height Off.
>>> M420 S1
SENDING:M420 S1
echo:Bed Leveling On.
echo:Fade Height Off.

MupetS, 2019-05-08:
This is my config

//#define AUTO_BED_LEVELING_3POINT
//#define AUTO_BED_LEVELING_LINEAR
#define AUTO_BED_LEVELING_BILINEAR
//#define AUTO_BED_LEVELING_UBL
//#define MESH_BED_LEVELING

//#define RESTORE_LEVELING_AFTER_G28

//#define DEBUG_LEVELING_FEATURE

#if ENABLED(MESH_BED_LEVELING) || ENABLED(AUTO_BED_LEVELING_BILINEAR) || ENABLED(AUTO_BED_LEVELING_UBL)

  #define ENABLE_LEVELING_FADE_HEIGHT

InsanityAutomation, 2019-05-08:
Try https://github.com/MarlinFirmware/Marlin/pull/13948 may be related

MupetS, 2019-05-08:
nothing changed

InsanityAutomation, 2019-05-08:
Ok, probably related so ill take a look. Can you confirm which exact menu path you were in, and if babystepping is enabled?

MupetS, 2019-05-08:
Marlin/src/lcd/menu/menu_bed_leveling.cpp and yes

InsanityAutomation, 2019-05-12:
Can you test with todays bugfix branch? A PR was just merged that may fix this. If not we may need to take a closer look. If this is now working, please go ahead and close the issue. Thanks.

closed 2019-05-12 by MupetS