MarlinFirmware/Marlin #13950: Compiling error with REPRAPWORLD_GRAPHICAL_LCD

thpparent, 2019-05-08:
### Description

Compilation doesn't complete anymore (was still able to compile the version of the 27/04 with exactly the same configuration (config.h and adv)

Using Arduino 1.8.9

Just load marlin.ino and compile

Error 

Arduino: 1.8.9 (Windows 10), Board: "Arduino Due (Programming Port)"

sketch\src\HAL\HAL_DUE\u8g_com_HAL_DUE_sw_spi_shared.cpp.o: In function `u8g_SetPILevel_DUE(_u8g_t*, unsigned char, unsigned char)':

sketch\src\HAL\HAL_DUE/u8g_com_HAL_DUE_sw_spi_shared.cpp:71: multiple definition of `u8g_SetPIOutput_DUE(_u8g_t*, unsigned char)'

sketch\src\HAL\HAL_DUE\u8g_com_HAL_DUE_st7920_sw_spi.cpp.o:sketch\src\HAL\HAL_DUE/u8g_com_HAL_DUE_st7920_sw_spi.cpp:68: first defined here

sketch\src\HAL\HAL_DUE\u8g_com_HAL_DUE_sw_spi_shared.cpp.o: In function `DELAY_CYCLES':

sketch\src\HAL\HAL_DUE/u8g_com_HAL_DUE_sw_spi_shared.cpp:76: multiple definition of `u8g_SetPILevel_DUE(_u8g_t*, unsigned char, unsigned char)'

sketch\src\HAL\HAL_DUE\u8g_com_HAL_DUE_st7920_sw_spi.cpp.o:sketch\src\HAL\HAL_DUE/u8g_com_HAL_DUE_st7920_sw_spi.cpp:73: first defined here

collect2.exe: error: ld returned 1 exit status

exit status 1
Error compiling for board Arduino Due (Programming Port).

[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/3158993/Marlin.zip)

Bob-the-Kuhn, 2019-05-08:
We're working on it.   See PR #13939

Bob-the-Kuhn, 2019-05-09:
The corrected code has been added to the PR.

Bob-the-Kuhn, 2019-05-10:
The PR has been merged.

Please close this issue if the new code works.

thpparent, 2019-05-11:
Fixed by PR

closed 2019-05-11 by thpparent