MarlinFirmware/Marlin #13952: Missing zip file in repo?

howroyd, 2019-05-08:
The readme.MD states:

> Modified FreeRTOS files are in a .ZIP file alongside the Marlin source files.

Yet I cannot seem to find these in the repo.

Am I missing something obvious?

howroyd, 2019-05-08:
...to clarify, on the [Marlin_RTOS](https://github.com/MarlinFirmware/Marlin/tree/Marlin_RTOS) branch

Phr3d13, 2019-05-30:
I have no experience with this branch, but I also cannot find the zip in question. I also looked through the commit history and can't find when the file disappeared.

boelle, 2019-07-21:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

closed 2019-07-21 by boelle

howroyd, 2019-07-22:
Still no zip file.  Keen to help out with the FreeRTOS stuff (I use it at work on a STM32F4) but can't due to missing config files!

boelle, 2019-07-22:
i dont think they even have started or completed RTOS

@thinkyhead

howroyd, 2019-07-22:
Ok, if that's the case may I suggest putting something in the readme.md ...it's a bit confusing.

Also, still keen to help out if I can.

boelle, 2019-07-22:
well if you can there is always bugs to be sorted out

https://github.com/MarlinFirmware/Marlin/issues?q=is%3Aissue+is%3Aopen+label%3A%22Bug%3A+Potential+%3F%22+sort%3Acreated-asc

InsanityAutomation, 2019-07-23:
As 32 bit platforms are becoming more available, tests showed very little if any true gain with RTOS. Take a look at the complexity of a custom linux kernel linux cnc needs to maintain for it...

howroyd, 2019-07-23:
Yes I understand. Not criticising in any way. Just suggesting that the readme might be updated to reflect this?

InsanityAutomation, 2019-07-24:
> 
> 
> Yes I understand. Not criticising in any way. Just suggesting that the readme might be updated to reflect this?

Id agree, but im not the maintainer or a member of the core team so I dont usually mess with the general house keeping stuff :)