MarlinFirmware/Marlin #13955: Stepper motors are now very slow

cat007mine, 2019-05-09:
Hi all,

I have just installed an MKS Gen L with 4x TMC2208 drivers on my Ender 3 running in UART mode.
I have noticed the stepper motor speeds, even when I try and crank up the feed rate speed via the control knob on the display, are slow. It seems to be printing at about 40mm/s - and won't go any faster. 

I have the following set:
#define DEFAULT_MAX_FEEDRATE          { 500, 500, 5, 100 }

Current of the drivers via M122 showing
Set current	760	760	760	900

Steps per unit I have set as:
X80.00 Y80.00 Z400.00 E97.00

Am I missing something?

reloxx13, 2019-05-09:
open a terminal, enter 

```
g28
g0 x0 y0 z20 f5000
g0 x200 y200 z50 f5000
```

if its moving fast, fix ur slicer config.

and yes, u r missing to upload ur fw config files.

cat007mine, 2019-05-09:
Thanks! 
I will send those commands and let you know how it goes. Currently  printing so can't do it just yet!
Attached are my 2x Marlin config files.

[Configuration.h.txt](https://github.com/MarlinFirmware/Marlin/files/3162065/Configuration.h.txt)
[Configuration_adv.h.txt](https://github.com/MarlinFirmware/Marlin/files/3162066/Configuration_adv.h.txt)

cat007mine, 2019-05-13:
> open a terminal, enter
> 
> ```
> g28
> g0 x0 y0 z20 f5000
> g0 x200 y200 z50 f5000
> ```
> 
> if its moving fast, fix ur slicer config.
> 
> and yes, u r missing to upload ur fw config files.

Ok they all work ok, and yes they move at an alright speed - but not fast. 
For example, I ran G28 to home and then the last command: 

g0 x200 y200 z50 f5000 

That last command took just under 10 seconds. 
Is that normal? How long should it take?

Cheers!
Hunt

robbycandra, 2019-05-13:
Its because your max feedrate for Z is only 5.

robbycandra, 2019-05-13:
5 mm / second... sot travel to Z50 takes 50/5 second = 10 second.

cat007mine, 2019-05-13:
All axis seem to finish the above movement at the same time - taking pretty much bang on 10 seconds (as best I can time).

robbycandra, 2019-05-13:
Maybe you can try this 
#define DEFAULT_MAX_FEEDRATE { 400, 400, 30, 100 }

cat007mine, 2019-05-13:
Ah great - that's moved the Z axis at a much nicer speed. Is 30mm/s max about right? How come the XY axis's are set to so much faster but they seem to move at about the same speed? Is it one of those "they all move at the same speed as the slowest" type thing?

robbycandra, 2019-05-13:
Yes... Now you can close the issue.

AnHardt, 2019-05-13:
If you'd move only X and Y, without Z at the same time, that move would be much faster. All axes have to arrive their target a the same time.

MasterPIC, 2019-05-23:
Please close the issue!

closed 2019-05-23 by cat007mine