MarlinFirmware/Marlin #13961: [FR] Homing Sensorless (X- X+) (Y- Y+)

Alissonverd, 2019-05-09:
Dear Friends.
I'm improving my printer with 32bit based board. But the problem is a small number of unused ports free to development.
In my setup, I use the Min/Max endstops, so only for XY I use 4 logic ports.

Maybe the firmware will be able to use the same pin for Min/Max endstops when using a sensorless system. So, in according of direction of movement, when TMC driver report a stop, firmware will detect the Min/Max stop.
It will free more 2 pins to development when using a TMC driver.

Thank you,
Best regards.

teemuatlut, 2019-05-10:
Why do you need both min and max on the same axis?

Alissonverd, 2019-05-10:
@teemuatlut 
Here we have many users using this printer. One of the users has already damaged the belts, and how this printer are shared, it's impact the other users. =(
To solve the problem and make the equipment more reliable and available to everyone, we put the maximum endstops, and it did not happen anymore.

Is it not possible to implement this feature?

Best regards.

robbycandra, 2019-05-10:
Actually, if we edit the code, it is posible. 
Another better scenario is, We use 1 pin for X MIN and Y MIN limit, and 1 pin for X MAX and Y MAX limit.

If we use same port for MIN and MAX, then if we start printer and the limit switch triggered, we dont know the axis is in MIN or MAX pos, and dont know where to move first.

teemuatlut, 2019-05-10:
Marlin will only ever home to one direction. If you home to min with stallGuard, what's the purpose of the mechanical max endstop if it is not used? Or if you ship with mechanical max endstops, why use sensorless homing at all?
An endstop at max also would not save you from a failed sensorless homing procedure to min direction.

Sensorless homing should always be calibrated first. If you are worried about damaged hardware, you can always err on the side of caution and ship with a bit too sensitive setting. The final working config depends on the assembly as well, like how tight the belts are or how much restriction there is on the rail.

I'm just trying to understand the situation where both min and max endstops at the same time would be useful or necessary.

Alissonverd, 2019-05-14:
Solved with NO_MOTION_BEFORE_HOMING

closed 2019-05-14 by Alissonverd