MarlinFirmware/Marlin #13964: BUG Delta settings will not allow editing  value above 327.76.

jmdearras, 2019-05-09:
### Description
On my Much Modified Hatchboc Alpha, with SKRV1.1 cpu, and rods 356.5, height 344.59, the delta settings show the correct value, but when you either click to edit, the value reverts to 327.76 and cannot be increased.
code pulled from 5/8 Marlin 2.0.x, though the bug is older than that.
[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/3164140/Configuration.zip)

### Steps to Reproduce

1. [First Step]
2. [Second Step]
3. [and so on...]

**Expected behavior:** [What you expect to happen]

**Actual behavior:** [What actually happens]

#### Additional Information

* Include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

jmdearras, 2019-05-13:
Why I did not notice that was the max 16bit signed value is beyond me!  32,767!  I will test and see if issue #13976 fixed it.

InsanityAutomation, 2019-05-13:
So far every other issue opened because of this has been verified so im hopeful!

jmdearras, 2019-05-13:
Yes, #13976 fixed this.  Everytime I look at this description, I feel like a moron not noticing the digits were equal to 0x7fff, and that's all  I can see, now!  Thanks for the fix.

closed 2019-05-15 by jmdearras