MarlinFirmware/Marlin #13965: [BUG] (Long beep on start or reset)

SISLANGER, 2019-05-10:
### Description
Printing well on SKR V1.3 set EEFFB

On startup and reset getting large beep even if speaker disabled.
Weird blips before the Marlin Logo not weird characters from firmware update via SD.

Any ideas Why the beeping?

### Steps to Reproduce
In my case turning on or resetting.

**Expected behavior:** [What you expect to happen]
No beep on startup

**Actual behavior:** [What actually happens]
Banned from using printer at night because of loud long beep when resetting or turning on. I am trying to get PS-On and neopixel working on it so constantly changing config at moment.

ManuelMcLure, 2019-05-10:
Let me guess... you have a ReprapDiscount Full Graphics display? Unfortunately there's nothing Marlin can do about this - the RRD display will beep from receiving power until it has been properly initialized and that doesn't happen until after the bootloader has actually started Marlin. You'll notice that it actually beeps for longer when you upload new firmware because the bootloader takes longer to start Marlin in that case.

SISLANGER, 2019-05-10:
Well that answers that one, crap umm will have to look at other displays or maybe cut the buzzer pin wire thank you for your help

closed 2019-05-10 by SISLANGER

doggyfan, 2019-05-10:
Try soldering a 10k resistor from pin on the EXP1 socket to GND.

That will default the buzzer to being OFF rather than ON via a weak pull-up in the CPU on the controller board.

https://reprap.org/mediawiki/images/5/51/RRD_FULL_GRAPHIC_SMART_CONTROLER_SCHEMATIC.pdf

SISLANGER, 2019-05-10:
@doggyfan Brillant will give that a go the resistor that is ;-)

SISLANGER, 2019-05-10:
@doggyfan thank you works a treat

ManuelMcLure, 2019-05-12:
Yes, thank you @doggyfan - it worked great here as well.

xenon2008, 2019-07-30:
Hello!

Today i have set up my first SKR 1.3 Board fornmy Ender 3 Pro.
When i Power on the Printer or use the Reset Button of the SKR 1.3, i will get a terrible beep.
So i think this is this issue.. but i have the stock ender 3 display (activated cr_10 Stockdisplay in marlin..)

Any ideas what i can do?
At the first time i thought i havr made a mistske because this is my first skr...
With marlin 2.0 bugfix (older release) on stock ender 3 melzi board it doesnt beep so terrible.
Its sounds like an little explosion *lol*

I hope everbody can understand my bad english :(

Best regards
Mark

Kabaniero, 2020-01-03:
> Hello!
>
> Today i have set up my first SKR 1.3 Board fornmy Ender 3 Pro.
> When i Power on the Printer or use the Reset Button of the SKR 1.3, i will get a terrible beep.
> So i think this is this issue.. [...]

Working solution: solder a 100KΩ resistor between the 1th and 9th pin of exp3 port on Creality stock LCD board. 

Мне помогло припаять резистор на 100КОм между первым и девятым пином на стоковой плате дисплея Creality.
[image]