MarlinFirmware/Marlin #13968: [FR] Languague Full Spanish Marlin 1.1.9

el-quique, 2019-05-10:
### Description

### Steps to Reproduce

1. Copiar los archivos language_es.h y language_es_utf8.h en el directorio de Marlin

2. En Configuration.h agregar las siguientes lineas:

// Preheat Constants
#define PREHEAT_1_LABEL       "PLA" // linea agregada
#define PREHEAT_1_TEMP_HOTEND 185
#define PREHEAT_1_TEMP_BED     50
#define PREHEAT_1_FAN_SPEED     0 // Value from 0 to 255

#define PREHEAT_2_LABEL       "ABS" // linea agregada
#define PREHEAT_2_TEMP_HOTEND 200
#define PREHEAT_2_TEMP_BED     90
#define PREHEAT_2_FAN_SPEED     0 // Value from 0 to 255

3. En configuration,h elegir:
#define LCD_LANGUAGE es_utf8
#define DISPLAY_CHARSET_HD44780 WESTERN
o 
#define LCD_LANGUAGE es
#define DISPLAY_CHARSET_HD44780 WESTERN

#### Additional Information
[language_espanol-Marlin_1.1.9.zip](https://github.com/MarlinFirmware/Marlin/files/3164437/language_espanol-Marlin_1.1.9.zip)

el-quique, 2019-05-10:
Description
Spanish Language Update for Marlin 1.1.9

Steps to follow
Copy the language_es.h and language_es_utf8.h files to the Marlin directory

In Configuration.h add the following lines:

// Preheat Constants
#define PREHEAT_1_LABEL "PLA" // line added
#define PREHEAT_1_TEMP_HOTEND 185
#define PREHEAT_1_TEMP_BED 50
#define PREHEAT_1_FAN_SPEED 0 // Value from 0 to 255

#define PREHEAT_2_LABEL "ABS" // added line
#define PREHEAT_2_TEMP_HOTEND 200
#define PREHEAT_2_TEMP_BED 90
#define PREHEAT_2_FAN_SPEED 0 // Value from 0 to 255

In configuration, h choose:
#define LCD_LANGUAGE es_utf8
#define DISPLAY_CHARSET_HD44780 WESTERN
or
#define LCD_LANGUAGE is
#define DISPLAY_CHARSET_HD44780 WESTERN
Attached files

Ludy87, 2019-05-10:
create a PR to submit changes

Ludy87, 2019-05-10:
You mix the 1.1.x version with 2.0.x, that's not okay.

Ludy87, 2019-05-10:
For special characters it does not need an extra file, but
```#define THIS_LANGUAGES_SPECIAL_SYMBOLS      _UxGT("ÁáçÉéÍíÑñÓóÚúüªº¡¿")```

marcio-ao, 2019-05-13:
It looks like there is a separate PR for Marlin 2.0.x https://github.com/MarlinFirmware/Marlin/issues/13941

el-quique, 2019-05-13:
Adapted for Marlin 1.1.9 and work fine

boelle, 2019-07-21:
PR's should not be submitted as an issue with a zip file

boelle, 2019-07-21:
will close this one as its not a FR nor a BUG, i have mentioned the post to @thinkyhead so he can grab the zip file and see if they are of any use

closed 2019-07-21 by boelle