MarlinFirmware/Marlin #13972: [BUG] Freeze on M500 with UBL 9x9 grid or larger

gandy92, 2019-05-10:
### Description

I'm running the latest bugfix-1.1.x on a MKS GEN_L v1.0 board with a BL-Touch bed-leveling sensor and UBL enabled. When trying to store a freshly acquired UBL mesh sized 9x9 or 10x10 to eeprom,  `M500` will freeze the firmware.

### Steps to Reproduce

1. Compile w/ platformio, using the attached configuration
2. Upload the firmware with OctoPrint firmware uploader
3. `G28` for homing
4. `G29 P1` to acquire the mesh data
5. `G29 T` to control the mesh (optional)
5. `G29 S1` to save the mesh to slot 1
6. `M500` to store settings to eeprom

**Expected behavior:** `M500` responds w/:
```
Send: M500
Recv: echo:Settings Stored (655 bytes; crc 22556)
Recv: Mesh saved in slot 1
Recv: ok
```

**Actual behavior:** The machine never responds to M500. Instead, a timeout occurs in octoprint and the firmware is reset. The grid data can not be restored from eeprom w/ `G29 L1`.

#### Additional Information

* W/ `GRID_MAX_POINTS_X` set to 7, everything works fine.
* [marlin.zip](https://github.com/MarlinFirmware/Marlin/files/3167768/marlin.zip) w/ configuration files (UBL `GRID_MAX_POINTS_X` currently set to 7, which works; if set to 9 or 10 the crash happens. No other values were tested)

shitcreek, 2019-05-11:
Anything greater than 7 can cause performance and memory issues. This is due to board limitation - it's not a bug.

InsanityAutomation, 2019-05-11:
With beefier boards Im running 15x15 right now fwiw, including a snapshot from a few hours ago.

gandy92, 2019-05-11:
@shitcreek I can see how too much memory consumption can be an issue. Is this a known issue that I just missed in the documentation? Just wondering what else I should know about the MKS GEN_L? I suppose a limitation like that is too complex for the sanity check to notice? 

@InsanityAutomation what kind of boards are you talking?

shitcreek, 2019-05-11:
@gandy92 I found this write up while researching the issue: https://www.thingiverse.com/thing:2783450 - it's not in the documentation, yet.

thinkyhead, 2019-05-16:
Marlin 1.1.x is end of life. Please test `bugfix-2.0.x` and if the issue is resolved this can be closed.

gandy92, 2019-06-02:
Couldn't get bugfix-2.0.x running on my MKS GEN_L v1.0 (Atmega2560) with TMC2208 in UART mode. This version fails communicating with the drivers (all LOW) and the motors wouldn't move at all, so I didn't quite get to testing larger mesh sizes. I'll be switching to a 32-bit ARM board, soon, so I won't be able to make further tests. 
May I suggest documenting the mesh size limitation of the ATmega2560 platform, somewhere?

closed 2019-06-02 by gandy92