MarlinFirmware/Marlin #13973: Marlin 2.0 Error[BUG] (short description)

inventabuild, 2019-05-10:
I followed directions for opening and uploading v2.0, but Arduino IDE gives me the error below.  When I open the .ino in Arduino I only see 6 tabs, not the usual numerous quantity of tabs.  Do all of the files need to be located in the same folder in order for Arduino to compile the firmware?

The Marlin Developers have all the files in separate folders, how do I put them in the same folder so Arduino will open the entire firmware and compile it?

Error Message When Trying To Compile v2.0.

Arduino: 1.8.5 (Windows 10), Board: "Arduino/Genuino Mega or Mega 2560, ATmega2560 (Mega 2560)"

C:\Program Files (x86)\Arduino\1.8.5\arduino-builder -dump-prefs -logger=machine -hardware C:\Program Files (x86)\Arduino\1.8.5\hardware -tools C:\Program Files (x86)\Arduino\1.8.5\tools-builder -tools C:\Program Files (x86)\Arduino\1.8.5\hardware\tools\avr -built-in-libraries C:\Program Files (x86)\Arduino\1.8.5\libraries -libraries C:\Users\Mitch\Documents\Arduino\libraries -fqbn=arduino:avr:mega:cpu=atmega2560 -ide-version=10805 -build-path C:\Users\Mitch\AppData\Local\Temp\arduino_build_751317 -warnings=none -build-cache C:\Users\Mitch\AppData\Local\Temp\arduino_cache_893 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.arduinoOTA.path=C:\Program Files (x86)\Arduino\1.8.5\hardware\tools\avr -prefs=runtime.tools.avr-gcc.path=C:\Program Files (x86)\Arduino\1.8.5\hardware\tools\avr -prefs=runtime.tools.avrdude.path=C:\Program Files (x86)\Arduino\1.8.5\hardware\tools\avr -verbose D:\3D Printers\lulzbot\Marlin\Printer 1 - Marlin-src_2.0.0.110_f1ffaa5da_From_Lulzbot_Website\Marlin-src_2.0.0.110_f1ffaa5da\Marlin\Marlin.ino
C:\Program Files (x86)\Arduino\1.8.5\arduino-builder -compile -logger=machine -hardware C:\Program Files (x86)\Arduino\1.8.5\hardware -tools C:\Program Files (x86)\Arduino\1.8.5\tools-builder -tools C:\Program Files (x86)\Arduino\1.8.5\hardware\tools\avr -built-in-libraries C:\Program Files (x86)\Arduino\1.8.5\libraries -libraries C:\Users\Mitch\Documents\Arduino\libraries -fqbn=arduino:avr:mega:cpu=atmega2560 -ide-version=10805 -build-path C:\Users\Mitch\AppData\Local\Temp\arduino_build_751317 -warnings=none -build-cache C:\Users\Mitch\AppData\Local\Temp\arduino_cache_893 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.arduinoOTA.path=C:\Program Files (x86)\Arduino\1.8.5\hardware\tools\avr -prefs=runtime.tools.avr-gcc.path=C:\Program Files (x86)\Arduino\1.8.5\hardware\tools\avr -prefs=runtime.tools.avrdude.path=C:\Program Files (x86)\Arduino\1.8.5\hardware\tools\avr -verbose D:\3D Printers\lulzbot\Marlin\Printer 1 - Marlin-src_2.0.0.110_f1ffaa5da_From_Lulzbot_Website\Marlin-src_2.0.0.110_f1ffaa5da\Marlin\Marlin.ino
Using board 'mega' from platform in folder: C:\Program Files (x86)\Arduino\1.8.5\hardware\arduino\avr
Using core 'arduino' from platform in folder: C:\Program Files (x86)\Arduino\1.8.5\hardware\arduino\avr
Detecting libraries used...
"C:\Program Files (x86)\Arduino\1.8.5\hardware\tools\avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -flto -w -x c++ -E -CC -mmcu=atmega2560 -DF_CPU=16000000L -DARDUINO=10805 -DARDUINO_AVR_MEGA2560 -DARDUINO_ARCH_AVR   "-IC:\Program Files (x86)\Arduino\1.8.5\hardware\arduino\avr\cores\arduino" "-IC:\Program Files (x86)\Arduino\1.8.5\hardware\arduino\avr\variants\mega" "C:\Users\Mitch\AppData\Local\Temp\arduino_build_751317\sketch\Marlin.ino.cpp" -o "nul"
"C:\Program Files (x86)\Arduino\1.8.5\hardware\tools\avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -flto -w -x c++ -E -CC -mmcu=atmega2560 -DF_CPU=16000000L -DARDUINO=10805 -DARDUINO_AVR_MEGA2560 -DARDUINO_ARCH_AVR   "-IC:\Program Files (x86)\Arduino\1.8.5\hardware\arduino\avr\cores\arduino" "-IC:\Program Files (x86)\Arduino\1.8.5\hardware\arduino\avr\variants\mega" "C:\Users\Mitch\AppData\Local\Temp\arduino_build_751317\sketch\src\HAL\HAL_AVR\HAL.cpp" -o "nul"
"C:\Program Files (x86)\Arduino\1.8.5\hardware\tools\avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -flto -w -x c++ -E -CC -mmcu=atmega2560 -DF_CPU=16000000L -DARDUINO=10805 -DARDUINO_AVR_MEGA2560 -DARDUINO_ARCH_AVR   "-IC:\Program Files (x86)\Arduino\1.8.5\hardware\arduino\avr\cores\arduino" "-IC:\Program Files (x86)\Arduino\1.8.5\hardware\arduino\avr\variants\mega" "C:\Users\Mitch\AppData\Local\Temp\arduino_build_751317\sketch\src\HAL\HAL_AVR\HAL_spi_AVR.cpp" -o "nul"
"C:\Program Files (x86)\Arduino\1.8.5\hardware\tools\avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -flto -w -x c++ -E -CC -mmcu=atmega2560 -DF_CPU=16000000L -DARDUINO=10805 -DARDUINO_AVR_MEGA2560 -DARDUINO_ARCH_AVR   "-IC:\Program Files (x86)\Arduino\1.8.5\hardware\arduino\avr\cores\arduino" "-IC:\Program Files (x86)\Arduino\1.8.5\hardware\arduino\avr\variants\mega" "C:\Users\Mitch\AppData\Local\Temp\arduino_build_751317\sketch\src\HAL\HAL_AVR\MarlinSerial.cpp" -o "nul"
"C:\Program Files (x86)\Arduino\1.8.5\hardware\tools\avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -flto -w -x c++ -E -CC -mmcu=atmega2560 -DF_CPU=16000000L -DARDUINO=10805 -DARDUINO_AVR_MEGA2560 -DARDUINO_ARCH_AVR   "-IC:\Program Files (x86)\Arduino\1.8.5\hardware\arduino\avr\cores\arduino" "-IC:\Program Files (x86)\Arduino\1.8.5\hardware\arduino\avr\variants\mega" "C:\Users\Mitch\AppData\Local\Temp\arduino_build_751317\sketch\src\HAL\HAL_AVR\fast_pwm.cpp" -o "nul"
"C:\Program Files (x86)\Arduino\1.8.5\hardware\tools\avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -flto -w -x c++ -E -CC -mmcu=atmega2560 -DF_CPU=16000000L -DARDUINO=10805 -DARDUINO_AVR_MEGA2560 -DARDUINO_ARCH_AVR   "-IC:\Program Files (x86)\Arduino\1.8.5\hardware\arduino\avr\cores\arduino" "-IC:\Program Files (x86)\Arduino\1.8.5\hardware\arduino\avr\variants\mega" "C:\Users\Mitch\AppData\Local\Temp\arduino_build_751317\sketch\src\HAL\HAL_AVR\fastio_AVR.cpp" -o "nul"
"C:\Program Files (x86)\Arduino\1.8.5\hardware\tools\avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -flto -w -x c++ -E -CC -mmcu=atmega2560 -DF_CPU=16000000L -DARDUINO=10805 -DARDUINO_AVR_MEGA2560 -DARDUINO_ARCH_AVR   "-IC:\Program Files (x86)\Arduino\1.8.5\hardware\arduino\avr\cores\arduino" "-IC:\Program Files (x86)\Arduino\1.8.5\hardware\arduino\avr\variants\mega" "C:\Users\Mitch\AppData\Local\Temp\arduino_build_751317\sketch\src\HAL\HAL_AVR\persistent_store_eeprom.cpp" -o "nul"
"C:\Program Files (x86)\Arduino\1.8.5\hardware\tools\avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -flto -w -x c++ -E -CC -mmcu=atmega2560 -DF_CPU=16000000L -DARDUINO=10805 -DARDUINO_AVR_MEGA2560 -DARDUINO_ARCH_AVR   "-IC:\Program Files (x86)\Arduino\1.8.5\hardware\arduino\avr\cores\arduino" "-IC:\Program Files (x86)\Arduino\1.8.5\hardware\arduino\avr\variants\mega" "C:\Users\Mitch\AppData\Local\Temp\arduino_build_751317\sketch\src\HAL\HAL_AVR\servo_AVR.cpp" -o "nul"
"C:\Program Files (x86)\Arduino\1.8.5\hardware\tools\avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -flto -w -x c++ -E -CC -mmcu=atmega2560 -DF_CPU=16000000L -DARDUINO=10805 -DARDUINO_AVR_MEGA2560 -DARDUINO_ARCH_AVR   "-IC:\Program Files (x86)\Arduino\1.8.5\hardware\arduino\avr\cores\arduino" "-IC:\Program Files (x86)\Arduino\1.8.5\hardware\arduino\avr\variants\mega" "C:\Users\Mitch\AppData\Local\Temp\arduino_build_751317\sketch\src\HAL\HAL_AVR\watchdog_AVR.cpp" -o "nul"
"C:\Program Files (x86)\Arduino\1.8.5\hardware\tools\avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -flto -w -x c++ -E -CC -mmcu=atmega2560 -DF_CPU=16000000L -DARDUINO=10805 -DARDUINO_AVR_MEGA2560 -DARDUINO_ARCH_AVR   "-IC:\Program Files (x86)\Arduino\1.8.5\hardware\arduino\avr\cores\arduino" "-IC:\Program Files (x86)\Arduino\1.8.5\hardware\arduino\avr\variants\mega" "C:\Users\Mitch\AppData\Local\Temp\arduino_build_751317\sketch\src\HAL\HAL_DUE\DebugMonitor_Due.cp [... cut ...]
C:\Users\Mitch\AppData\Local\Temp\arduino_build_751317\sketch\src\inc/../pins/pins_ARCHIM2.h:40:4: error: #error "Oops! Select 'Archim' in 'Tools > Board.'"

   #error "Oops! Select 'Archim' in 'Tools > Board.'"

    ^

In file included from C:\Users\Mitch\AppData\Local\Temp\arduino_build_751317\sketch\src\Lulzbot_Extras.cpp:24:0:

C:\Users\Mitch\AppData\Local\Temp\arduino_build_751317\sketch\src\module/stepper_indirection.h:59:26: fatal error: TMCStepper.h: No such file or directory

   #include <TMCStepper.h>

                          ^

compilation terminated.

exit status 1
Error compiling for board Arduino/Genuino Mega or Mega 2560.

ManuelMcLure, 2019-05-10:
You need to upgrade to Arduino 1.8.9 or better to compile 2.0 - 1.8.5 will not work. Or better still use PlatformIO instead of Arduino: http://marlinfw.org/docs/basics/install_platformio.html

inventabuild, 2019-05-10:
I tried 1.8.9 and got the same error:

Arduino: 1.8.9 (Windows 10), Board: "Arduino/Genuino Mega or Mega 2560, ATmega2560 (Mega 2560)"

C:\Program Files (x86)\Arduino\arduino-builder -dump-prefs -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\Mitch\Documents\Arduino\libraries -fqbn=arduino:avr:mega:cpu=atmega2560 -ide-version=10809 -build-path C:\Users\Mitch\AppData\Local\Temp\arduino_build_42049 -warnings=none -build-cache C:\Users\Mitch\AppData\Local\Temp\arduino_cache_845280 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.avr-gcc.path=C:\Program Files (x86)\Arduino\hardware\tools\avr -prefs=runtime.tools.avr-gcc-5.4.0-atmel3.6.1-arduino2.path=C:\Program Files (x86)\Arduino\hardware\tools\avr -prefs=runtime.tools.avrdude.path=C:\Program Files (x86)\Arduino\hardware\tools\avr -prefs=runtime.tools.avrdude-6.3.0-arduino14.path=C:\Program Files (x86)\Arduino\hardware\tools\avr -prefs=runtime.tools.arduinoOTA.path=C:\Program Files (x86)\Arduino\hardware\tools\avr -prefs=runtime.tools.arduinoOTA-1.2.1.path=C:\Program Files (x86)\Arduino\hardware\tools\avr -verbose D:\3D Printers\lulzbot\Marlin\Printer 1 - Marlin-src_2.0.0.110_f1ffaa5da_From_Lulzbot_Website\Marlin-src_2.0.0.110_f1ffaa5da\Marlin\Marlin.ino
C:\Program Files (x86)\Arduino\arduino-builder -compile -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\Mitch\Documents\Arduino\libraries -fqbn=arduino:avr:mega:cpu=atmega2560 -ide-version=10809 -build-path C:\Users\Mitch\AppData\Local\Temp\arduino_build_42049 -warnings=none -build-cache C:\Users\Mitch\AppData\Local\Temp\arduino_cache_845280 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.avr-gcc.path=C:\Program Files (x86)\Arduino\hardware\tools\avr -prefs=runtime.tools.avr-gcc-5.4.0-atmel3.6.1-arduino2.path=C:\Program Files (x86)\Arduino\hardware\tools\avr -prefs=runtime.tools.avrdude.path=C:\Program Files (x86)\Arduino\hardware\tools\avr -prefs=runtime.tools.avrdude-6.3.0-arduino14.path=C:\Program Files (x86)\Arduino\hardware\tools\avr -prefs=runtime.tools.arduinoOTA.path=C:\Program Files (x86)\Arduino\hardware\tools\avr -prefs=runtime.tools.arduinoOTA-1.2.1.path=C:\Program Files (x86)\Arduino\hardware\tools\avr -verbose D:\3D Printers\lulzbot\Marlin\Printer 1 - Marlin-src_2.0.0.110_f1ffaa5da_From_Lulzbot_Website\Marlin-src_2.0.0.110_f1ffaa5da\Marlin\Marlin.ino
Using board 'mega' from platform in folder: C:\Program Files (x86)\Arduino\hardware\arduino\avr
Using core 'arduino' from platform in folder: C:\Program Files (x86)\Arduino\hardware\arduino\avr
Detecting libraries used...
"C:\\Program Files (x86)\\Arduino\\hardware\\tools\\avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -flto -w -x c++ -E -CC -mmcu=atmega2560 -DF_CPU=16000000L -DARDUINO=10809 -DARDUINO_AVR_MEGA2560 -DARDUINO_ARCH_AVR "-IC:\\Program Files (x86)\\Arduino\\hardware\\arduino\\avr\\cores\\arduino" "-IC:\\Program Files (x86)\\Arduino\\hardware\\arduino\\avr\\variants\\mega" "C:\\Users\\Mitch\\AppData\\Local\\Temp\\arduino_build_42049\\sketch\\Marlin.ino.cpp" -o nul
"C:\\Program Files (x86)\\Arduino\\hardware\\tools\\avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -flto -w -x c++ -E -CC -mmcu=atmega2560 -DF_CPU=16000000L -DARDUINO=10809 -DARDUINO_AVR_MEGA2560 -DARDUINO_ARCH_AVR "-IC:\\Program Files (x86)\\Arduino\\hardware\\arduino\\avr\\cores\\arduino" "-IC:\\Program Files (x86)\\Arduino\\hardware\\arduino\\avr\\variants\\mega" "C:\\Users\\Mitch\\AppData\\Local\\Temp\\arduino_build_42049\\sketch\\src\\HAL\\HAL_AVR\\HAL.cpp" -o nul
Error while detecting libraries included by C:\Users\Mitch\AppData\Local\Temp\arduino_build_42049\sketch\src\HAL\HAL_AVR\HAL.cpp
"C:\\Program Files (x86)\\Arduino\\hardware\\tools\\avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -flto -w -x c++ -E -CC -mmcu=atmega2560 -DF_CPU=16000000L -DARDUINO=10809 -DARDUINO_AVR_MEGA2560 -DARDUINO_ARCH_AVR "-IC:\\Program Files (x86)\\Arduino\\hardware\\arduino\\avr\\cores\\arduino" "-IC:\\Program Files (x86)\\Arduino\\hardware\\arduino\\avr\\variants\\mega" "C:\\Users\\Mitch\\AppData\\Local\\Temp\\arduino_build_42049\\sketch\\src\\HAL\\HAL_AVR\\HAL_spi_AVR.cpp" -o nul
Error while detecting libraries included by C:\Users\Mitch\AppData\Local\Temp\arduino_build_42049\sketch\src\HAL\HAL_AVR\HAL_spi_AVR.cpp
"C:\\Program Files (x86)\\Arduino\\hardware\\tools\\avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -flto -w -x c++ -E -CC -mmcu=atmega2560 -DF_CPU=16000000L -DARDUINO=10809 -DARDUINO_AVR_MEGA2560 -DARDUINO_ARCH_AVR "-IC:\\Program Files (x86)\\Arduino\\hardware\\arduino\\avr\\cores\\arduino" "-IC:\\Program Files (x86)\\Arduino\\hardware\\arduino\\avr\\variants\\mega" "C:\\Users\\Mitch\\AppData\\Local\\Temp\\arduino_build_42049\\sk [... cut ...]
c:\users\mitch\appdata\local\temp\arduino_build_42049\sketch\src\pins\pins_ARCHIM2.h:40:4: error: #error "Oops! Select 'Archim' in 'Tools > Board.'"

   #error "Oops! Select 'Archim' in 'Tools > Board.'"

    ^

In file included from C:\Users\Mitch\AppData\Local\Temp\arduino_build_42049\sketch\src\Lulzbot_Extras.cpp:24:0:

C:\Users\Mitch\AppData\Local\Temp\arduino_build_42049\sketch\src\module/stepper_indirection.h:59:26: fatal error: TMCStepper.h: No such file or directory

compilation terminated.

exit status 1
Error compiling for board Arduino/Genuino Mega or Mega 2560.

inventabuild, 2019-05-11:
The issue is as I stated in my first post that the Marlin v2.0 folder with Marlin.ino only contains 4 files (see 1st attachment), the Marlin v1.1.9 folder with Marlin.ino contains all the files needed to open and compile Marlin (see 2nd attachment).
[image]
[image]
Why do you separate the firmware files into separate folders for v2.0?  Arduino doesn't know how to grab them all to create the firmware.

Please advise how I can open Marlin v2.0 so that all the files are open in the same environment so I can modify and compile the firmware and upload it, preferably w/ Arduino.

InsanityAutomation, 2019-05-11:
Arduino does grab them, just doesnt show them in the UI for easy editing. We are also supporting boards that work outside of the Arduino frameworks now. Ensure you are using Arduino IDE version 1.8.7 or newer in order to have the proper libraries and path support. Or use PlatformIO with VSCode.

wesk-ms, 2019-05-11:
The Marlin 2.0 source layout is entirely different than 1.x.x.  The files are **supposed** to be in the folder hierarchy you see when unzipping the package.  The header files (.h) specify where all the various files are located.  It's not up to the Arduino IDE to figure it out.

If you are manually moving all of the source files into a single folder, then that's the problem.

boelle, 2019-07-21:
so this is more a user misunderstanding than a bug?

boelle, 2019-07-21:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

closed 2019-07-21 by boelle