MarlinFirmware/Marlin #13977: [BUG] Custom Boot Scree not shown n skr v1.3 and mks_mini 12864

DrDitto, 2019-05-11:
### Description

### Steps to Reproduce

1. Dopwnload latest version
2  show_custom_boot_screen
3. copy the _Bootscreen.h to location
4. Compile and upload

**Expected behavior:** [Custom boot screen must to be shown]

**Actual behavior:** [Blank scree and then the marlin logo]

#### Additional Information

* Include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

reloxx13, 2019-05-11:
You know, you are supposed to fill the issue form.

Have you read Marlin's Code of Conduct? By filing an Issue, you are expected to comply with it, including treating everyone with respect: https://github.com/MarlinFirmware/Marlin/blob/master/.github/code_of_conduct.md

Do you want to ask a question? Are you looking for support? Please don't post here. Instead please use the Marlin Firmware forum at http://forums.reprap.org/list.php?415 or the Marlin Facebook Group https://www.facebook.com/groups/1049718498464482/.

Before filing an issue be sure to test the 1.1 and/or 2.0 "bugfix" branches to see whether the issue is already addressed.

### Description

### Steps to Reproduce

1. [First Step]
2. [Second Step]
3. [and so on...]

**Expected behavior:** [What you expect to happen]

**Actual behavior:** [What actually happens]

#### Additional Information

* Include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

Patag, 2019-06-05:
@reloxx13 : Any other constructive and smart replies such this one ? Maybe a fix since all this time ?

reloxx13, 2019-06-05:
Not with missing informations.

DrDitto, 2019-06-06:
Ok try this
PR #14216 but change the data start to this
static const uint8_t u8g_dev_uc1701_mini12864_HAL_data_start[] PROGMEM = {
  //DrDitto
      #if ENABLED(MKS_MINI_12864)
          U8G_ESC_ADR(0),             /* instruction mode */
          U8G_ESC_CS(1),              /* enable chip */
          0x040, /* set display start line to 0 */
          0x0a0, /* ADC set to reverse */
          0x0c8, /* common output mode */
          0x0a6, /* display normal, bit val 0: LCD pixel off. */
          0x0a2, /* LCD bias 1/9 */
          0x02f, /* all power control circuits on */
          0x0f8, /* set booster ratio to */
          0x000, /* 4x */
          0x023, /* set V0 voltage resistor ratio to large */
          0x0ac, /* indicator */
          0x000, /* disable */
          0x0af, /* display on */
          0x010, /* set upper 4 bit of the col adr to 0 */
          U8G_ESC_END                 /* end of sequence */
      #else
          U8G_ESC_ADR(0),             /* instruction mode */
          U8G_ESC_CS(1),              /* enable chip */
          UC1701_COLUMN_ADR(0),       /* address 0 */
          U8G_ESC_END                 /* end of sequence */
      #endif

};

boelle, 2019-07-21:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

closed 2019-07-21 by boelle

DrDitto, 2019-07-28:
I am loading the fix in a new fork