MarlinFirmware/Marlin #13979: Thermal runaway:bed at exactly halfway through print

bill-orange, 2019-05-11:
### Description

### Steps to Reproduce

1. [First Step]
2. [Second Step]
3. [and so on...]

**Expected behavior:** [What you expect to happen]

**Actual behavior:** [What actually happens]

#### Additional Information

* Include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

JamesKey-iq, 2019-05-11:
I also seem to be having this problem on longer prints

boelle, 2019-07-21:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

closed 2019-07-21 by boelle

bill-orange, 2019-07-21:
I did solve it.  The failure being located exactly at the halfway point was a red herring.  My bed heat power is provided by a FET board.  There was a cold and broken solder joint on the FET board output.  I re-soldered the joint and the problem was solved.

boelle, 2019-07-21:
:-)