MarlinFirmware/Marlin #13980: [QUESTION] Compile error on SKR1.1 with BLTOUCH

DavidThijs, 2019-05-11:
Whenever I try to get a very recent build of Marlin 2.0 to work on my Bigtreetech SKR1.1 board while enabling a BL_TOUCH probe. 
The pins I want to use are the regular Z min pin and as Servo0 pin P1_26

This compile error is thrown:

_Compiling .pioenvs\LPC1768\src\src\feature\closedloop.cpp.o
In file included from Marlin\src\module/../HAL/shared/servo.h:76,
from Marlin\src\module/servo.h:29,
from Marlin\src\Marlin.cpp:93:
Marlin\src\module/../HAL/shared/../HAL_LPC1768/MarlinServo.h: In member function 'void MarlinServo::move(int)':
Marlin\src\module/../HAL/shared/../HAL_LPC1768/MarlinServo.h:59:22: error: 'servo_info' was not declared in this scope
if (this->attach(servo_info[this->servoIndex].Pin.nbr) >= 0) {    // try to reattach
^~~~~~~~~~
In file included from Marlin\src\feature\../module/../HAL/shared/servo.h:76,
from Marlin\src\feature\../module/servo.h:29,
from Marlin\src\feature\bltouch.cpp:31:
g:\marlin20_skr\marlin\marlin\src\hal\hal_lpc1768\marlinservo.h: In member function 'void MarlinServo::move(int)':
g:\marlin20_skr\marlin\marlin\src\hal\hal_lpc1768\marlinservo.h:59:22: error: 'servo_info' was not declared in this scope
if (this->attach(servo_info[this->servoIndex].Pin.nbr) >= 0) {    // try to reattach_

My current configs are:

**Configuration.h:**
```
#define NUM_SERVOS 1 
#define SERVO_DELAY { 300 }
#define Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN
#define Z_PROBE_SERVO_NR 0 
#define BLTOUCH
```

**Pins_BIQU_SKR_v1.1.h:**
```
#define SERVO0_PIN P1_26
//disabled because reused as SERVO0_PIN
//#define Y_MAX_PIN          P1_26
#define Z_MIN_PIN          P1_25
#define Z_MAX_PIN          P1_24
```

The board is working with an LCD128x64 and TMC stepper drivers and SCK is connected to P2.6, so that one isn't free anymore. Has someone a clue because the code in MarlinServo.h and Servo.h doesn't make me much wiser.

InsanityAutomation, 2019-05-11:
Servo output needs to be an analog capable pin. Not sure the one stolen from the y endstop is, I'm not an expert on the platform. That'd be the first thing I'd confirm though.

p3p, 2019-05-11:
Make sure your frameworks and libraries are up to date, if the error persists post complete configs here in a zip file.

@InsanityAutomation Every pin on the LPC176x can be used as a PWM pin due to software fallback the framework uses, it's not as stable as the Hardware pins but usable, saying that P1_26 is a hardware capable pin, shares a channel with P2_05.

p3p, 2019-05-11:
> The pins I want to use are the regular Z min pin and as Servo0 pin P1_26

@DavidThijs Although this should build, endstop input pins tend to have filtering circuits on the board this will not interact well with using it as an output pin.

DavidThijs, 2019-05-11:
> 
> 
> Servo output needs to be an analog capable pin. Not sure the one stolen from the y endstop is, I'm not an expert on the platform. That'd be the first thing I'd confirm though.

Strange, I tought that the PIN should be PWM capable to provide commands to the ATTINY to stow, deploy, reset etc...

For Smoothieboards (also LPC1768), the pins 1.23, 1.24 and 1.26 should be ok. Hardware wise, a resistor is to be added but that doesn't affect compiling :)
See: [(https://www.antclabs.com/wiring32)]

@p3p,

My issue is a compile error. The hardware will be dealt with later on. I can unsolder the cap for debouncing, no issue !

p3p, 2019-05-11:
> My issue is a compile error.

I can see your issue is a compile error, tis why I asked you to make sure you are using the current version of the framework (framework-arduino-lpc176x @ 0.1.3) and if the error persists supply full configs in a zip file.

DavidThijs, 2019-05-11:
@p3p 
Chris, I have just updated PlatformIO and also the platform/libraries:
NXPLPC from 4.2.0 > 4.3.0
Neopixel from 1.2.1 > 1.2.2
Servo from 1.1.3 > 1.1.4
These were the only ones marked in Atom as requiring an update. 
The Marlin bugfix 2.0 version that I used was the one available on 21/04/2019, so quite recent.

All modified config/pins files are attached.
[config_marlin20_21042019.zip](https://github.com/MarlinFirmware/Marlin/files/3169489/config_marlin20_21042019.zip)

I thank you all for the quick response on this topic !

p3p, 2019-05-11:
I downloaded Marlin, copied in the supplied files and the built without problem using `pio run` from the terminal, can you remove any platformio cache directories such as `.platformio` in your home directory and `.pio*` in the Marlin directory.

gloomyandy, 2019-05-11:
I think you really need to test with a more recent version of Marlin if you can. Quite a bit has changed since your version.

DavidThijs, 2019-05-11:
Ok, will make a new project with the most recent Marlin BF2.0 build. I'll report back :)

DavidThijs, 2019-05-11:
! I just needed to download the latest Marlin BF2.0, unzipped it in a new folder and then copied my configs and pin file over + edit of platformio.ini to set LPC1768 and ...

Tadaa... it compiled successfully

Still puzzled why, but I can test further :+1:

closed 2019-05-11 by DavidThijs