MarlinFirmware/Marlin #13983: [BUG] BLTouch deploys and stows before homing X and/or Y

martinroger, 2019-05-11:
### Description

I am not 100% sure this is a bug, as there is some logic to the behaviour. After recently upgrading to one of the latest May commits of the 2.x branch, my BLTouch (on Z) deploys and stows systematically before any homing move, even if it just a X or Y move.

This presents an issue with my print sequence, where at the end of a print I home X and Y and then let things cool down. In such cases, the BL touch deploys, hits an error by touching the print, stays deployed and gets dragged across the print, potentially catching/breaking its tip

### Steps to Reproduce

1. G28 X or G28 Y or G28 X0 Y0

Expected behavior: 
Non-bltouch linked axis is being homed while BLTouch stays stowed with no deploy of the probe.

Actual behavior:
Probe deploys, and stows if there is no error, and then homes X and/or Y. If the probe touches a print or the bed, it gets potentially dragged in deployed state.

Current version of my Marlin 2.0 for this printer is the default branch (vKing) at : https://github.com/martinroger/Marlin

InsanityAutomation, 2019-05-12:
New revision probes arrive Monday. After some review, this will be adjusted with any changes needed there. @FanDjango has an open PR with this change if you want it now.

martinroger, 2019-05-12:
Thanks, that clarifies it for me. For the time being I can patiently wait... And raise my Z before homing at the end of a print. It's not like I am printing at max Z often!

FanDjango, 2019-05-12:
> In such cases, the BL touch deploys, hits an error by touching the print, stays deployed and gets dragged across the print, potentially catching/breaking its tip

Ooof. Not good.

@InsanityAutomation You are right in saying that the PR I've got open will fix that by itself. **BUT**: In addition, it would be more logical to move the BLTouch initialization downwards in the code to the place where Z home actually will take place - it shouldn't happen at all if someone is only homing X and/or Y (takes 2x500ms for nothing). I will analyze further and put in another commit for that.

FanDjango, 2019-05-12:
@martinroger Ok, I have reconfigured and can reproduce this - and it can also happen to anyone that does a home and the nozzle is not high enough.

martinroger, 2019-05-12:
I guess I can thank the bavarian rain today ! Thanks for the great work.

martinroger, 2019-05-15:
Issue seems resolved with today's commit. Issue closed.

closed 2019-05-15 by martinroger