MarlinFirmware/Marlin #13984: [BUG] Compiling buffer overflow warnings bugfix 2 - should show as error

doggyfan, 2019-05-12:
When compiling the current Marlin bugfix 2 (12th May 2019) using Atom & platformio for the SKR 1.3 board (LPC1768), I get a couple of compile warnings (buffer overflows) ..

I'm surprised writing data past a buffers limits only shows up as a warning and not an error. 

Writing data past a buffers limit really is not good, you'll no doubt be corrupting some other functions data or whatever.

==================

In file included from c:\projects\3d printer\marlin 2\mine\marlin\src\hal\shared\marduino.h:33,
from Marlin\src\feature\../sd/../inc/../HAL/HAL_LPC1768/HAL.h:40,
from Marlin\src\feature\../sd/../inc/MarlinConfig.h:30,
from Marlin\src\feature\../sd/cardreader.h:24,
from Marlin\src\feature\power_loss_recovery.h:28,
from Marlin\src\feature\power_loss_recovery.cpp:31:
Marlin\src\feature\power_loss_recovery.cpp: In static member function 'static void PrintJobRecovery::resume()':
Marlin\src\feature\power_loss_recovery.cpp:406:23: warning: '%s' directive writing up to 104 bytes into a region of size 46 [-Wformat-overflow=]
char *fn = info.sd_filename;
~
sprintf_P(cmd, PSTR("M23 %s"), fn);
C:\users\####\.platformio\packages\framework-arduino-lpc176x\cores\arduino/Arduino.h:56:18: note: in definition of macro 'PSTR'
#define PSTR(v) (v)
^
Marlin\src\feature\power_loss_recovery.cpp:406:12: note: 'sprintf' output between 5 and 109 bytes into a destination of size 50
sprintf_P(cmd, PSTR("M23 %s"), fn);

==================

Marlin\src\sd\cardreader.cpp: In static member function 'static void CardReader::presort()':
Marlin\src\sd\cardreader.cpp:765:72: warning: array subscript 27 is above array bounds of 'char [27]' [-Warray-bounds]
sortnames[I][SORTED_LONGNAME_MAXLEN] = '\0'; }while(0)
~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^
Marlin\src\sd\cardreader.cpp:942:11: note: in expansion of macro 'SET_SORTNAME'
SET_SORTNAME(0);
^~~~~~~~~~~~
Marlin\src\sd\cardreader.cpp:765:72: warning: array subscript 27 is above array bounds of 'char [27]' [-Warray-bounds]
sortnames[I][SORTED_LONGNAME_MAXLEN] = '\0'; }while(0)
~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^
Marlin\src\sd\cardreader.cpp:852:13: note: in expansion of macro 'SET_SORTNAME'
SET_SORTNAME(i);
^~~~~~~~~~~~

==================

boelle, 2019-07-21:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

sjasonsmith, 2019-09-21:
The configs are too old to build easily against modern Marlin. I enabled POWER_LOSS_RECOVERY on my own SKR config and I don't see the warnings mentioned in the description.

Considering the original reporter is a ghost, this should probably be closed.

closed 2020-06-20 by boelle