MarlinFirmware/Marlin #13987: [FR] Any way to disable cooldown when stopping print ?

WADZoQUADZ, 2019-05-12:
In earlier marlin versions when hitting stop print it used to remain hot. Now it 1.1.9Bf it automatically cools down. Is there a feature or can there be to disable it from cooling down?

Sometimes we have it preheated then click the wrong file then stop it or sometimes we are adjusting the first layer height and stop it . Now everytime we have to wait for it to preheat all over again.

reloxx13, 2019-05-12:
this is supposed, safe and correct. 

if u stop a print, the printer thinks its done.

stop selecting the wrong files.

WADZoQUADZ, 2019-05-16:
i think your mistaken . It has always been this way in past marlin versions.

alexzkter, 2019-06-23:
I agree. 99% of the time I cancel a print (due to poor adhesion on the first layer) so that I can retry moments laters. At least the printer now homes XY...because having the nozzle sit on top of the print was rather annoying. Now I still have to go to Temperature and preheat menus.

boelle, 2019-07-21:
is this a feature request or bug?

InsanityAutomation, 2019-10-15:
@boelle it was a FR and its done.

closed 2019-10-15 by boelle