MarlinFirmware/Marlin #13989: [BUG] [2.0] EXTRA_LIN_ADVANCE_K: 'bitRead' was not declared in this scope

LastDragon-ru, 2019-05-12:
### Description

```
Marlin\src\gcode\feature\advance\M900.cpp: In static member function 'static void GcodeSuite::M900()':
Marlin\src\gcode\feature\advance\M900.cpp:58:21: error: 'bitRead' was not declared in this scope
bool ext_slot = bitRead(lin_adv_slot, tmp_extruder);
^~~~~~~
Marlin\src\gcode\feature\advance\M900.cpp:58:21: note: suggested alternative: 'spiRead'
bool ext_slot = bitRead(lin_adv_slot, tmp_extruder);
^~~~~~~
spiRead
Marlin\src\gcode\feature\advance\M900.cpp:64:9: error: 'bitWrite' was not declared in this scope
bitWrite(lin_adv_slot, tmp_extruder, slot);
^~~~~~~~
Marlin\src\gcode\feature\advance\M900.cpp:64:9: note: suggested alternative: 'finite'
bitWrite(lin_adv_slot, tmp_extruder, slot);
^~~~~~~~
finite
Compiling .pioenvs\LPC1768\src\src\gcode\feature\fwretract\G10_G11.cpp.o
*** [.pioenvs\LPC1768\src\src\gcode\feature\advance\M900.cpp.o] Error 1
```

### Steps to Reproduce

1. Uncomment `#define EXTRA_LIN_ADVANCE_K` 
2. Compile

#### Additional Information

MKS SBase + Marlin 2 (https://github.com/MarlinFirmware/Marlin/commit/940ff8e7c8a5919681dcd6d6e87668381010151a)

FanDjango, 2019-05-12:
I confirm. Same here. 
If `EXTRA_LIN_ADVANCE_K` is enabled, the code uses `bitRead` and `bitWrite`, which is part of the `platformio-pkg-framework-arduinosam`. Not present/referenced in the LPC1768 env.

SJ-Innovation, 2019-05-12:
Oop! On it!

closed 2019-05-15 by LastDragon-ru