MarlinFirmware/Marlin #13990: [Stupid (?) question or BUG] Sensorless homing sensitivity settings

Patag, 2019-05-12:
Sorry in advance if you consider this as a “support” question but I don’t feel it is.
I'm trying to set-up a printer SKRv1.3-based with sensorless homing feature, using TMC5160 on X and Y axis.
As far as I've understood, the menu “Configuration/Advanced settings/TMC drivers/Sensorless homing” is intended to set Stallguard sensitivity, thus homing bump sensitivity. Right ?
But X and Y values are always set to 0 after initilisation (Power on or Reset) while M122 command shows default values defined in Marlin. Moreover, every change made from LCD panel are correctly taken into account, still confirmed by M122. Did I miss something in configuration process ?
In the end, sensorless homing in working fine (only tested on X only yet) with values defined in config_adv.h, but it’s very confusing behavior.
Thanks for your brilliant opinions about that.
Michel

teemuatlut, 2019-05-12:
The boot process should initiate the LCD variables accordingly but it's possible it isn't working correctly with TMC5160 (or others). It's also possible that since TMC5160 support is still so new, not all bugs have been fixed.
You can try saving EEPROM with M500 and see if the LCD is properly initiated after that. It's possible the issue only presents itself the first time you've enabled TMC5160 in config.

thisiskeithb, 2019-05-13:
I am also running an SKR 1.3 with genuine 5160 Watterott drivers in SPI/sensorless homing mode on an Ender-3 with the bugfix-2.0.x commit (940ff8e) from yesterday and the a613bca commit from 5/10/12.

If I revert back to the 15357af commit from 5/3/19, the values are populated in the TMC drivers menu. 

Sensorless homing works fine and running `M122` , returns the correct values from EEPROM for all three of these commits. Running `M500` does not change anything.

doggyfan, 2019-05-13:
If this is connected to the TMC2208 problem then that problem was caused by changed to Marlin in this commit .. 

https://github.com/MarlinFirmware/Marlin/tree/15357af67ceb74b14606eba9fbb75d20914f8909

I see they changed the main boot-up order of things in there @teemuatlut if that means anything ?

The TMC2208 problem has been present ever since.

thisiskeithb, 2019-05-13:
This should be fixed with PR https://github.com/MarlinFirmware/Marlin/pull/14008 thanks to @marcio-ao.

Patag, 2019-05-14:
Fix in progress in #14008. I close

closed 2019-05-14 by Patag