MarlinFirmware/Marlin #13994: [BUG] PT1000 - SKR V1.3 M2 not working with table 1047 - Heater failure

RoyBerntsenDesign, 2019-05-12:
### Description

### Steps to Reproduce

1. [First Step]
2. [Second Step]
3. [and so on...]

**Expected behavior:** [What you expect to happen]

**Actual behavior:** [What actually happens]

#### Additional Information

* Include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

InsanityAutomation, 2019-05-13:
I was using 147 with a pt100 unamplified earlier today. To confirm, you're directly connected with no amplifier right?

RoyBerntsenDesign, 2019-05-13:
I am using a PT1000 not a PT100 - They need no amplifier

Best regards/Vennlig hilsen
Roy Berntsen
M.Sc. Design Engineer
Tel: +47 928 26 012
E-mail: roy@pro3d.no
Web: www.pro3d.no

gloomyandy, 2019-05-13:
Please provide your configuration files. It is also not at all clear (to me at least) what problem you are seeing. Are you getting an error when you boot (if so please provide a screenshot or the wording)? Are you getting an incorrect temperature reading (if so please supply the actual reading and what you expect to see)?

RoyBerntsenDesign, 2019-05-13:
On bootup the screen shows heater failure - I dont remember what failure it
was as I took it off before picture. There was never any temp reading of
the thermistor on the screen just straight to heater error

Config attached

Best regards/Vennlig hilsen
Roy Berntsen
M.Sc. Design Engineer
Tel: +47 928 26 012
E-mail: roy@pro3d.no
Web: www.pro3d.no

gloomyandy, 2019-05-13:
You can't attach your files via email, you must upload them using the github web interface.

How recent a version of Marlin are you using? Is it the standard version from the Marlin github repo or are you using the version from Bigtreetech. Please test using a very recent version from the standard download.

InsanityAutomation, 2019-05-13:
Can also try adding #define BOGUS_TEMPERATURE_FAILSAFE_OVERRIDE to your configuration.h to see if it starts up correctly.

RoyBerntsenDesign, 2019-05-13:
It is the standard Marlin branch from yesterday 2019/05/12 - No bigtrree
branch

Best regards/Vennlig hilsen
Roy Berntsen
M.Sc. Design Engineer
Tel: +47 928 26 012
E-mail: roy@pro3d.no
Web: www.pro3d.no

RoyBerntsenDesign, 2019-05-13:
I added a zip file

InsanityAutomation, 2019-05-16:
Any update in trying with BOGUS_TEMPERATURE_FAILSAFE_OVERRIDE

RoyBerntsenDesign, 2019-05-18:
> Any update in trying with BOGUS_TEMPERATURE_FAILSAFE_OVERRIDE

So I made some work on this today

First Here is the heater failure in the picture. Then I added bogus temp override and now it shows room temp. I will try to heat it later

[image]

[image]

boelle, 2019-07-21:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

closed 2019-07-21 by boelle

jcabrerazuniga, 2020-06-14:
I am also having a similar problem with my stock CR-10s. I set the thermistor value to 1047, set the maxtemp to 300 but after booting, my screen shows me 300 and if I try to increase the temperature, the hotend starts warming up but the displayed temperature goes down.  I added #define BOGUS_TEMPERATURE_FAILSAFE_OVERRIDE to my config.h file but nothing works. It might be my motherboard but I just want to be sure. Any help will be appreciated.

thinkyhead, 2020-06-15:
@jcabrerazuniga — This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- MarlinFW.org [Marlin Documentation](http://marlinfw.org)
- RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
- [Tom's 3D Forums](https://discuss.toms3d.org/)
- Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
- Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- Marlin Discord server. Join link: https://discord.gg/n5NJ59y

After seeking help from the community, if the consensus points to a bug in Marlin, then you should post a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose).