MarlinFirmware/Marlin #13999: MKS Sbase + optical laser filament sensor[FR] (feature request title)

Studna3D, 2019-05-12:
Hi, 
Does anyone know configuration for MKS Sbase with this sensor:
[https://www.aliexpress.com/item/Prusa-i3-Laser-Sensor-3D-Printer-Filament-Sensor-v0-2e-Detect-Stuck-Filament-Offer-User/32906489907.html]

shitcreek, 2019-05-14:
Please provide your config files. 

Are you trying to use SWITCHING_EXTRUDER with a servo?

Studna3D, 2019-05-14:
[config.zip](https://github.com/MarlinFirmware/Marlin/files/3179680/config.zip)

shitcreek, 2019-05-14:
it looks like you have both SWITCHING_EXTRUDER and SWITCHING_NOZZLE defined in your config file when only one should be.

Studna3D, 2019-05-16:
I use both. This is picture of my extruder. 
[image]

shitcreek, 2019-05-16:
You cannot use both as it is currently written in the codes.

For a one stepper motor (extruder) with two nozzles and a servo setup, what you want is SWITCHING_NOZZLE
http://marlinfw.org/docs/configuration/configuration.html#switching-nozzle

shitcreek, 2019-05-16:
You cannot use both as it is currently written in the codes.

Studna3D, 2019-05-16:
When I set only #define SWITCHING_EXTRUDER, its the same problem with "NUM_RUNOUT_SENSORS cannot exceed the number of E steppers."

shitcreek, 2019-05-16:
You mean your servo? Is it plugged into the correct pins? Have you tried M280?

Studna3D, 2019-05-19:
When I set only #define SWITCHING_NOZZLE, the servo switches the second nozzle, but the stepper motor does not move. Motor move only one direction when set first nozzle. That's why I have both #define SWITCHING_EXTRUDER and #define SWITCHING_NOZZLE.

shitcreek, 2019-05-20:
These two features currently do not work in combination. This would likely need to be a Feature Request. 
You'd need to modify the codes to get it to work.
If you look inside servo.h, you'd see SWITCHING_NOZZLE and SWITCHING_EXTRUDER are separate if cases:
`#if HAS_SERVO_ANGLES

  #if ENABLED(SWITCHING_EXTRUDER)
    #ifndef SWITCHING_EXTRUDER_E23_SERVO_NR
      #define SWITCHING_EXTRUDER_E23_SERVO_NR -1
    #endif
    #if EXTRUDERS > 3
      #define REQ_ANGLES 4
    #else
      #define REQ_ANGLES 2
    #endif
    #define SADATA    SWITCHING_EXTRUDER_SERVO_ANGLES
    #define ASRC(N,E) (SWITCHING_EXTRUDER_SERVO_NR == N ? asrc[E] : SWITCHING_EXTRUDER_E23_SERVO_NR == N ? asrc[E+2] : 0)
  #elif ENABLED(SWITCHING_NOZZLE)
    #define SADATA    SWITCHING_NOZZLE_SERVO_ANGLES
    #define ASRC(N,E) (SWITCHING_NOZZLE_SERVO_NR == N ? asrc[E] : 0)
  #elif defined(Z_PROBE_SERVO_NR)
    #define ASRC(N,E) (Z_PROBE_SERVO_NR == N ? asrc[E] : 0)
    #if ENABLED(BLTOUCH)
      #include "../feature/bltouch.h"
    #endif
    #ifdef BLTOUCH_ANGLES
      #define SADATA  BLTOUCH_ANGLES
    #elif defined(Z_SERVO_ANGLES)
      #define SADATA  Z_SERVO_ANGLES
    #else
      #error "Servo angles are needed!"
    #endif
  #endif`

boelle, 2019-07-21:
is this a bug or feature request?

Studna3D, 2019-08-07:
Feature request