MarlinFirmware/Marlin #14001: [FR] (feature request title)

eikeime, 2019-05-13:
### Description

master and slave style dual controller board support.

Some controller board like lpc1768 based doesn't have much custom io for us to tinker with, some does, but sometimes it still is not enough.
So is it possible we can put the critical functions like temperature sensors, xyze motons,endstops on the master controller board, and put none timing sensitive functions like led light, rgb light, extruder cooling fan, controller cooling fan on a slave controller board, which is controlled by master board send gcode through serial port or spi.
During printing master board could just pass through all gcode to the slave board and they decide how to handle them by them self.

#### Additional Information

* Provide pictures or links that demonstrate a similar feature or concept.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

robbycandra, 2019-05-18:
With PLC1768,,, you can expand the board to 12 stepper. But you have modify the hardware.

alexxy, 2019-05-20:
Well. To make controller distributed (e.g. many comunicating boards) we need some bus to connect all boards toogether. Good candidate for such bus is CAN bus.