MarlinFirmware/Marlin #14002: [BUG] bugfix-2.0.x- g29 la tete ne se déplace pas parellèle au lit

francobusc, 2019-05-13:
bonjour, 

mon problème est sur une delta diy...

G28 fonctionne , je vais bien en home ok

G33 fonctionne aussi ok

G29 ne fonctionne pas correctement .... pas bon

la tête descend touche le bed ok...puis la tête se dirige vers le 2ème point et au lieu de se déplacer parallèle au bed et bien la tête monte en se déplaçant de 20mm plus ou moins puis descend pour sonder mais est trop haut

voir vidéo : 

g29 pas bon

https://mega.nz/#!HzoEjQLL!rm3g37M9xF8tXARcU3mOqXrF7KsfAVkH66dczQzf4SI

G33 est bon

https://mega.nz/#!ayRlhCTQ!oRwFuG2UxqqOFuXkH5VCDIVTTipCK-Q_NM7mvn_-5xo

j ai tester ici

AUTO_BED_LEVELING_BILINEAR

et

AUTO_BED_LEVELING_UBL

et j ai le même souci

merci pour votre aide

[marlin.zip](https://github.com/MarlinFirmware/Marlin/files/3171436/marlin.zip)

francobusc, 2019-05-13:
j ai trouvé mon souci....je suis parti sur un fichier existant tevo little monster et as de souci...je ne sais pas pourquoi....
peut etre un mauvais fichier

reloxx13, 2019-05-14:
english

Bob-the-Kuhn, 2019-05-14:
Please close this issue if your problem has been solved.

---

Veuillez fermer ce problème si votre problème a été résolu.

closed 2019-05-14 by francobusc