MarlinFirmware/Marlin #14003: [BUG] (TMC2208 hybrid_threshold not updating board)

SISLANGER, 2019-05-13:
### Description
TMC2208 hybrid_threshold not updating on board always shows 0 with TMC2208 UART on SKR V1.3 

### Steps to Reproduce

SEE DESC

**Expected behavior:** 
Show updated values or any values other than 0

**Actual behavior:** [What actually happens]
only shows set values of 0

doggyfan, 2019-05-13:
Could you maybe post the output from an **M122** command ?  ..  assuming you've enabled **TMC_DEBUG**

Maybe also zip and post your two config files ?

teemuatlut, 2019-05-13:
See if your drivers can communicate first
http://marlinfw.org/docs/hardware/tmc_drivers.html#troubleshooting

SISLANGER, 2019-05-13:
--
M122
X	Y	Z	E	E1
Enabled		false	false	false	false	false
Set current	800	800	800	800	800
RMS current	795	795	795	795	795
MAX current	1121	1121	1121	1121	1121
Run current	25/31	25/31	25/31	25/31	25/31
Hold current	12/31	12/31	12/31	12/31	12/31
CS actual		12/31	12/31	12/31	12/31	12/31
PWM scale	14	14	14	14	14
vsense		1=.18	1=.18	1=.18	1=.18	1=.18
stealthChop	true	true	true	true	true
msteps		16	16	16	16	16
tstep		max	max	max	max	max
pwm
threshold		78	78	66	64	164
[mm/s]		100.25	100.04	30.21	30.00	50.29
OT prewarn	false	false	false	false	false
OT prewarn has
been triggered	false	false	false	false	false
off time		3	3	3	3	3
blank time	24	24	24	24	24
hysteresis
-end		-1	-1	-1	-1	-1
-start		1	1	1	1	1
Stallguard thrs
DRVSTATUS	X	Y	Z	E	E1
stst		X	X	X	X	X
olb
ola
s2gb
s2ga
otpw
ot
157C
150C
143C
120C
s2vsa
s2vsb
Driver registers:
X	0xC0:0C:00:00
Y	0xC0:0C:00:00
Z	0xC0:0C:00:00
E	0xC0:0C:00:00
E1	0xC0:0C:00:00
 
 
Testing X connection... OK
Testing Y connection... OK
Testing Z connection... OK
Testing E connection... OK
Testing E1 connection... OK

doggyfan, 2019-05-13:
Looks OK.

When you say the threshold value shows as '0', do you mean on an LCD display ? .. if so, what LCD display do you have ?

SISLANGER, 2019-05-13:
[Archive.zip](https://github.com/MarlinFirmware/Marlin/files/3172808/Archive.zip)

@doggyfan 
@teemuatlut

sreichholf, 2019-05-13:
I can confirm this issue on an Ender 3 with CR10_STOCKDISPLAY.
All hybrid threshold values show "0" on LCD

SISLANGER, 2019-05-13:
@doggyfan yes all other values show fine, except the hybrid_threshold which I can hear and see the changes when I change it live during a print.

SISLANGER, 2019-05-13:
I get the following on compelling

Marlin/src/feature/tmc_util.cpp:146:28: warning: 'TMC_driver_data get_driver_data(TMC2208Stepper&)' defined but not used [-Wunused-function]
static TMC_driver_data get_driver_data(TMC2208Stepper &st) {
^~~~~~~~~~~~~~~
Marlin/src/feature/tmc_util.cpp:143:23: warning: 'uint32_t get_pwm_scale(TMC2208Stepper&)' defined but not used [-Wunused-function]
static uint32_t get_pwm_scale(TMC2208Stepper &st) { return st.pwm_scale_sum(); }
^~~~~~~~~~~~~

SISLANGER, 2019-05-13:
> I can confirm this issue on an Ender 3 with SKR 1.3 and CR10_STOCKDISPLAY.
> All hybrid threshold values show "0" on LCD.
> 
> I have left //STEALTHCHOP_E disabled, though I do have 4x2208 as it doesn't work anyways.

OK so its not just board specific I know I have seen them in the settings to start with so maybe a bug, maybe a conflict of settings

doggyfan, 2019-05-13:
I don't get that compile warning, but yes I too can confirm something has gone wrong with the Marlin code at some point. I too am getting '0' with my four TMC2208 thresholds on the LCD with the SKR 1.3 board.

M122 looks OK too though.

00: 		X	Y	Z	Z2
Enabled		false	false	false	false
Set current	1000	1000	1000	1000
RMS current	994	994	994	994
MAX current	1402	1402	1402	1402
Run current	17/31	17/31	17/31	17/31
Hold current	8/31	8/31	8/31	8/31
CS actual		8/31	8/31	8/31	8/31
PWM scale	10	10	10	10
vsense		0=.325	0=.325	0=.325	0=.325
stealthChop	true	true	true	true
msteps		16	16	16	16
tstep		max	max	max	max
pwm
threshold		29	29	38	38
[mm/s]		136.31	136.31	13.00	13.00
OT prewarn	false	false	false	false
off time		4	4	4	4
blank time	24	24	24	24
hysteresis
-end		2	2	2	2
-start		1	1	1	1
Stallguard thrs				
DRVSTATUS	X	Y	Z	Z2
stst		X	X	X	X
olb					
ola					
s2gb					
s2ga					
otpw					
ot					
157C					
150C					
143C					
120C					
s2vsa					
s2vsb					
Driver registers:
		X	0xC0:08:00:00
		Y	0xC0:08:00:00
		Z	0xC0:08:00:00
		Z2	0xC0:08:00:00

Testing X connection... OK
Testing Y connection... OK
Testing Z connection... OK
Testing Z2 connection... OK
ok

gloomyandy, 2019-05-13:
Please test with a very recent version of Marlin. There have been changes to menu code in the last day or so that impact the way that some values are displayed and edited.

SISLANGER, 2019-05-13:
@gloomyandy it was a recent fresh bugfix-2.0.x from a week or so ago

doggyfan, 2019-05-13:
@gloomyandy just compiled with the latest code on github and still no change. All thresholds are showing '0' in the LCD menu.

Patag, 2019-05-13:
Any relation with this #13990 ?

gloomyandy, 2019-05-13:
@SISLANGER a week or so ago is way too old, this change was like 2 days ago I think. But it looks like there is still a problem with the latest code. 

@doggyfan if you change the threshold using the LCD (so change the value from zero), does that new value show up in an M122?

SISLANGER, 2019-05-13:
@gloomyandy I had a look at the changes and wasn't related to the board or issue

doggyfan, 2019-05-13:
> @doggyfan if you change the threshold using the LCD (so change the value from zero), does that new value show up in an M122?

Yes it does, if I edit the the threshold on the LCD menu it shows up in the M122 response. So it's editing it, just not displaying it correctly.

Strangley, the LCD menu setting is remember if I exit the TMC menu and go back into it.

gloomyandy, 2019-05-13:
@doggyfan and @Patag it looks very much like the same issue that is causing  #13990

doggyfan, 2019-05-13:
yep, okey doke @gloomyandy 

I'd download previous code commits and see exactly where the problem occured if only I could get on with github.

I have windows 7, git and tortoisegit installed. How do I do download previous commits I wonder ?

SISLANGER, 2019-05-13:
@doggyfan If possible could you let me know the fix or solution when you get a chance. Thank you

doggyfan, 2019-05-13:
Yes I will do that for you @SISLANGER with pleasure. I just need to find out how to download previous Marlin commits from github first.

SISLANGER, 2019-05-13:
Thank you @doggyfan 
Is this it https://github.com/MarlinFirmware/Marlin/tree/15357af67ceb74b14606eba9fbb75d20914f8909

doggyfan, 2019-05-13:
No, need to go back further than that as that version has the problem too.

How did you select that tree ? .. I need to be able to do this myself.

SISLANGER, 2019-05-13:
Sorry I'm just getting used to GitHubI found a list under commits 

https://github.com/MarlinFirmware/Marlin/compare/bugfix-2.0.x

doggyfan, 2019-05-13:
OK, I'm getting there slowly.

This problem was created between the 2nd May and the 4th May. I'm just homing in on the exact commit that has caused this.

SISLANGER, 2019-05-13:
Cool

doggyfan, 2019-05-13:
Yes sorry @SISLANGER, it was that one that's caused this problem.

https://github.com/MarlinFirmware/Marlin/tree/15357af67ceb74b14606eba9fbb75d20914f8909

The previous commit is fine.

SISLANGER, 2019-05-13:
No worries @doggyfan have fun ;-)

doggyfan, 2019-05-13:
I'm not yet familiar enough with Marlin to fix this. Whoever made those changes needs to be brought in to fix it I think.

SISLANGER, 2019-05-13:
@doggyfan Thank you for trying

@thinkyhead 
@marcio-ao 

Any possibility you can help on this bug / issue

teemuatlut, 2019-05-13:
Good job tracking down the commit. Helps a ton!

At a quick glance I see `ui.init` was moved to before `settings.load`.
`ui.init` is responsible for setting up the LCD along withe TMC LCD section.
`settings.load` calls the function that initiates the TMC drivers and their configuration.

Revert that and see if that helps any.

SISLANGER, 2019-05-13:
@teemuatlut sorry on what file

teemuatlut, 2019-05-13:
Marlin.cpp on `setup()`

doggyfan, 2019-05-13:
Yes, reverting that ui boot-up position does indeed fix the problem @teemuatlut.

But I suspect we can't just move it back without concequences to other parts of the code. Seems the ui needs to be done first.

Really, the very first thing that should be setup is the ui/LCD, because it's the only feedback for the user when things go wrong at boot-up.

If it were me, I'd have the entire boot-up sequence displayed on the LCD as it goes through it all. That way the user gets to see where bad problems occur.

marcio-ao, 2019-05-13:
It looks like there is conflicting requirements that need to sorted through. When I moved "ui.init()" before "settings.load()", I did my best to check whether it would introduce any problems, but I must have failed to see this one.

Anyway, the reason I made the change is that our implementation of the ExtUI is storing settings in the EEPROM related to things like brightness, touch register configuration, etc. The following sequence of things happen:

1. - `ui.init()` is called, which calls the ExtUI to initializes the display hardware to the point in which it can begin accepting commands.
2. - `settings.load` is now called, which reads data from the EEPROM and calls the ExtUI with EEPROM data pertaining to the UI.
3. - The ExtUI takes this data and loads it into the registers on the graphics chip.

The reason for the re-org is that if 2 and 3 happens before 1, then the graphics chip isn't initialized so it cannot receive the settings.

It sounds like the TMC code is expecting things to happen in another order. I'll read through this thread and see if I can understand what is going on.

marcio-ao, 2019-05-13:
I guess in going through the comments it isn't clear to me why the TMC code would require the EEPROM to be read before the UI is initialized. @teemuatlut: Could you perhaps clarify?

sreichholf, 2019-05-13:
I just did a quick "flyover" but in init_tmc_section()  (which is at the en dof MarlinUI::init) init_lcd_variables() is called for each tmc stepper.
That in return requires the settings to be initialized: as it sets the initial values from   
`planner.settings.axis_steps_per_mm[spmm_id]`

marcio-ao, 2019-05-13:
I suspect the issue is with the function `init_tmc_section()` which is called at the end of `ui.init()`. It looks like it shadows the TMC state into separate state variables in RAM. Before the re-ordering, that shadowing happens after the EEPROM had been loaded and after the initial state had been loaded into the TMC drivers. After the re-ordering, it happens before the EEPROM has been loaded, so it shadows the default values rather than EEPROM values.

The fix to this appears to be simple, the call to `init_tmc_section()` could be placed at the end of `_load()` in `configuration_store.cpp`.

marcio-ao, 2019-05-13:
@sreichholf: Yes, looks like you identified the problem a split second before I did :)

marcio-ao, 2019-05-13:
In fact, the ExtUI already implements a method for doing stuff after the EEPROM has been loaded. We could put `init_tmc_section` in the same location to make the relationship clear:

```
  bool MarlinSettings::load() {
    if (validate()) {
      const bool success = _load();
      #if ENABLED(EXTENSIBLE_UI)
        ExtUI::onConfigurationStoreRead(success);
      #endif
      #if HAS_TRINAMIC && HAS_LCD_MENU
        init_tmc_section();
      #endif
      return success;
    }
    reset();
    #if ENABLED(EEPROM_AUTO_INIT)
      (void)save();
      SERIAL_ECHO_MSG("EEPROM Initialized");
    #endif
    return true;
  }
```

marcio-ao, 2019-05-13:
Actually, it probably needs to be done in case of failure as well:

```
  bool MarlinSettings::load() {
    if (validate()) {
      const bool success = _load();
      #if ENABLED(EXTENSIBLE_UI)
        ExtUI::onConfigurationStoreRead(success);
      #endif
      #if HAS_TRINAMIC && HAS_LCD_MENU
        init_tmc_section();
      #endif
      return success;
    }
    reset();
    #if ENABLED(EEPROM_AUTO_INIT)
      (void)save();
      SERIAL_ECHO_MSG("EEPROM Initialized");
    #endif
    #if HAS_TRINAMIC && HAS_LCD_MENU
        init_tmc_section();
    #endif
    return true;
  }
```

marcio-ao, 2019-05-13:
Okay, third revision, here is what I am leaning towards:

In "configuration_store.cpp", replace "bool MarlinSetting::load()" with:

```
  bool MarlinSettings::load() {
    bool success = validate();
    if (success)
      success = _load();
    else {
      reset();
      #if ENABLED(EEPROM_AUTO_INIT)
        (void)save();
        SERIAL_ECHO_MSG("EEPROM Initialized");
      #endif
    }
    #if ENABLED(EXTENSIBLE_UI)
      ExtUI::onConfigurationStoreRead(success);
    #endif
    #if HAS_TRINAMIC && HAS_LCD_MENU
      init_tmc_section();
    #endif
    return success;
  }
```

In "ultralcd.cpp", remove the following from `void MarlinUI::init()`:

```
  #if HAS_TRINAMIC && HAS_LCD_MENU
    init_tmc_section();
  #endif
```

teemuatlut, 2019-05-13:
Perhaps the whole TMC init configuration should be moved to its own section.
The overall order should be something along lines of 

1. Init UART/SPI
2. Configure static driver parameters
3. Configure variables that are EEPROM supported
4. Init TMC LCD <==> Copy some of the values into shadow variables

marcio-ao, 2019-05-13:
My only concern with my above suggestion is that now it means `init_tmc_section` will be called at startup, but also anytime the user runs `M501`. Presumably this would need to happen anyway, to update the shadow variables, but I don't know how this was handled before.

teemuatlut, 2019-05-13:
I never really liked having the initialization tied to `M50x` calls. Restoring the config doesn't need to send all the configuration parameters to the driver, just the ones that can be modified with gcodes.
The steps 2, 3 and 4 also need to happen with `Power::power_on` because the settings don't "stick" if there is no VMOT power.

marcio-ao, 2019-05-13:
@teemuatlut: Just a thought, rather than keeping shadow variables and having to deal with keeping their state correct, couldn't you just copy them on entry to TMC menu? The advantage of that is that the correct settings will always used in the LCD interface, regardless of whether some other code changed elsewhere.

This is the approach I take in my ExtUI. I keep no copies of the machine state, except when the enters a screen where there that setting needs to be changed.

teemuatlut, 2019-05-13:
Sure, that would be ideal as I believe that would save a bit of SRAM, but I really dislike dealing with the monster that is the LCD section of Marlin and the macros needed references to variables. I didn't think of using local variables or just didn't spend the time to figure it out.

SISLANGER, 2019-05-13:
@marcio-ao is that the current fix

marcio-ao, 2019-05-13:
> I never really liked having the initialization tied to M50x calls. Restoring the config doesn't need to send all the configuration parameters to the driver, just the ones that can be modified with gcodes.

I agree. In a way, it seems like the TMC drivers have the same issues that I ran into with the graphics chip. They are some external bits of hardware with their own settings registers that need to be copied from the EEPROM. I suggest that "main.cpp" do the following:

- As early as possible, initialize the TMC drivers with default values.
- Call `ui.init()`
- Call `settings.load();`
- From within `Configuration_store.cpp`, load whatever values are read from EEPROM into TMC chips as well as the shadow variables.

> EDIT: Would that also mean the variables are read from the drivers on each LCD update cycle when the menu is selected?

Ideally these values would be copied only when transitioning into and out of the menus, although the UltraLCD code doesn't make it easy to do this, since it lacks entry and exit routines. In my ExtUI, I explicitly have "onEntry" and "onExit" events where I shuffle things in and out of shadow variables.

marcio-ao, 2019-05-13:
How about keeping things simple. For now, we could change `Marlin.cpp' to have the following init sequence:

```
  // UI must be initialized before EEPROM
  // (because EEPROM code calls the UI).
  ui.init();
  ui.reset_status();

  ...

  // Load data from EEPROM if available (or use defaults)
  // This also updates variables in the planner, elsewhere
  (void)settings.load();

  #if HAS_TRINAMIC && HAS_LCD_MENU
    init_tmc_section();
  #endif
```

I feel like this restores things to as closely as they were to prior to commit 15357af67ceb74b14606eba9fbb75d20914f8909 without introducing any more corner cases to test. Thoughts?

teemuatlut, 2019-05-13:
I'm fine with patching things for now and making a proper better thought-out solution later.

marcio-ao, 2019-05-13:
I'll work on a PR, since I broke it :)

tpruvot, 2019-05-13:
oh, about that...
ui.reset_status() is supposed to show "Ready" on the LCD

settings.load() later can display some data on lcd and erase the Ready status

SISLANGER, 2019-05-13:
// Load data from EEPROM if available (or use defaults)
  // This also updates variables in the planner, elsewhere
  (void)settings.load();

  #if HAS_TRINAMIC && HAS_LCD_MENU
    init_tmc_section();
  #endif

causing identifier "SENSITIVE_PINS" is undefined

marcio-ao, 2019-05-13:
Can you include the entire error message?

SISLANGER, 2019-05-13:
#define SENSITIVE_PINS { _X_PINS _Y_PINS _Z_PINS _X2_PINS _Y2_PINS _Z2_PINS _Z3_PINS _Z_PROBE _E0_PINS _E1_PINS _E2_PINS _E3_PINS _E4_PINS _E5_PINS _BED_PINS _H0_PINS _H1_PINS _H2_PINS _H3_PINS _H4_PINS _H5_PINS _PS_ON _HEATER_BED _FAN0 _FAN1 _FAN2 _FANC HAL_SENSITIVE_PINS }
Expands to:

{ _X_PINS _Y_PINS _Z_PINS _X2_PINS _Y2_PINS _Z2_PINS _Z3_PINS _Z_PROBE _E0_PINS _E1_PINS _E2_PINS _E3_PINS _E4_PINS _E5_PINS _BED_PINS _H0_PINS _H1_PINS _H2_PINS _H3_PINS _H4_PINS _H5_PINS _PS_ON _HEATER_BED _FAN0 _FAN1 _FAN2 _FANC HAL_SENSITIVE_PINS }
identifier "SENSITIVE_PINS" is undefined

marcio-ao, 2019-05-13:
@SISLANGER: That does not seem related to the Trinamic change we have been discussing. Did anything else change? Or did you pull new code?

SISLANGER, 2019-05-13:
it comes up as soon as I paste in 

#if HAS_TRINAMIC && HAS_LCD_MENU
init_tmc_section();
#endif

gloomyandy, 2019-05-13:
@SISLANGER please make sure you are using the latest version of Marlin so that everyone is in the same state.

marcio-ao, 2019-05-13:
@SISLANGER: Let's see if the Travis build process catches this in the PR. I'm not seeing it in our local builds.

[50 more replies]

closed 2019-05-28 by SISLANGER