MarlinFirmware/Marlin #14004: [BUG] SKR 1.3 USB Connectivity with Octoprint not working

jjacobsfs, 2019-05-13:
Description

When I connect my printer through USB to Octoprint, no temperature is displayed and after a minute or so I will get a timeout.  If I am lucky, the SKR will accept a command or two before it loses comms with Octoprint.

Steps to Reproduce
•Connect the printer to Octoprint with USB cable
•Open Octoprint,, no temperature is displayed and barely any gcode is scrolling in the terminal

Expected behavior: [What you expect to happen]
 be able to control the printer through USB

Actual behavior: [What actually happens]
 -> Not working

Additional Information

My Configuration.h file :

#define SERIAL_PORT 0
#define SERIAL_PORT_2 -1
#define BAUDRATE 250000

When I connect OctoPrint to my other printers, it works fine so I can rule out the RPi, Octoprint, and USB cable. 

Thanks for your help

gloomyandy, 2019-05-13:
Are you using the very latest version of Marlin from the Marlin github or are you using the version from Bigtreetech (or some other repository). Please retest with the latest version of Marlin.  Several people have reported this issue when using the Bigtreetech version. 

Also please do as requested by the issue template and include a zip file of your configuration files.

jjacobsfs, 2019-05-13:
I am using the BigTreeTech version.  I will try the latest Marlin tonight and report back.

Thanks

Patag, 2019-05-13:
@jjacobsfs : I'm quite sure you have to set baud rate to 115200 on SKR V1.3.

jjansen85, 2019-05-13:
My setup:
```
#define SERIAL_PORT 0
#define SERIAL_PORT_2 -1
#define BAUDRATE 115200
```

doggyfan, 2019-05-13:
The baudrate has no effect at all with CDC USB comms (SERIAL_PORT).
It does however affect real life com-port comms (such as SERIAL_PORT2).

jjacobsfs, 2019-05-13:
I tried several different rates and all results were the same.

jjansen85, 2019-05-13:
I just realise that I had something similar with my SKR 1.3 connected to my macbook pro. When trying to communicate via Repetier-Host, it did not work as expected. No temperature, timeout etc, sometimes it did connect.. Later on I figured out that this was because MacOS mounted the on-board SD-card. After unmounting the SD-card in MacOS I was able to connect.
Btw, comms between SKR and RPi with OctoPrint never had issues...
Maybe this can be of help, maybe not ;-)

gloomyandy, 2019-05-13:
@jjansen85 there should be no problem with having the SD card mounted (it happens all of the time with Windows), if you have a problem with that (with a recent version of Marlin), please create a new bug report and supply all of the details.

Patag, 2019-05-13:
I had some communication problems with SKR V1.3 until...I changed USB cable !

doggyfan, 2019-05-14:
When using the USB to the SKR, you need to make sure the DTR line is set (simulated in CDC USB I know) otherwise the SKR will not respond.
I'm using OctoPrint in windows and it doesn't have that option (not that I can find) but it works all the same, but I don't know what serial comms options there are in a linux (raspberry) setup.

Just curious. Why do you specify SERIAL_PORT as the hardware port (UART) on the SKR and SERIAL_PORT_2 as the USB port ? .. it doesn't matter I know.

jjacobsfs, 2019-05-14:
> Just curious. Why do you specify SERIAL_PORT as the hardware port (UART) on the SKR and SERIAL_PORT_2 as the USB port ?
I just used the settings Bigtreetech had.  Are you saying that is incorrect?

I used the latest Marlin 2.0 and its working fine now.  Not sure what Bigtreetech did to their firmware.  For anyone else with this issue, you cannot use the Bigtreetech config files directly as they are a bit different than Marlins.  I just used theirs for reference.

gloomyandy, 2019-05-14:
It depends what you are trying to do. At the moment you have two serial ports configured. One runs over the built in USB stack (port -1) and the other runs over the hardware UART (port 0) which comes out to the AUX connection on the board. It sounds like you are using the USB serial port for Octopriint. Do you have anything else connected to the AUX port (like a TFT display)? If not you may want to set things up like below so only the USB port is active...
```C
#define SERIAL_PORT -1
//#define SERIAL_PORT_2 0
```

jjacobsfs, 2019-05-14:
I do not have the TFT.  I am using the onboard USB port.  I was considering hooking the Pi up to the UART instead of the USB.  Not sure if it would be any more reliable even though I have never had any issues with using USB.

gloomyandy, 2019-05-14:
I'm in the process of setting up something similar. It will almost certainly be less reliable using the UART, but I want to be able to talk to my PC via the USB connection. If you are using a Pi then remember that you may need to make changes to the way Linux is configured to be able to use the rPi UART.

jjacobsfs, 2019-05-14:
Do you have an SKR 1.3?

gloomyandy, 2019-05-14:
Yep and a V1.1.

jjacobsfs, 2019-05-14:
A bit off topic, do you use a sd card extension cable?  I had all kinds of issues with that.

gloomyandy, 2019-05-14:
No I don't, but I'm not surprised you have problems. I did a lot of work on the LPC176x USB stack and when testing that I tried various setups including providing USB access to the SD card reader attached to the LCD display. But they would not work reliably using the relatively high speed that the framework/USB/SD code uses to access the SD card over SPI.

Alissonverd, 2019-05-15:
https://github.com/bigtreetech/BIGTREETECH-SKR-V1.3
https://www.dropbox.com/s/ppjfflhf3j5yzh2/MarlinV2.0%20SKRV1.1%20instruction.docx?dl=0

> #define SERIAL_PORT -1
> #define SERIAL_PORT_2 0
> #define BAUDRATE 115200

gloomyandy, 2019-05-15:
@Alissonverd I'm sorry I don't understand your post. Do you have a problem?

@jjacobsfs if your problem is now resolved, please close this issue. Folks will still be able to find it if they do a search.

jmdearras, 2019-05-16:
-1 is the USB, it should be SERIAL_PORT.

jimmyeao, 2019-07-06:
Have some strange behaviour with my SKR1.3 - Octoprint will only connect if I set the port and Baud to Auto. It then detects Baud as 115200, but if I then try and connect by specifying the BAUD as 115200 it just sits there and refuses to connect. MArlin is configured:
#define SERIAL_PORT -1
#define SERIAL_PORT_2 0
#define BAUDRATE 115200

jmdearras, 2019-07-10:
baud is meaningless on the USB (-1) on the SKR v1.x

jimmyeao, 2019-07-10:
That doesn’t answer the question though

Jimmy White

gloomyandy, 2019-07-10:
@jimmyeao I doubt if your question has anything to do with the original issue. This issue has already drifted all over the place and should ideally be closed. Please start a fresh one and post your configuration files, details of your printer setup (are you actually using SERIAL_PORT_2 for anything?), how recent a version of Marlin are you using and full details of your octoprint setup. Having said that you may be better off asking this question on the Octoprint forum as it seems more related to how Octoprint decides to connect than directly to Marlin.

boelle, 2019-07-21:
as @gloomyandy advices i will close this one, OP found the issue in that he could not directly copy over config files

closed 2019-07-21 by boelle

Snowist, 2019-08-30:
Set config as following

#define SERIAL_PORT 0
#define SERIAL_PORT_2 -1

And connect your USB cable to Pi and SKR 1.3 and reboot the Pi. Pi will reconfigure the port ttyACM0 and will work.

ipatch, 2019-10-20:
> Set config as following
> 
> #define SERIAL_PORT 0
> #define SERIAL_PORT_2 -1
> 
> And connect your USB cable to Pi and SKR 1.3 and reboot the Pi. Pi will reconfigure the port ttyACM0 and will work.

tried with these settings using the latest bugfix branch of marlin, 
- built the FW using platformio, 
- uploaded the freshly built bin to sd card on the skr
- connect the usb cable to one of the 4 usb ports on my pi
- connect the other end of the cable to skr
- reboot skr using built reset button on board
- open web interface for octoprint try to connect and get the following error messages

```
onnecting to: /dev/ttyACM0
Changing monitoring state from "Offline" to "Opening serial port"
Connected to: Serial<id=0x6b20b1f0, open=True>(port='/dev/ttyACM0', baudrate=115200, bytesize=8, parity='N', stopbits=1, timeout=10.0, xonxoff=False, rtscts=False, dsrdtr=False), starting monitor
Changing monitoring state from "Opening serial port" to "Connecting"
There was a timeout while trying to connect to the printer
Changing monitoring state from "Connecting" to "Offline"
Connection closed, closing down monitor
```

**~/.octoprint/logs/serial.log**

```
2019-10-20 13:11:44,981 - Connecting to: /dev/ttyACM0
2019-10-20 13:11:44,999 - Changing monitoring state from "Offline" to "Opening serial port"
2019-10-20 13:11:45,004 - Connected to: Serial<id=0x6b20b1f0, open=True>(port='/dev/ttyACM0', baudrate=115200, bytesize=8, parity='N', stopbits=1, timeout=10.0, xonxoff=False, rtscts=False, dsrdtr=False), starting monitor
2019-10-20 13:11:45,004 - Changing monitoring state from "Opening serial port" to "Connecting"
2019-10-20 13:12:15,043 - There was a timeout while trying to connect to the printer
2019-10-20 13:12:15,054 - Changing monitoring state from "Connecting" to "Offline"
2019-10-20 13:12:15,067 - Connection closed, closing down monitor
```

I've tried connecting octoprint via both the USB cable and the GPIO dupont pins on the header, and neither can connect to the SKR via octoprint **FWR**

All that said, when I do connect the skr 1.3 directly to my computer using the same usb cable I am able to interface with the board SD card _in_ or _out_ and I can successfully run a `m115` cmd from prontoface _printrun_ on macOS.

ipatch, 2019-10-20:
Can now confirm that this issue with connecting via the USB cable to a Raspberry Pi using Octoprint was Octoprint related and not marlin.

Leaving this here for future me,

If running into connection issues with Octoprint and an SKR v1.3 logic board, try toggling the below settings within octoprint,

[image]

---

Uncheck all the boxes for firmware related settings until a proper connection can be confirmed.

[image]

SirRedZ, 2019-11-06:
Thanks for posting this. I have been struggling to customize my (printers) to co-operate and I have aborted/interrupter numerous prints by auto-detecting usb port screwing with another printer on the same Rpi3B+.

My goal is to have (2) Ender 3 printers running off 1 Pi; SKR v1.3 in "A" and stock mb. in "B" Eventually "B" will get a SKR 1.3 also. I followed Chris Rileys tutorials on multiple Printers, 1 Pi and it works. The challenge is with using AUTO-ON plugins to power the Printer on when a slicer spools the gcode to octoprint, which works very well on the stock MB. With the SKR, the board is not powered and even though OctoPrint gets the Gcode with M80 command, it serial times out before actually turning on the printer... Im a little lost and don't know where to go from here. I know the issue is serial-port related between octoprint and printer response.

[-=SrZ=-]