MarlinFirmware/Marlin #14009: [BUG] Contrast not working on MKS12864 on SKRV1.3

jmdearras, 2019-05-13:
### Description

Contrast adjustment stopped working on MKS12864

### Steps to Reproduce

1. Adjust Contrast with LCD.  Note it does not change.
2. Press knob to leave, notice the contrast does change instantaneously.
3. Reset CPU and notice contrast changes to the stored value until the main screen.

**Expected behavior:** [What you expect to happen]

**Actual behavior:** [What actually happens]

#### Additional Information
This was working on the April 28 version, but failed when updating to 5/13
There also seem to be issues with eeprom and default settings for TMC2130.  I need to look further

* Include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

shitcreek, 2019-05-14:
regarding TMC2130, there's this issue: 14003

jmdearras, 2019-05-16:
This was my error, when i pulled the latest libraries, contrast worked as expected

closed 2019-05-16 by jmdearras