MarlinFirmware/Marlin #14010: [BUG] TMC2130 code defaults are not restored as default, SKRV1.3

jmdearras, 2019-05-13:
### Description

X and Y_STALL_SENSITIVITY default values in code are not restored.

### Steps to Reproduce

1. Define non-zero STALL_SENSITIVITY values, compile and load.
2. Start and clear eeprom.
3. restore defaults.
4. see that X,Y_STALL_SENSITIVITY are not your entered values, but are 0
5. setting from LCD, and storing works fine.
[Configuration_adv.zip](https://github.com/MarlinFirmware/Marlin/files/3175189/Configuration_adv.zip)
[Configuration_adv.zip](https://github.com/MarlinFirmware/Marlin/files/3175192/Configuration_adv.zip)

**Expected behavior:** [What you expect to happen]

**Actual behavior:** [What actually happens]

#### Additional Information

* Include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

gloomyandy, 2019-05-14:
see https://github.com/MarlinFirmware/Marlin/issues/14003

jmdearras, 2019-05-16:
Latest pull (5/16) all work except storing sensorless endstop values.  Always restore as 0

boelle, 2019-07-21:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

closed 2019-07-21 by boelle