MarlinFirmware/Marlin #14014: Error compiling  MArlin 2.0 on board BOARD_RURAMPS4D_11  with REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER

ozzysv, 2019-05-14:
Hello
I download the latest archive and in the file  configuration.h  I change only two lines
#defaine MOTHERBOARD BOARD_RURAMPS4D_11
uncoment 
#define  REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER
and compilation fails

Compiler Arduino 1.8.9, u8glib installed

**What could be the error?**

Arduino: 1.8.9 (Windows 10), Плата:"Arduino Due (Programming Port)"

```
In file included from sketch\src\HAL\HAL_DUE\u8g_com_HAL_DUE_st7920_sw_spi.cpp:62:0:

sketch\src\HAL\HAL_DUE\u8g_com_HAL_DUE_sw_spi_shared.h:27:26: error: variable or field 'u8g_SetPIOutput_DUE' declared void

 void u8g_SetPIOutput_DUE(u8g_t *u8g, uint8_t pin_index);

                          ^

sketch\src\HAL\HAL_DUE\u8g_com_HAL_DUE_sw_spi_shared.h:27:26: error: 'u8g_t' was not declared in this scope

sketch\src\HAL\HAL_DUE\u8g_com_HAL_DUE_sw_spi_shared.h:27:33: error: 'u8g' was not declared in this scope

 void u8g_SetPIOutput_DUE(u8g_t *u8g, uint8_t pin_index);

                                 ^

sketch\src\HAL\HAL_DUE\u8g_com_HAL_DUE_sw_spi_shared.h:27:46: error: expected primary-expression before 'pin_index'

 void u8g_SetPIOutput_DUE(u8g_t *u8g, uint8_t pin_index);

                .......

exit status 1
```

Bob-the-Kuhn, 2019-05-14:
I believe PR #14005 will fix this issue.

ozzysv, 2019-05-14:
> I believe PR #14005 will fix this issue.

thank

in folder  ...**Marlin\src\HAL\HAL_DUE**  
in files 
u8g_com_HAL_DUE_st7920_sw_spi.cpp
u8g_com_HAL_DUE_sw_spi.cpp
u8g_com_HAL_DUE_sw_spi_shared.cpp

I changed the sequence of definitions so that  
`#include "u8g_com_HAL_DUE_sw_spi_shared.h"`
 was located behind
` #include <U8glib.h`

And everything works

closed 2019-05-14 by ozzysv