MarlinFirmware/Marlin #14016: LCD doesn't work perfectly

D3bugg3RR, 2019-05-14:
Hello everyone, i installed bootloader in the tronxy x5s mainboard (atmega 1284P) and after that, the latest release of marlin firmware, i copied  all tronxy x5s files in the main directory and i uploaded the sketch, all works but the lcd is "disturbed"

Tronxy x5s have only configuration.h file (missing configuration_adv.h)

i used sanguino board and 1284P (16 Mhz)
[image]
[image]

closed 2019-05-14 by D3bugg3RR

jefersonbentesdacosta, 2019-10-18:
How did you solve your problem?

InsanityAutomation, 2019-10-18:
@jefersonbentesdacosta it seems ardunio ide has been causing lcd issues, compiling with VS Code seems to fix it for most

jefersonbentesdacosta, 2019-10-18:
I just added this lines to configuration.h

#define ST7920_DELAY_1 DELAY_NS(250)
#define ST7920_DELAY_2 DELAY_NS(250)
#define ST7920_DELAY_3 DELAY_NS(250)