MarlinFirmware/Marlin #14017: G0/G1 Commands send Y to max position

gmg23, 2019-05-14:
I am having a problem running 1.1.9 and 1.1.X bug fix. Rig in question is Tevo Tornado, stock everything (other than authentic bltouch). All axis work perfectly during homing. I can move the bed from the LCD. Bed moves operates perfectly during bed leveling (bltouch pre-print). Been pulling my hair out all day. Basically an G0 or G1 command cause the Y axis to go to max position (300). After that Y is completely unrecoverable unless homed. This causes prints to literally just print in a single line on the back of the bed. Y ends up at ~280 after bed leveling and as soon as the print tries to send the bed back to center, it goes to 300 and locks up. Terminal shows the print working normally. The LCD reports the Y position as if the bed is moving appropriately. I am at a total loss. Would really appreciate any suggestions. 

Let me also add that this printer was running previously. Once I updated to 1.1.9, it all went sideways. I do know that 1.1.9 is generally solid, as I have it running on other printers. After this problem popped up, I grabbed a config.h from another machine that is running well and modified only the necessary settings - same problem.

Thanks in advance.

Not sure how you can reproduce, as this seems to be an uncommon issue. 

I would expect G1 Y0 would send Y to the end stop after homing

What actually happens is Y moves to 300/max position. (this happens with any linear move sent to Y)

Clearly since all homing and LCD control is working, the printer is capable of making the required moves. 

I will dig into it deeper when I have some time but would love any suggestions.

Here is copy of Configuration.h
[Configuration.txt](https://github.com/MarlinFirmware/Marlin/files/3177958/Configuration.txt)

reloxx13, 2019-05-14:
what terminal do u use to enter the g0/g1? octoprints?
tried printing from sd card?

if octoprint, check printer settings inside octoprint settings for bed size/center origin.

if move axis and sd print works, i would think sth is wrong with ur host (baudrate is high, bad cable).

gmg23, 2019-05-14:
Hi, thanks for the quick response.

Yes, I am using OctoPrint. However I to share this suspicion, and went down
this road of testing yesterday. I’m experiencing the same problem when
connecting to the USB through rep-host.

I also tried slicing a calibration cube simplify 3-D and printing that
directly from the SD card, and I experience the same issue.

I was getting ready to start enabling additional debug information, when  I
discovered the printer doesn’t think that anything is wrong. It’s not
providing any errors, and it’s indicating that the Y axis is doing
everything that it should.

Very odd one huh?

FanDjango, 2019-05-14:
For temporary testing try setting your X AND Y max bed size to (say) 200.

See if same happens or goes to 300 still.

Also swap X and Y steppers and see if problem swaps.

Stuff like that maybe?

gmg23, 2019-05-14:
I like the suggestion about the bed size. I would expect to see if stop at 200, but we will see. I will try that when I get home from work. 

Yes I kept keying in on the stepper drivers, however I also kept coming back to the fact that the bed was able to function properly during homing and bed leveling. So that left me thinking about a possible overheating driver and possibly stepper, however I have ruled that out. Also since this was printing normally up until new firmware, pretty sure I'm not looking at a hardware issue. However I do agree it's worth testing, and when I get home I will try to swap steppers, then drivers.

Thank you for the suggestion. Much appreciated.

gmg23, 2019-05-14:
You think the Arduino 1.8.9 compiler may have any relevance here? I did just recently update the IDE before loading this firmware. 

Among the other suggestions, I am going to recompile on 1.8.8 and see what happens

FanDjango, 2019-05-14:
No, 1.8.9 is ok. And 1.8.8 is not it either.

FanDjango, 2019-05-14:
LCD AUTO HOME and then MOVE AXIS Y movement to Y 150 goes into the middle right, and display shows Y 150?

gmg23, 2019-05-14:
yes, when I move it via LCD it works and tracks correctly. This got me to thinking that maybe this specific to absolute positioning... after confirm LCD worked normally, I then connected via terminal and switched to relative positioning mode and jogged Y 10mm... same thing. Responds to the first command by extending to 300, locks up, Y axis wont accept any movement commands until homed.

FanDjango, 2019-05-14:
> Very odd one huh?

One way to find out if the movement to the limit on Y is infinite or just to (say) 500 is to (for testing) reducing the steps:

#define DEFAULT_AXIS_STEPS_PER_UNIT   { 8, 8, 399.2901, 400 }

(making the universe smaller)

Just grasping at straws here

FanDjango, 2019-05-14:
Doesn't happen for G1 X150?

FanDjango, 2019-05-14:
Also try
#define DEFAULT_AXIS_STEPS_PER_UNIT   { 80, 80, 400, 400 }

Long time ago there was some limit parsing this, but I am not sure at all. And if so it would be also for X

gmg23, 2019-05-14:
No. All commands to X, both relative and absolute, work every time as expected.

gmg23, 2019-05-14:
Additional suggestion noted. I will add that to list of things to try when I get home.

FanDjango, 2019-05-14:
One more idea:
You compiled. How close to the storage limit are you?

gmg23, 2019-05-14:
Dynamic memory 64%
Prog Mem 62%
MKS Gen L (2560)

FanDjango, 2019-05-14:
2560, ok no danger

FanDjango, 2019-05-14:
After homing, try some other commands, like M48
Also did you try G1 X150 Y150 or just G1 Y150? Can you go through all possible permutations and try to see if combining or adding X,Z, etc. or the order causes the command?

FanDjango, 2019-05-14:
And then bugfix-2.0.x on the 2560, I recommend definitely

gmg23, 2019-05-14:
I started digging through the code to see the difference between how the LCD manipulates the axis vs a gcode command from terminal, but I was frustrated and falling asleep at that point lol

FanDjango, 2019-05-14:
I'll leave you to it for now

gmg23, 2019-05-14:
I have tried all G1 and G0 permutations.  starting with G1 X150 Y150 Z15 F2000, and pairing down from there.

gmg23, 2019-05-14:
thanks for all the suggestions. this seems like a very odd one off problem. I will try these things and let you know. Really appreciate everyones time - and contributions to Marlin in general!!!

gmg23, 2019-05-15:
So quick update here... I have tried changing the default steps to 80,80,400,400, no luck. 

Interestingly when I decreased the bed size to 250, Y continued to go to 300. However when I lowered it to 100, the Y axis had the same action, but it stops at 100. This time around the LCD shows Y at 100, and it continues to sit there regardless of how many commands are being sent to move X and Z around. 

I didn't even bother trying to print anything. I got sick of fighting an uphill battle with what is clearly a deep seeded gremlin. I loaded Marlin 2.0.X and everything works great. It seems someone else had a similar problem a few days ago in bug report #13645.  

I really would like to figure out whats happening, but I have to put it on the back burner at the moment as I have already smoked up a couple days troubleshooting this.

reloxx13, 2019-05-15:
did u reset eeprom with M502 ?

gmg23, 2019-05-15:
yes I run 502/500 after flashing. 

Marlin 2.0 makes it through 75% of bed leveling and errors out due to bl-touch. I will deal with that later I guess. 

Back to 1.1.9 - and making headway - seems that babystepping is the root cause of this bug. With baby stepping disabled entirely, the printer is once again working normally. Symptom was only present on the Y axis, so perhaps there is something about it causing it to malfunction. Going to dig a bit deeper, but need to get this rig printing again first. 

Thanks

gmg23, 2019-05-15:
Just wanted to add that I have re-enabled #DEFINE_BABYSTEPPING , but left #DEFINE_BABYSTEP_XY commented out. The printer is still working normally with these settings. It would seem something about the XY baby stepping is cause of this particular issue.

boelle, 2019-07-21:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

chasbecht, 2019-09-23:
Earlier today I encountered this issue, but with travel to the home position as in #13645 and was able to work around it by disabling software endstops.  I'm Commenting here because that bug is closed and this one is open.  I'm using a pretty odd cobbled-together version of Marlin at the moment, so I don't know if it's the same issue.

boelle, 2019-10-12:
@gmg23 still having issues?

boelle, 2019-10-21:
i cant confirm this one, i tried 

Send: G0 X0
Recv: ok
[...]
Send: G0 X125
Recv: ok
[...]
Send: G0 Y0
Recv: ok
[...]
Send: G0 Y105
Recv: ok
[...]

so everything is working as expected

given the lack of activity i will close this one, we can always reopen when its confirmed that tehre is an issue

closed 2019-10-21 by boelle

Cristaux, 2019-11-19:
Hello everyone, I am in a similar case:
Mks Gen L, TB6600, Nema 17, Double Axis Y & Z.
everything looks OK except on Y axis -1 impossible and Position Y = 0.
Ex:
G0 Y100 -> oK
G0 Y10 -> remains at position Y100 and indicates Y = 0
Thank you in advance....

[Configuration.txt](https://github.com/MarlinFirmware/Marlin/files/3866191/Configuration.txt)
[Configuration_adv.txt](https://github.com/MarlinFirmware/Marlin/files/3866192/Configuration_adv.txt)
[pins_RAMPS.txt](https://github.com/MarlinFirmware/Marlin/files/3866193/pins_RAMPS.txt)
But Files Below