MarlinFirmware/Marlin #14019: [BUG] LPC1768 Compilation Error - platformio.ini problem

Rigid3D, 2019-05-14:
### Description

Compilation throws error about Neopixel library. Marlin is configured for LPC1768 board and the Neopixel features are not defined.

The problem is solved when the libdep record of the neopixel library (neopixel github link) is commented on the platformio.ini

#### Additional Information

The error message

.piolibdeps\Adafruit NeoPixel\Adafruit_NeoPixel.cpp: In member function 'void Adafruit_NeoPixel::show()':

.piolibdeps\Adafruit NeoPixel\Adafruit_NeoPixel.cpp:153:3: error: 'noInterrupts' was not declared in this scope

noInterrupts(); // Need 100% focus on instruction timing

^~~~~~~~~~~~

.piolibdeps\Adafruit NeoPixel\Adafruit_NeoPixel.cpp:153:3: note: suggested alternative: 'detachInterrupt'

noInterrupts(); // Need 100% focus on instruction timing

^~~~~~~~~~~~

detachInterrupt

.piolibdeps\Adafruit NeoPixel\Adafruit_NeoPixel.cpp:2037:3: error: 'interrupts' was not declared in this scope

interrupts();

^~~~~~~~~~

.piolibdeps\Adafruit NeoPixel\Adafruit_NeoPixel.cpp:2037:3: note: suggested alternative: 'intptr_t'

interrupts();

^~~~~~~~~~

intptr_t

Compiling .pioenvs\LPC1768\CMSIS\driver\lpc17xx_exti.c.o

*** [.pioenvs\LPC1768\lib91e\Adafruit NeoPixel\Adafruit_NeoPixel.cpp.o] Error 1

 [ERROR] Took 66.10 seconds

Bob-the-Kuhn, 2019-05-14:
I was not able to reproduce the problem.

I grabbed the latest 2.0.x, deleted my **.piolibdeps** directory and did a build with a LCD that has Neopixel LEDS in it.  There were no problems

Please do the above steps and see if the problem goes away.

Deleting  **.piolibdeps** results in PlatformIO getting the latest version of the libraries and only the needed libraries.  Chances are that your old Neopixel library was out of date or there was a second incompatible one there.

  Bob

Rigid3D, 2019-05-14:
Actually neopixel leds are not defined in my configuration.h and configuration_adv.h.

I have retried it and as you have suggested. Ran PIO Clean. Deleted .piolibdeps folder from the project folder. Ran Auto Build. Again the same error.

Let's wait a couple of days and If noone else confirms the issue, we can close it. Maybe it is related with my setup.

Below is the content of .piolibdeps after autobuild.

PS C:\Users\Legion\Desktop\Marlin-bugfix-2.0.x\.piolibdeps> dir

    Directory: C:\Users\Legion\Desktop\Marlin-bugfix-2.0.x\.piolibdeps

Mode                LastWriteTime         Length Name
----                -------------         ------ ----
d-----       15.05.2019     01:04                Adafruit NeoPixel
d-----       15.05.2019     01:04                LiquidCrystal_ID136
d-----       15.05.2019     01:04                Servo_ID883
d-----       15.05.2019     01:04                TMCStepper_ID5513
d-----       15.05.2019     01:04                U8glib-HAL_ID1932

p3p, 2019-05-14:
You need to update your PlatformIO frameworks, NeoPixel support was added a while ago.

Terminal: `platformio update`
Visual Studio Code: `Alien head button ->  updates -> update all`
Atom: PlatformIO menu -> `Update platforms, packages and libraries`

Rigid3D, 2019-05-15:
@p3p thank you. Updating did not solve the issue but deleting all the platforms and libraries did.

closed 2019-05-15 by Rigid3D