MarlinFirmware/Marlin #14020: [BUG] 2.0 "bugfix" unplug / short both(hotend,bed) thermistor only lock board not send ERR..MSG serial

PC-maniak, 2019-05-14:
### Description

Hi I am trying Marlin 2.0 on RE-ARM I would say everything works as it should except MIN MAX temperature hotend and bed .. if i unplug the thermistor .. board only locks and it needs to be restarted but the serial does not write back the error nothing .. just freezes it it happens both for hotend (0) and for bed .. Thermal runaway, Heating failed, it works as it should

### Steps to Reproduce

disconnecting the thermistor plug may not make sense to describe =)

**Expected behavior:** serial port message error

**Actual behavior:** nothing

#### Additional Information

excuse my English

doggyfan, 2019-05-15:
Hi @PC-maniak 

mmmm the lock-up is not user friendly at all is it, it's not really the best thing for a firmware to be doing.

Are you using the most recent Marlin I wonder, or maybe a version from a week or so ago ?

PC-maniak, 2019-05-15:
it's version +- 4days back .. downloaded directly from marlin website

but the "nothing" is better when is locked even without an error vote than it burns :D

shitcreek, 2019-07-06:
Any update on this?

PC-maniak, 2019-07-06:
hello i don't know yet i haven't had time to check if anything has changed since last update

gloomyandy, 2019-07-06:
When you say it locks up, do you have a display? Is any message shown on the display? How are you connected to the Re-Arm board, via USB? It would help if you post your configuration files.

boelle, 2019-09-24:
@PC-maniak is the issue still there? can you attach your 2 config files?

boelle, 2019-10-12:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

closed 2019-10-12 by boelle