MarlinFirmware/Marlin #14022: [BUG] Layer Shift on Tool-Change

P4CM4N76, 2019-05-15:
### Description

Print suffers a layer shift only on tool-change.
This layer shift only ocours in the s axis.

### Steps to Reproduce

1. Slice a dual colour print
2. Print it

**Expected behavior:**

Layer lines aligned.

**Actual behavior:**

In every tool-change the layer suffers a shift in the x axis of about 0.5mm.
This shift only ocours in the x axis and in the negative direction.
I've tried the version of marlin 1.1.9, 1.1.9-bugfix and 2.0.0-bugfix (the last one dated of today (15/05/2019)).
I've tried with and without tool change script in the slicer. With and without prime pillar.
In every version the result is the same.

#### Additional Information

[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/3182565/Configuration.zip)
[image]
[image]
Link to video: https://youtu.be/BAppu4Juvp0

InsanityAutomation, 2019-05-15:
This looks more like an offset calibration issue than a firmware issue. Have you adjusted your offsets?

P4CM4N76, 2019-05-15:
> This looks more like an offset calibration issue than a firmware issue. Have you adjusted your offsets?

This is dual extruder, single nozzle. What offsets is there to adjust?

InsanityAutomation, 2019-05-16:
@P4CM4N76 Sorry, missed that in the config. Most of the single nozzle testing was done with the automatic pause and filament swap functions enabled to eliminate tool change scripts. Can you post what you have as tool change scripts, and if possible try the following from Config Adv and eliminate them : 

#if EXTRUDERS > 1
  // Z raise distance for tool-change, as needed for some extruders
  #define TOOLCHANGE_ZRAISE     2  // (mm)

  // Retract and prime filament on tool-change
  #define TOOLCHANGE_FILAMENT_SWAP
  #if ENABLED(TOOLCHANGE_FILAMENT_SWAP)
    #define TOOLCHANGE_FIL_SWAP_LENGTH         100  // (mm)
    #define TOOLCHANGE_FIL_EXTRA_PRIME           2  // (mm)
    #define TOOLCHANGE_FIL_SWAP_RETRACT_SPEED 3600  // (mm/m)
    #define TOOLCHANGE_FIL_SWAP_PRIME_SPEED   3600  // (mm/m)
  #endif

  /**
   * Position to park head during tool change.
   * Doesn't apply to SWITCHING_TOOLHEAD, DUAL_X_CARRIAGE, or PARKING_EXTRUDER
   */
  #define TOOLCHANGE_PARK
  #if ENABLED(TOOLCHANGE_PARK)
    #define TOOLCHANGE_PARK_XY    { X_MIN_POS + 10, Y_MIN_POS + 10 }
    #define TOOLCHANGE_PARK_XY_FEEDRATE 6000  // (mm/m)
  #endif
#endif

P4CM4N76, 2019-05-16:
My Tool-Change Scrip is:

G91 ; Switch from absolute positioning to relative positioning
G0 E-100 F3600 ; Retract filament by 6.5+100=106.5mm at 60 mm/s (3600 mm/min) on old tool
G90 ; Return to absolute positioning
M211 S0 ; Disable Soft Endstops
G0 Y302 F6000 ; Go to Y=302 at 100 mm/sec
{IF NEWTOOL=1}T1 ; Activate tool 1
{IF NEWTOOL=0}T0 ; Activate tool 0
G91 ; Switch from absolute positioning to relative positioning
G0 E100 F3600 ; Extrude 100mm of filament at 60 mm/s on new tool
G0 E45 F180 ; Extrude 45mm of filament at 3 mm/s on new tool
G0 E-6.5 F3600 ; Retract filament by 6.5 mm at 60 mm/s
G90 ; Return to absolute positioning
G0 X[next_position_y] Y[next_position_y] F6000 ; Move to next position on print at 100 mm/s
M211 S1  ; Enable Soft Endstops
G91 ; Switch from absolute positioning to relative positioning
G0 E6.8 F3600 ; Extrude filament by 6.8 mm at 60 mm/s
G90 ; Return to absolute positioning

In the final Scrip the movement (G0 Y302 F6000 ; Go to Y=302 at 100 mm/sec) will be changed to G0 X320 F6000 for a purge bucket, for now is in the Y-axis for debugging.

I've tested without tool change script and with the automatic pause and filament swap functions enabled.
This is the result (don't mind the stinging). The problem persists.

[image]

InsanityAutomation, 2019-05-16:
Ok, I'll unbury my crx and get a bit of testing in most likely this weekend.

P4CM4N76, 2019-05-29:
I've found out what is causing this.
It seems that is something about skew correction that is making the printer "misbehave".
I've disabled the feature and the result is this.
[image]

boelle, 2019-07-21:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

boelle, 2019-09-24:
@P4CM4N76 is the issue still there?

closed 2019-10-12 by boelle

ZOKY-3D, 2019-11-01:
I suspect that the issue is definitely a bug in Marlin (1.1.9. bugfix) software.
I have a similar situation like P4CM4N76.

The settings are as follows:

-2 extruders and 2 heaters (the second one is used for support)
-mb gt2560
-slicer: simplify 3d

During the tool switching phase of printing there are small shifts over y axis. After that, the model continues being printed normally without shifts (when only 1 tool is used).
[image]

When I try to print the same model with only 1 head (both for the model and support) everything works perfectly (Same acceleration and speed settings).
[image]

I use skew correction only for y/z axis because my other axis dont need correction. **When I turn skew correction off everything works like it should. The issue is that skew correction creates a problem when using 2 heads.**

Being pretty new here on github, was wondering whats the bug report process like? Where can I find a potential future solution?

midopple, 2020-03-23:
I suspect that the issue is definitely a bug in Marlin, I use the Version V2.0.4 and have the Same Problem.

Settings:
Dual X with 2 Printheads
RAMBO Board
Bedlevel UBL

During the tool switching phase of printing there are small shifts over y axis

When i set the Skew faktor M852 S0.002 the i have an small shifts over y axis
When i set Skew M852 S0 i have no shifting

[image]

Thanks vor help

Michael

thinkyhead, 2020-03-24:
Good data, thanks! We'll continue on your new issue.