MarlinFirmware/Marlin #14031: [BUG] error: 'struct job_recovery_info_t' has no member named 'active_hotend'

guestisp, 2019-05-16:
When building with `EXTRUDERS` >= 2 and `SINGLENOZZLE`

I'm not able to test with `SINGLENOZZLE` disabled as I only have a dual extruder single nozzle machine

```
sketch/src/feature/power_loss_recovery.cpp: In static member function 'static void PrintJobRecovery::resume()':
sketch/src/feature/power_loss_recovery.cpp:285:40: error: 'struct job_recovery_info_t' has no member named 'active_hotend'
     sprintf_P(cmd, PSTR("T%i S"), info.active_hotend);
                                        ^
exit status 1
```

boelle, 2019-07-21:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

closed 2019-07-21 by boelle

guestisp, 2019-07-21:
I forgot to close.
I did a pull request, already merged, to fix this

boelle, 2019-07-21:
:-D