MarlinFirmware/Marlin #14034: Wrong Zoffset after start print

D3bugg3RR, 2019-05-16:
I changed start gcode in the slicer a lot of times, but the problem is marlin because old gcode were printed wrong after firmware update.
I executed home in all axis, calibrated the bed,when the printer starts printing, the bed goes down more than necessary, and when the filament is being extruded the bed is still down (more or less 2 cm)

is home position the zoffset is correct.

reloxx13, 2019-05-16:
config? abl probe? z-offset m851? startcode? printer/board? eeprom resetted m502? from sd card?

robbycandra, 2019-05-18:
Use STORE SETTING

D3bugg3RR, 2019-05-18:
(update) since yesterday "set home offset" doesn't work and return a message that say's "err too far", today i flashed a custom marlin and it seems to work, this is my start gcode
G21
G90
M82
M107
G28
G1 Z5 F1200
G1 X10 E6 F5000
G92 E0 ; reset extrusion distance
G1 Z0.3 ; Raise and begin printing.
M117 Printing...

I put G1 Z0.3 to make the distance correct between bed and nozzle because the first layer is 3 mm depth, but with this gcode the nozzle goes up first, and later goes in the print start point, with what code can i move the nozzle from (example) Z=4 to Z=0.3 while it goes in the start point?

[image]

shitcreek, 2019-05-21:
'G1 Z5 F1200' - this moves the nozzle up
Use G1/G0 to move axis. http://marlinfw.org/docs/gcode/G000-G001.html

This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/1.1.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
- [Tom's 3D Forums](https://discuss.toms3d.org/)
- Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
- Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- Marlin Discord server. Join link: https://discord.gg/n5NJ59y

After seeking help from the community, if the consensus points to to a bug in Marlin, then you should post a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose).

closed 2019-07-06 by shitcreek