MarlinFirmware/Marlin #14035: [FR] M117 command)

sandroxGH, 2019-05-16:
Is possible only for the M117 command have the possibility to print the ":" character on screen,
I wold like to display the estimate printi time for example M117 Time 1:01:00

Ludy87, 2019-05-16:
try & Error 😉

doggyfan, 2019-05-17:
Works for me ..

M117 Hi :Good lookin:

PC-maniak, 2019-05-18:
Hello i try and ":" not working ..  i send M117 Layer : 1/69

and i get "Layer " only.. hmmm

tested on latest /// 1.1.x "bugfix" snapshot Patches after 1.1.9 ///

igorepi, 2019-05-22:
M177 with ":" does work when this command is received from UART. Same command doesn't work when printing from SD card, at least in Marlin 1.1.9.

PC-maniak, 2019-05-22:
Aa yes i tested from SD card

InsanityAutomation, 2019-05-24:
Can you check with bugfix-2.0.x? 1.1.x is EOL and wont be getting more updates. The SD parser may be stripping them out as comments, but 2.0 has had about a year of development on top of 1.1.9....

PC-maniak, 2019-05-24:
yes i will try it on weekend on re-arm board
and write the result

igorepi, 2019-05-28:
Just checked with latest bugfix-2.0.x - same behavior as in 1.1.9. Everything after ":" is stripping out by SD parser.

PC-maniak, 2019-05-28:
Yes same here ->blah:bluh i get only "blah"

AnHardt, 2019-05-28:
There is a **half** a special handling of ':' in ttps://github.com/MarlinFirmware/Marlin/blob/bugfix-2.0.x/Marlin/src/gcode/queue.cpp#L741 when read from SD.
The line is ended, but I can't see  special *function* of ':'.
Try to remove and see what happens.

sandroxGH, 2019-06-17:
Can be possible remove from 
https://github.com/MarlinFirmware/Marlin/blob/1.1.x/Marlin/Marlin_main.cpp#L1189
the "sd_char == ':' " to print correctly on the screen

AnHardt, 2019-06-17:
The line seems to belong to a lost/vergotten feature. At least i can neither find documentation nor handling  '#' and ':' anywhere in the code.
Try to comment out the complete line
```
//          || ((sd_char == '#' || sd_char == ':') && !sd_comment_mode)
```
Test that and come back with the results. Do you see any regressions or improvements?

boelle, 2019-07-21:
so is this a bug or feature request?

sandroxGH, 2019-08-02:
Hello I've try to comment the line suggest on Marlin_main.ccp line 1189 "||
((sd_char == '#' || sd_char == ':') && !sd_comment_mode)" it working fine
from two weeks  printing from SD with out any side effect.
Thanks

jimb15122, 2020-01-31:
Hi.  This issue should be reclassified as a bug.
The M117 command cannot display the :  and # characters when the command is read from SD card.  This bug also causes the text that appears after the symbol to be truncated, and be processed as a subsequent command.  (Can you spell attack surface?) The : and # characters are being interpreted as end of line characters.  

I remember seeing a mention somewhere on the web that the : character was used in some gcode generators as a way to pack multiple commands on a single line.  But for the life of me I could not anything on the net mentioning this practice.   Since this  "support" is not mentioned on the official reprap gcode, the special checks for these characters should be removed from queue.cpp.   I do not think these were typos.  They are either intentional (but undocumented) support for multi command packing, or a left over developer debug hack.  This problem exists in both the 1.x and 2.x versions.

thinkyhead, 2020-02-01:
I believe this must be left over from early attempts to do sub-calls. They aren't needed any more, so have been removed.

closed 2020-02-01 by thinkyhead