MarlinFirmware/Marlin #14038: MKS Gen V1.4 - Dual Z setup, Z2/E1 doesn't move

Earluke, 2019-05-16:
### Description

I have a Sunhokey Prusa I3 with a new MKS Gen V1.4 board and a mix of TMC2100 (X,Y,E0) and A4988 (both Z moptors).
 
As i have dual z stepper motors, but only one extruder (E0),  i have to use E1 for the 2nd Z stepper motor.

I did use the Marlin Configurator  to configure the driver types for each motor (X,Y,Z,E0,E1) and activated Z dual stepper drivers in configuration_adv.

### Steps to Reproduce

1. Use Marlin Configurator to create configuration + upload to printer
2. increase Z height manually via repetier host 

**Expected behavior:** both Z stepper motors to turn 
Only one z stepper motor turns (and the other can be turned by hand (no power))

I allready tried 10 different configuration files incl. 1.1.9 and 1.1.x and swapping motors (then the other one works, but never both)

I also made sure that my bed temp probe is activated (see other bug report).

FanDjango, 2019-05-16:
Next thing to check, pin assignments for E1

Earluke, 2019-05-16:
How exactly, what pin should E1 be assigned to?

FanDjango, 2019-05-16:
Ähm. pins_..... file. Check the definitions, is there something there that might explain the failure to activate that stepper

FanDjango, 2019-05-16:
In your configuration.h->you have defined the board. That points you to the pins file

robbycandra, 2019-05-18:
Check if you are using correct MotherBOard at configuration.h

Earluke, 2019-05-20:
Reinstalled Repetier host and now it works perfectly.

THX for your help, BR Norbert

shitcreek, 2019-07-06:
cool beans

closed 2019-07-06 by shitcreek