MarlinFirmware/Marlin #14039: Failed to build code with VIKI2 display for AZTEEG_X5_GT

dmitrygorbenko, 2019-05-16:
### Description

I'm trying to build the source code with platformIO for AZTEEG_X5_GT with VIKI2 display, and I'm getting compilation errors.

### Steps to Reproduce

1. Apply my patch
2. Run: platformio run --environment LPC1769

**Expected behavior:**
To see the line:

> Environment LPC1769                     [SUCCESS]

**Actual behavior:**

I'm getting error:
> Marlin/src/HAL/HAL_LPC1768/u8g/u8g_com_HAL_LPC1768_sw_spi.cpp:70:10: fatal error: gpio.h: No such file or directory

#### Additional Information
OS: Ubuntu 18.04
Marlin branch: 2.0.x, latest

Attaching patch and compilation output:

[VIKI2_error_patch.txt](https://github.com/MarlinFirmware/Marlin/files/3189021/VIKI2_error_patch.txt)

[VIKI2_compilation_error.txt](https://github.com/MarlinFirmware/Marlin/files/3189011/VIKI2_compilation_error.txt)

p3p, 2019-05-16:
Make sure your platformio packages are up to date `platformio update` framework-arduino-lpc176x should be version 0.1.3.

dmitrygorbenko, 2019-05-17:
It helped !
I'm sorry for this case...
Can we add this step into some documents ? To avoid such cases when people forgot up update their tools ?
Or add some tool which detects the environment ?

Anyway, thanks !

closed 2019-05-17 by dmitrygorbenko