MarlinFirmware/Marlin #14045: [BUG] Undefined reference to `Backlash::distance_mm'

Ubermeisters, 2019-05-17:
Marlin 2.0 bugfix

Latest, Clean install of Arduino IDE

When compiling, after enabling backlash compensation, i am met with the following error:

```
C:\Users\Chris\AppData\Local\Temp\ccQyj2hA.ltrans6.ltrans.o: In function `add_correction_steps':

sketch\src\feature/backlash.cpp:79: undefined reference to `Backlash::distance_mm'

sketch\src\feature/backlash.cpp:79: undefined reference to `Backlash::distance_mm'

collect2.exe: error: ld returned 1 exit status

exit status 1
Error compiling for board Arduino/Genuino Mega or Mega 2560.
```
A snip from my configuration_adv.h:

```
#define BACKLASH_COMPENSATION
#if ENABLED(BACKLASH_COMPENSATION)
  // Define values for backlash distance and correction.
  // If BACKLASH_GCODE is enabled these values are the defaults.
  #define BACKLASH_DISTANCE_MM { 0.85, 0.85, 0.001 } // (mm)
  #define BACKLASH_CORRECTION    0.9       // 0.0 = no correction; 1.0 = full correction
```

As you can see, my XY vales are really high (seriously), so i'd like to also enable backlash smoothing. The motors are very weak, and belt tension cant be set as well as necessary for no backlash compensation.

My Marlin DIR:

[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/3193300/Marlin.zip)

thinkyhead, 2019-05-21:
The compiler doesn't like `constexpr` for some reason. So `distance_mm` has now been changed to a `const` instead.

closed 2019-05-21 by thinkyhead

Ubermeisters, 2019-05-21:
Thanks! I'm new to using GitHub, what do I do now to enact the change on my files? I see the commit but I'm unsure which of the 2 code section is new vs. old.

Ubermeisters, 2019-05-22:
@thinkyhead Upon compiling [your  fix](https://github.com/MarlinFirmware/Marlin/tree/cfca07852a8cfa5f15527b19d60093641f3063d4), i am receiving the following error:

```
Arduino: 1.8.9 (Windows 10), Board: "Arduino/Genuino Mega or Mega 2560, ATmega2560 (Mega 2560)"

sketch\src\module\configuration_store.cpp: In static member function 'static bool MarlinSettings::_load()':

sketch\src\module\configuration_store.cpp:1969:57: error: binding 'const float [3]' to reference of type 'float (&)[3]' discards qualifiers

           float (&backlash_distance_mm)[XYZ] = backlash.distance_mm;

                                                         ^

sketch\src\module\configuration_store.cpp:1974:51: error: binding 'const uint8_t {aka const unsigned char}' to reference of type 'uint8_t& {aka unsigned char&}' discards qualifiers

           uint8_t &backlash_correction = backlash.correction;

                                                   ^

exit status 1
Error compiling for board Arduino/Genuino Mega or Mega 2560.

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

```

italocjs, 2019-07-23:
Just a follow-up and close,  I tried burning the firmware (downloaded today) and it failed when using backlash compensation, what fixed was simply putting some small values (0.01 instead o 0, which is the default), i'm not sure why tho, 

Error i had:
`error: binding 'const float [3]' to reference of type 'float (&)[3]' discards qualifiers float (&backlash_distance_mm)[XYZ] = backlash.distance_mm;`

Fixed using:
`//
// Backlash Compensation
// Adds extra movement to axes on direction-changes to account for backlash.
//
#define BACKLASH_COMPENSATION
#if ENABLED(BACKLASH_COMPENSATION)
  // Define values for backlash distance and correction.
  // If BACKLASH_GCODE is enabled these values are the defaults.
  #define BACKLASH_DISTANCE_MM { 0.01, 0.01, 0.01 } // (mm)
  #define BACKLASH_CORRECTION    1.0       // 0.0 = no correction; 1.0 = full correction

  // Set BACKLASH_SMOOTHING_MM to spread backlash correction over multiple segments
  // to reduce print artifacts. (Enabling this is costly in memory and computation!)
  //#define BACKLASH_SMOOTHING_MM 3 // (mm)

  // Add runtime configuration and tuning of backlash values (M425)
  #define BACKLASH_GCODE

  #if ENABLED(BACKLASH_GCODE)
    // Measure the Z backlash when probing (G29) and set with "M425 Z"
    #define MEASURE_BACKLASH_WHEN_PROBING

    #if ENABLED(MEASURE_BACKLASH_WHEN_PROBING)
      // When measuring, the probe will move up to BACKLASH_MEASUREMENT_LIMIT
      // mm away from point of contact in BACKLASH_MEASUREMENT_RESOLUTION
      // increments while checking for the contact to be broken.
      #define BACKLASH_MEASUREMENT_LIMIT       0.5   // (mm)
      #define BACKLASH_MEASUREMENT_RESOLUTION  0.005 // (mm)
      #define BACKLASH_MEASUREMENT_FEEDRATE    Z_PROBE_SPEED_SLOW // (mm/m)
    #endif
  #endif
#endif
`

Ubermeisters, 2019-07-24:
Thank you. This interesting. good to know. i will attempt the same.

commit cfca0785 (2019-05-21): Backlash linker error workaround

Fix #14045