MarlinFirmware/Marlin #14047: [BUG] M43 (pins report/debug) on SKR 1.3 causes firmware reboot

doggyfan, 2019-05-18:
Issuing simple **M43** command causes the SKR 1.3 board to reboot, though it does output a random length of pin reporting text before doing so.

Need to enable pin debug option by having ..

**#define PINS_DEBUGGING**

in your configuration_adv.h file.

doggyfan, 2019-05-18:
Actually, so is an **M122** command (causing reboot).

Bob-the-Kuhn, 2019-05-18:
I haven't figured out why the M43 and M43 I commands sometimes cause a reboot.  

It appears that if you wait until the mapped drive pops up on the PC that the M43 command does not cause a reset.  Is that true for you?

Patag, 2019-05-18:
Same here, whatever the mapped drive status.

doggyfan, 2019-05-18:
<details>

I get the M43 output OK, then about 2 seconds after, it reboots.

<summary>M43 output is OK</summary>

``` 
04:17:36: M43
PIN: 0.00        E1_DIR_PIN                             protected 
.                Z2_DIR_PIN                             protected 
PIN: 0.01        E1_STEP_PIN                            protected 
.                Z2_STEP_PIN                            protected 
PIN: 0.02 (A 7)  <unused/unknown>   Analog in =     0   Input  = 1
PIN: 0.03 (A 6)  <unused/unknown>   Analog in =     0   Input  = 1
PIN: 0.04        <unused/unknown>                       Input  = 1
PIN: 0.05        <unused/unknown>                       Input  = 1
PIN: 0.06        <unused/unknown>                       Output = 1
PIN: 0.07        <unused/unknown>                       Output = 0
PIN: 0.08        <unused/unknown>                       Input  = 1
PIN: 0.09        <unused/unknown>                       Output = 1
PIN: 0.10        E1_ENABLE_PIN                          protected 
.                Z2_ENABLE_PIN                          protected 
PIN: 0.11        E0_DIR_PIN                             protected 
PIN: 0.15        DOGLCD_SCK                             Output = 0
.                SCK_PIN                                Output = 0
PIN: 0.16        LCD_SDSS                               Output = 1
.                SDSS                                   Output = 1
.                SS_PIN                                 Output = 1
PIN: 0.17        MISO_PIN                               Input  = 1
PIN: 0.18        DOGLCD_MOSI                            Output = 1
.                MOSI_PIN                               Output = 1
PIN: 0.19        Y_STEP_PIN                             protected 
PIN: 0.20        Y_DIR_PIN                              protected 
PIN: 0.21        Z_ENABLE_PIN                           protected 
PIN: 0.22        Z_STEP_PIN                             protected 
PIN: 0.23 (A 0)  <unused/unknown>   Analog in =   988   Input  = 0
PIN: 0.24 (A 1)  <unused/unknown>   Analog in =   988   Input  = 0
PIN: 0.25 (A 2)  <unused/unknown>   Analog in =     0   Input  = 1
PIN: 0.26 (A 3)  <unused/unknown>   Analog in =     0   Input  = 1
PIN: 0.27        <unused/unknown>                       Input  = 0
PIN: 0.28        BTN_ENC                                Input  = 1
PIN: 0.29        <unused/unknown>                       Input  = 1
PIN: 0.30        <unused/unknown>                       Input  = 0
PIN: 1.00        <unused/unknown>                       Input  = 1
PIN: 1.01        E1_CS_PIN                              Input  = 1
.                Z2_CS_PIN                              Input  = 1
.                Z2_SERIAL_RX_PIN                       Input  = 1
.                E1_SERIAL_RX_PIN                       Input  = 1
PIN: 1.04        Z2_SERIAL_TX_PIN                       Output = 1
.                E1_SERIAL_TX_PIN                       Output = 1
PIN: 1.08        E0_CS_PIN                              Input  = 1
.                E0_SERIAL_RX_PIN                       Input  = 1
PIN: 1.09        E0_SERIAL_TX_PIN                       Input  = 1
PIN: 1.10        Z_CS_PIN                               Input  = 1
.                Z_SERIAL_RX_PIN                        Input  = 1
PIN: 1.14        Z_SERIAL_TX_PIN                        Output = 1
PIN: 1.15        Y_CS_PIN                               Input  = 1
.                Y_SERIAL_RX_PIN                        Input  = 1
PIN: 1.16        Y_SERIAL_TX_PIN                        Output = 1
PIN: 1.17        X_CS_PIN                               Input  = 1
.                X_SERIAL_RX_PIN                        Input  = 1
PIN: 1.18        LCD_PINS_ENABLE                        Output = 0
PIN: 1.19        LCD_PINS_RS                            Output = 0
PIN: 1.20        LCD_PINS_D4                            Output = 0
PIN: 1.21        LCD_PINS_D5                            Output = 0
PIN: 1.22        LCD_PINS_D6                            Input  = 1
PIN: 1.23        LCD_PINS_D7                            Input  = 1
PIN: 1.24        <unused/unknown>                       Input  = 1
PIN: 1.25        Z_MIN_PIN                              protected 
PIN: 1.26        <unused/unknown>                       Input  = 1
PIN: 1.27        Y_MIN_PIN                              protected 
PIN: 1.28        FIL_RUNOUT_PIN                         Input  = 1
PIN: 1.29        X_MIN_PIN                              protected 
PIN: 1.30 (A 4)  BEEPER_PIN                             Output = 0
PIN: 1.31 (A 5)  SD_DETECT_PIN                          Input  = 0
PIN: 2.00        CONTROLLER_FAN_PIN                     protected 
.                SERVO0_PIN                             protected 
PIN: 2.01        X_ENABLE_PIN                           protected 
PIN: 2.02        X_STEP_PIN                             protected 
PIN: 2.03        FAN_PIN                                protected 
PIN: 2.04        E0_AUTO_FAN_PIN                        protected 
.                FAN1_PIN                               protected 
PIN: 2.05        HEATER_BED_PIN                         protected 
PIN: 2.06        X_DIR_PIN                              protected 
PIN: 2.07        HEATER_0_PIN                           protected 
PIN: 2.08        Y_ENABLE_PIN                           protected 
PIN: 2.09        <unused/unknown>                       Input  = 0
PIN: 2.10        <unused/unknown>                       Input  = 1
PIN: 2.11        Z_DIR_PIN                              protected 
PIN: 2.12        E0_ENABLE_PIN                          protected 
PIN: 2.13        E0_STEP_PIN                            protected 
PIN: 3.25        BTN_EN2                                Input  = 1
PIN: 3.26        BTN_EN1                                Input  = 1
PIN: 4.28        <unused/unknown>                       Output = 0
PIN: 4.29        X_SERIAL_TX_PIN                        Output = 1
ok
``` 
</details>

doggyfan, 2019-05-18:
When you say the **mapped drive** @Bob-the-Kuhn, do you mean the boards SD Card via the USB (just checking) ?

doggyfan, 2019-05-18:
@gloomyandy has mentioned it could well be the watchdog causing the reboot, but not sure why it's doing so.

Bob-the-Kuhn, 2019-05-18:
Yes, by mapped drive I mean the SD card via the USB.

Fruit & double fruit -  just exactly the type of problem that is a real pain to nail down.  Intermittent and symptoms that seem to vary.  

Maybe I'll be lucky and some WDT resets in the M43 code will keep it from happening.

doggyfan, 2019-05-18:
What's the watchdog timeout period I wonder on the LPC1768 (SKR 1.3 board), ie how long has Marlin got to do a WDT reset before it times out ?

gloomyandy, 2019-05-18:
Watchdog timeouts on LPC176x based boards are a real pain, because the default bootloader "hides" the fact that a watchdog has occurred from Marlin. It would be nice to get that sorted out if we can.

The time it takes to map a drive depends a lot on the size of the drive and the PC OS. For some reason both Windows and Mac insist on reading all of the FAT blocks when they mount a drive. On a large drive this can take a fair while, then to add insult to injury if this is the second time that Windows has seen the drive (so if you have used it locally and then released it back to Windows), Windows will read the FAT twice! I spent a long time digging into this when working on the USB code.

In this case I have a suspicion that output a large volume of text is causing the problem especially if that output is going to both the hardware serial port and to the USB serial port. I think that if you just use the USB port it may work, but I've not had chance to test that theory yet.

Patag, 2019-05-18:
@gloomyandy : Can I help for some tests (with Win 10 OS) ?

doggyfan, 2019-05-18:
The text output across the USB serial port is quite fast. Maybe issue a WD reset at the beginning of the M43 command ?

doggyfan, 2019-05-18:
@gloomyandy, I've disabled SERIAL_PORT_2 (I normally have it to UART-0 for the WiFi adapator), but it still reboots about 2 seconds after the M43 has outputted it's text.

Bob-the-Kuhn, 2019-05-18:
I'm 95% sure that the analog reads are involved with this problem.  You can never be sure with this type of a problem until root cause is found.

Please replace the **pinsDebug.h** file in the **Marlin\src\pins\** directory with this one.  It has the analog reads disabled in the M43 command.  Let me know if this "fixes" the hangs/resets for you.
[pinsDebug.zip](https://github.com/MarlinFirmware/Marlin/files/3194150/pinsDebug.zip)

It's definitely a WDT timeout.  Disabling the WDT in Configuration_adv.h  resulst in no hangs or resets.  But ... adding WDT resets after each pin report, after each M43 analog read and before each M43 analog read made no difference.

I started looking at the analogRead when I noticed that all the partial reports stopped just before reporting an analog value.  SInce the thermal manager uses an interrupt to oversample the ADC I tried the following:
- Moved the analog reads out of the print statements.
- Disabled all interrupts before the relocated analog reads
- Added a statement at the very beginning of the **uint16_t analogRead(pin_t adc_pin)** function that skips the ADC conversion if one is already in progress.

Bob-the-Kuhn, 2019-05-18:
The LPC1769 also hanges/resets.

doggyfan, 2019-05-18:
Yes that seems OK now, no reboots. So it's something to do with analogue pin interrogation ?

Patag, 2019-05-18:
Runs fine here also. 10 M43 in a raw, no reboot at all.

Bob-the-Kuhn, 2019-05-21:
@p3p - I'm out of ideas.  Suggestions would be welcomed.

FanDjango, 2019-05-21:
My M43 output hangs after a random amount of lines. Only one time have I ever seen the full output.

p3p, 2019-05-21:
I've never really been through the M43 code, I'l have to find time to investigate.. there are many reasons accessing a pin wrong could cause the MCU to panic.

tpruvot, 2019-05-21:
its more or less what i experienced porting it to STM32F1 : see #14072

Bob-the-Kuhn, 2019-05-21:
**pins\pinsDebug.h** has the code that is accessing the pins.

It uses **digitalRead_mod** to read the pins.  **digitalRead_mod** is supplied by the HAL's **pinsDebug.h** .  In the LPC1768's case it has` #define digitalRead_mod(p) extDigitalRead(p)`

It does not write to any pins.

For analog inputs it uses `(long)analogRead(DIGITAL_PIN_TO_ANALOG_PIN(pin))`

VanessaE, 2019-05-21:
> Actually, so is an M122 command (causing reboot).

@doggyfan, same here, with TMCStepper 0.3.4, and Marlin at commit 7b4c3bd92a5f327c4118483fe2eeb45d5f1416a1, but with my TMC2208's *removed* for debugging reasons.  I connect over USB; the output from `M122` trickles in slowly, but otherwise looks correct and complete, up until it causes the SKR v1.1 to reboot (after which, it stays in its bootloader).

@gloomyandy, I don't think it has to do with the volume of text.  I can do `M115` or `M503` or the like, over and over (by just holding down "Enter" to keep repeating the command), flooding the console with kilobytes worth of responses, without crashing the firmware.

teemuatlut, 2019-05-23:
I'm reducing the number of TMC2208::read retries to prevent watchdog reset.
`M122` has a lot of reads because it currently doesn't cache any of the registers. That's on the works as soon as I get a few other things out of the way.

boelle, 2019-07-21:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

gruvin, 2019-07-29:
"Me too", precisely as described by others -- M43. Haven't seen issues with TMC2208 yet, which I am using, in UART mode.

mylife4aiurr, 2019-08-12:
Dont close, this still effects LPC1768, skr v1.3. 
Cant M43, but M122 works without crash.

sl1pkn07, 2019-08-12:
M43 causes hard hang (not reboot) in my ReARM (LPC1768)

p3p, 2019-08-13:
I know about the issue and have a fix, it will eventually turn up when I roll out an update to the framework (it's not an issue with Marlin).

DiegoGiovany, 2019-09-05:
> I know about the issue and have a fix, it will eventually turn up when I roll out an update to the framework (it's not an issue with Marlin).

Can you release, or point to fix?

DiegoGiovany, 2019-11-04:
> I know about the issue and have a fix, it will eventually turn up when I roll out an update to the framework (it's not an issue with Marlin).

Even with updated framework, and marlin, it still freeze after m43.

p3p, 2019-11-04:
That is unlikely, you sure you flashed to the new firmware correctly, the code that was causing the halt is not there anymore.

sjasonsmith, 2019-11-04:
I just updated to the updated framework and verified I have no issues with `M43 I` on my SKR 1.3.

DiegoGiovany, 2019-11-05:
Correction... It kills the uart connection, If using USB cable, M43 works, but permanently disable uart connection... so it canot be operated over esp3d wifi

p3p, 2019-11-05:
That should be an easy fix, `M43` shouldn't be trying to debug protected pins, those should include the active serial ports but perhaps isn't been defined correctly. Although `M43 I` will override that (apart from usb)

closed 2020-06-20 by boelle