MarlinFirmware/Marlin #14051: [2.0.x] TMC Settings Ignored

thisiskeithb, 2019-05-18:
### Description

TMC driver settings are using values not set by me which causes the steppers to not move at all. Homing halts the printer since there is no axis movement.

**Expected behavior:** Use TMC values set in Marlin configs

**Actual behavior:** Using random values for TMC drivers like 256 microstepping and _really_ high hybrid thresholds.

#### Additional Information

After porting my settings from the 7ccbcf2 commit to the 1dad6e7 commit from earlier today (both are attached below), my TMC5160 drivers no longer work as intended on a BigTreeTech SKR 1.3.

Running M122 after flashing the 1dad6e7 commit from today returns very odd values:
```
Send: M122
Recv: 		X	Y	Z	E
Recv: Enabled		false	false	false	false
Recv: Set current	635	950	530	890
Recv: RMS current	631	942	518	876
Recv: MAX current	890	1328	730	1235
Recv: Run current	31/31	31/31	31/31	31/31
Recv: Hold current	15/31	15/31	15/31	15/31
Recv: Global scaler	53/256	79/256	44/256	74/256
Recv: CS actual		8/31	8/31	8/31	8/31
Recv: PWM scale	8	8	8	8
Recv: stealthChop	false	false	false	false
Recv: msteps		256	256	256	256
Recv: tstep		max	max	max	max
Recv: pwm
Recv: threshold		49	49	39	31
Recv: [mm/s]		3227.04	3227.04	810.90	491.64
Recv: OT prewarn	false	false	false	false
Recv: OT prewarn has
Recv: been triggered	false	false	false	false
Recv: off time		0	0	0	0
Recv: blank time	36	36	36	36
Recv: hysteresis
Recv: -end		-1	-1	-1	-1
Recv: -start		6	6	6	6
Recv: Stallguard thrs	1	-2	0	0
Recv: DRVSTATUS	X	Y	Z	E
Recv: stallguard
Recv: sg_result		0	0	0	0
Recv: fsactive
Recv: stst		X	X	X	X
Recv: olb				X
Recv: ola				X
Recv: s2gb
Recv: s2ga
Recv: otpw
Recv: ot
Recv: Driver registers:
Recv: 		X	0x80:08:00:00
Recv: 		Y	0x80:08:00:00
Recv: 		Z	0xE0:08:00:00
Recv: 		E	0x80:08:00:00
Recv: 
Recv: 
Recv: Testing X connection... OK
Recv: Testing Y connection... OK
Recv: Testing Z connection... OK
Recv: Testing E connection... OK
Recv: ok
```

Running M122 on the 7ccbcf2 commit from 5/16/19 returns the correct values:
```
Send: M122
		X	Y	Z	E
Enabled		false	false	false	false
Set current	635	950	530	890
RMS current	631	942	518	876
MAX current	890	1328	730	1235
Run current	31/31	31/31	31/31	31/31
Hold current	15/31	15/31	15/31	15/31
Global scaler	53/256	79/256	44/256	74/256
CS actual		15/31	15/31	15/31	15/31
PWM scale	0	0	0	0
stealthChop	true	true	true	false
msteps		16	16	16	16
tstep		max	max	max	max
pwm
threshold		49	49	39	31
[mm/s]		201.69	201.69	50.68	30.73
OT prewarn	false	false	false	false
OT prewarn has
been triggered	false	false	false	false
off time		4	4	4	4
blank time	24	24	24	24
hysteresis
-end		2	2	2	2
-start		1	1	1	1
Stallguard thrs	1	-2	0	0
DRVSTATUS	X	Y	Z	E
stallguard
sg_result		0	0	0	0
fsactive
stst		X	X	X	X
olb
ola
s2gb
s2ga
otpw
ot
Driver registers:
		X	0x80:0F:40:00
		Y	0x80:0F:40:00
		Z	0x80:0F:40:00
		E	0x80:0F:00:00

Testing X connection... OK
Testing Y connection... OK
Testing Z connection... OK
Testing E connection... OK
ok
```
I always run M502 and M500 after flashing since it's part of my post-flash script in OctoPrint, but just for good measure, I issued the commends manually and nothing changes.

Config files:
[Ender-3 - SKR 1.3 - Marlin-bugfix-2.0.x - 1dad6e7 - 5.18-19.zip](https://github.com/MarlinFirmware/Marlin/files/3194106/Ender-3.-.SKR.1.3.-.Marlin-bugfix-2.0.x.-.1dad6e7.-.5.18-19.zip)

[Ender-3 - SKR 1.3 - Marlin-bugfix-2.0.x - 7ccbcf2 - 5-16-2019.zip](https://github.com/MarlinFirmware/Marlin/files/3194103/Ender-3.-.SKR.1.3.-.Marlin-bugfix-2.0.x.-.7ccbcf2.-.5-16-2019.zip)

Patag, 2019-05-18:
Maybe you could try M502 (EEPROM reset) and M500 (save current settings).

thisiskeithb, 2019-05-18:
@Patag: As I said above:

> I always run M502 and M500 after flashing since it's part of my post-flash script in OctoPrint, but just for good measure, I issued the commands manually and nothing changes.

thinkyhead, 2019-05-18:
We'll have to trace where TMC drivers are initialized at startup and see if any obvious errors jump out. I don't have a Trinamic setup at this time, but I may be able to run the code on a bare board and follow the bouncing bits.

thisiskeithb, 2019-05-19:
This looks to be an issue with the latest TMC Stepper library (@teemuatlut). I specified v0.3.2 in platformio.ini and recompiled the 1dad6e7 commit and now I'm back in business:

```
Send: M122
		X	Y	Z	E
Enabled		false	false	false	false
Set current	635	950	530	890
RMS current	631	942	518	876
MAX current	890	1328	730	1235
Run current	31/31	31/31	31/31	31/31
Hold current	15/31	15/31	15/31	15/31
Global scaler	53/256	79/256	44/256	74/256
CS actual		15/31	15/31	15/31	15/31
PWM scale	0	0	27	0
stealthChop	true	true	true	false
msteps		16	16	16	16
tstep		max	max	max	max
pwm
threshold		49	49	39	31
[mm/s]		201.69	201.69	50.68	30.73
OT prewarn	false	false	false	false
OT prewarn has
been triggered	false	false	false	false
off time		4	4	4	4
blank time	24	24	24	24
hysteresis
-end		2	2	2	2
-start		1	1	1	1
Stallguard thrs	1	-2	0	0
DRVSTATUS	X	Y	Z	E
stallguard
sg_result		0	0	0	0
fsactive
stst		X	X	X	X
olb
ola
s2gb
s2ga
otpw
ot
Driver registers:
		X	0x80:0F:40:00
		Y	0x80:0F:40:00
		Z	0x80:0F:40:00
		E	0x80:0F:00:00

Testing X connection... OK
Testing Y connection... OK
Testing Z connection... OK
Testing E connection... OK
ok
```

jmdearras, 2019-05-19:
@thisiskeithb  how do you specify a particular version in platformio?

thisiskeithb, 2019-05-19:
@jmdearras: Look for `TMCStepper@<1.0.0` under the LPC1768 environment section and change it to `TMCStepper=https://github.com/teemuatlut/TMCStepper/archive/v0.3.2.zip`

teemuatlut, 2019-05-19:
I was able to confirm your issue going when from TMCStepper v0.3.2 to v0.3.3.

kAdonis, 2019-05-19:
I tested which commit started breaking things for me, last working commit of the TMCStepper library is teemuatlut/TMCStepper@d33aba2

teemuatlut, 2019-05-19:
That doesn't make changes to the code.
The problem is with SW SPI trying to follow MODE3 but it isn't quite there yet.

thisiskeithb, 2019-05-20:
@teemuatlut:  Reverting SW SPI MODE3 changes in v0.3.4 resolved my issues, thank you!

I had to specify that particular version in platformio.ini (maybe due to some propagation with CDNs?), but hopefully PIO will start grabbing the latest copy off Github for everyone soon.

closed 2019-05-20 by thisiskeithb

VanLaser, 2019-05-27:
I know this is closed, but (since I probably don't understand it exactly), would this affect anyone anymore? I.e. can I just update my fork with latest Marlin bugfix-2.0.x/TMCStepper and re-apply my configs over it, or do I have to do something extra (e.g. remove TMCStepper library folder)?  Thanks.

teemuatlut, 2019-05-27:
Rebase your fork and apply settings.
You can remove the `.pioenvs` and `.piolibdeps` folders for good measure if you're using PIO.
Then just make sure you're compiling the latest library version.

VanLaser, 2019-05-27:
Thanks for the prompt answer!