MarlinFirmware/Marlin #14052: Running out of Memory[FR] (feature request title)

davidcgu, 2019-05-18:


doggyfan, 2019-05-18:
mmmm, it comes with age I'm afraid.

robbycandra, 2019-05-18:
Too many things to remember. So many things to forget.

Bob-the-Kuhn, 2019-05-18:
My understanding is mesh slots exist only in EEPROM and that only one is ever loaded into RAM. 

I don't think that cutting down on mesh slots will affect RAM usage.  Reducing the number of points in the mesh will reduce RAM usage.

Two things you might try:.  I know that these save FLASH.  Maybe they also save RAM.
- Compile 1.1.9 with PlatformIO
- Switch to 2.0.x and compile with PlatformIO

Roxy-3D, 2019-05-18:
You can lower the number of mesh points in both the X and Y direction.   

You probably should leave #define BLOCK_BUFFER_SIZE 16    That is a reasonable amount of buffer storage.   If you really need to save some RAM memory, you can bump it down to 8.   But you might see some stuttering on very small arcs and circles.

MasterPIC, 2019-05-18:
SlimLCD

guestisp, 2019-05-18:
Same situation here.
Will the RX buffer set to 1024 increase reliability during prints?

As far as i can see, when printing from USB, a BLOCK_BUFFER_SIZE set to 16 could be too low but it's almost impossible to set it to 32 without loosing features

What if i see stuttering with BLOCK_BUFFER_SIZE as 16? Any workaround?

I have to try to print with baud rate set to 500000 insted 250000. The connection was done properly, i'm able to manage the printer but i don't know if during a print a so high baud rate lead to issues

Roxy-3D, 2019-05-18:
> What if i see stuttering with BLOCK_BUFFER_SIZE as 16? Any workaround?

You can switch to a faster board...   Any one of the 32-bit boards will be faster.  

You can also reduce your max feed rates and acceleration so the printer is a little bit slower.  (That will give the processor more time between line segments to process each command.)

You can switch to a 20x4 LCD Panel instead of a graphical LCD.

You can raise the number in:   #define LCD_UPDATE_INTERVAL 100    to something like 500.   The LCD Panel will be more sluggish, but it will consume considerably less CPU cycles.

davidcgu, 2019-05-18:
That is actually good to know as I'm having an 20x4 LCD but only use it in a very punctual situation and really don't need it normally so will increase refresh time for sure if helps somehow.

I have considered about 32 bit board but I think wait a few months is better as most probably will be a big amount of them at a good price, even today it's available some duet wifi 2 clone at a very reasonable price.

I thought having several mesh slots considering each point is a floating number was consuming several resources but understand that is not the case then.

Regards

guestisp, 2019-05-18:
@Roxy-3D  wait, AFAIK, the issue is not related to the printer CPU or the firmware unable to print fast, I think it's related to the USB connection too slow. The printer is printing too fast for the USB connection..

I've set the baud rate to 500000 (it was 250000), i was able to move the printer, connect, disconnect and also flash the firmware. Still have to try a print, but if 500000 works properly, it should send twice faster than with 250000, thus resolving the issue

My feedrates are 200-400 or 500 (it depens on the test i'm doing), so, pretty low and I don't print faster than 60mm/s

Roxy-3D, 2019-05-18:
It maybe you have a slow USB connection or maybe some software installed that is slowing it down.
But just as an FYI:    I have never seen the USB Serial port have trouble keeping the receive buffer queue full.    

I have the Max7219 Debug LED Matrix's on my printers.   And the idle task can be configured to display how full the receive buffer queue is.    I've never seen it drain all the way empty with a setting of 16 buffers.   Usually...   It sits all the way full.       It might be worth $3 to buy a Matrix and see how your queue is holding up.

davidcgu, 2019-05-18:
Thanks you guys, I have not tested personally but read bout it, seems maybe the article is not right and really 16 is enough, will make some test to check.

guestisp, 2019-05-19:
Thank you for the suggestions.
Look at this video https://youtu.be/SBZ7MfvAsGc but there are tons of articles or deeper tests showing a bottleneck in the USB printing

robbycandra, 2019-05-19:
In my experience, printing from SD card is slower than USB.

guestisp, 2019-05-19:
Strange .... Which baud rate and sd card are you using?

robbycandra, 2019-05-19:
Try to print vase mode which many segment, big gcode file. It will be easier if you print from USB.

guestisp, 2019-05-19:
I think a vase mode is faster to handle as you just have one perimeter, no infill, no retraction, a constant Z increase and so on 

In fact, our vases are printed properly. The same object printed in standard mode, most of the time is full of blobs and yesterday i've seen they could be caused by an insufficient USB speed to keep the buffer full (or a too small buffer)

A workaround could be to implement a buffer stored on the printer SD card. From USB you start to print, marlin writes a chunk of command to SD (maybe 5-10 seconds) then start to run these commands. In this way you can have huge buffers without any memory constraints.

A sort of a filesystem journal or the writeback cache of any raid controller

davidcgu, 2019-05-19:
I think I will make some tests my own to get a conclusion (at least with my config) and will share the results later on

FanDjango, 2019-05-19:
Octoprint is your culprit, to a certain degree. It processes a lot of stuff for every GCODE. And the python serial framework is single threaded. And then on 8 bit boards you cannot realistically increase receive buffers and also not the baud rate and stay reliable. Always compare the "stuttering" in tight curves to printing the same object from SD-card, by the way. In the Octoprint github there is an epic issue thread about this problem. My only universal solution to this problem was a move to a 32 bit board. Since then, this is no longer a problem for me. Baudrate 250000, BLOCK_BUFFER_SIZE 64, RX_BUFFER_SIZE 128. So what are you waiting for?

guestisp, 2019-05-19:
I'm using octoprint without any plugins processing gcode, but yes, I think octoprint is the culprit.. The epic thread is saying that the issue is related to the 8bit board, but I don't think so, is much more likely that octoprint is unable to keep the buffer full.

I'm running on a Pi B+, the load avg is 0.50

Which 32bit board do you suggest, 100% compatibile with Marlin 2?

davidcgu, 2019-05-19:
I'm actually considering this step however I have a customer Hbot and had issues with TMC drivers as seems were not providing enough power however works fine with DRV.
I saw Duet wifi clone but is still over 80€ and the economic option of BIGTREETECH SKR V1.3 does only works with TMC, so that's my real problem actually.

Working on it.

Regards

davidcgu, 2019-05-19:
> I'm using octoprint without any plugins processing gcode, but yes, I think octoprint is the culprit.. The epic thread is saying that the issue is related to the 8bit board, but I don't think so, is [...]

I think you have the reply on my previous one...

davidcgu, 2019-05-19:
FYI I've being able to move from 8 to 16

// The ASCII buffer for serial input
#define MAX_CMD_SIZE 96
#define BUFSIZE 16

By disabling on Marlin:

 //#define ABL_BILINEAR_SUBDIVISION

Anyway I think I will move to 32 bit...

guestisp, 2019-05-19:
> #define BUFSIZE 16

Mine is 4. But is this the buffer we have to increase ? i've thought it was `BLOCK_BUFFER_SIZE`

FanDjango, 2019-05-20:
I am using MKS SBASE 1.3 and am happy with it. But I am not saying it is the best - it was cheap.

Vertabreak, 2020-01-19:
can confirm the issues with octopi is with the speed of the pi its being run on the pi3b+ and pi4b have zero issues when marlin has the default bufsize 4 and block buffer of 16. this was a issue with mostly pi0 simply didn't have enough CPU to handle octopi they suggest using a pi3 or better on the site now. 
@boelle i think this one is safe to close

closed 2020-01-24 by boelle