MarlinFirmware/Marlin #14053: [BUG] STM32 SKR-MINI compile failure duplicate INLINE statements/

jmdearras, 2019-05-18:
### Description

STM32 compile errors with SKR-MINI, claims many duplicate "INLINE" statements.
Using BigTreeTech branch, but the problem seems generic to STM32.
[Configuration_adv.zip](https://github.com/MarlinFirmware/Marlin/files/3194264/Configuration_adv.zip)

### Steps to Reproduce

1. [First Step]
2. [Second Step]
3. [and so on...]

**Expected behavior:** [What you expect to happen]

**Actual behavior:** [What actually happens]

#### Additional Information

* Include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

jmz52, 2019-05-19:
From platformio.ini
`[env:BIGTREE_SKR_MINI]`
`platform      = ststm32@<4.4.0`
`framework     = arduino`
`board         = genericSTM32F103RC`
`extra_scripts = buildroot/share/PlatformIO/scripts/STM32F1.py`

"duplicate inline" problem caused by old ststm32 platform used by BigTreeTech (ststm32@<4.4.0).
Problem was resolved in newer versions of ststm32 platform.
https://github.com/MarlinFirmware/Marlin/issues/13075

Please note that script  buildroot/share/PlatformIO/scripts/STM32F1.py will not work with ststm32 version 5.x.x
Line `env['CPPDEFINES'].remove(("VECT_TAB_ADDR", 134217728))` will fail as value '134217728' is specific for old ststm32 platform.
You can use working code from https://github.com/MarlinFirmware/Marlin/blob/bugfix-2.0.x/buildroot/share/PlatformIO/scripts/mks_robin.py

tpruvot, 2019-05-19:
its not even merged, cant be a bug here... see #14041

jmdearras, 2019-05-19:
> its not even merged, cant be a bug here... see #14041 
I cannot get the MOTHERBOARD BOARD_STM32F1R to compile, either, and that is in the main build.

Bob-the-Kuhn, 2019-05-19:
I ran into the duplicate "INLINE" problem a couple of days ago.  Here's the short term fix that I did:
1. In **.platformio** I did a search for **static inline __always_inline**
2. I pulled all the listed files into Notepad++
3. I replaced that phrase with **static inline** in all the files & wrote them back.

I'm new to the STM32 world.  This may not be the best advice.

jmz52, 2019-05-19:
> 134217728 is 0x08000000 in decimal :p

@tpruvot , I know, but python does not care about dec2hex conversion.
ststm32 platform 4.x.x uses decimal number to define flash address space. In 5.x.x it was changed to use hex.
That is why updated MKS Robin's script uses search by key only rather that key and value pair.

`for define in env['CPPDEFINES']:`
&nbsp;&nbsp;&nbsp;&nbsp;`if define[0] == "VECT_TAB_ADDR":`
&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;`env['CPPDEFINES'].remove(define)`
`env['CPPDEFINES'].append(("VECT_TAB_ADDR", "0x08007000"))`

@Bob-the-Kuhn , I would recommend to update PlatformIO and stm32 packages and remove stmst32 version requirements (from platformio.ini) if it is set in your build environment.
Also what 'board' you are using in your build environment?
Bluepill for example need few more tricks to compile code using stm32duino as default board's definitions force platformio to use stm32 core libraries.

jmdearras, 2019-05-19:
I am also new to STM32.  Did it compile and run?

Bob-the-Kuhn, 2019-05-19:
I'm using Marlin's default platformio.ini.  As best I can tell there are no version requirements in it nor in the supporting python scripts.

I deleted all my PlatformIO packages and platforms.  On the next F1 build it re-installed 4.4.0.

Bob-the-Kuhn, 2019-05-19:
@jmdearras - I'm getting an "unknown motherboard"  error using your config files..  Please post your pins_BIGTREE_SKR_MINI_V1_1.h file.

Bob-the-Kuhn, 2019-05-19:
Sometimes I amaze myself.  I just looked at PR #14041.  Never mind ...

@jmdearras - I suggest you wait until PR #14041 is merged and then try again.  Unfortunately the timeline is indeterminate.  If you can't wait then I suggest downloading the PR #14041 code and trying it.

jmdearras, 2019-05-19:
I tried MOTHERBOARD BOARD_STM32F1R with the default config, and it will not
compile, either.

jmdearras, 2019-05-19:
tried again with a main branch from 5/18 and a supported board.

^~~~~~~~~~~~~~~
C:\users\jimde\.platformio\packages\framework-arduinoststm32\STM32F1\cores\maple/ext_interrupts.h:123:15: error: duplicate 'inline'
static inline __always_inline void noInterrupts() {
^~~~~~~~~~~~~~~
C:\users\jimde\.platformio\packages\framework-arduinoststm32\STM32F1\system\libmaple\include/libmaple/nvic.h:112:15: error: duplicate 'inline'
static inline __always_inline void nvic_globalirq_enable() {

^~~~~~~~~~~~~~~
C:\users\jimde\.platformio\packages\framework-arduinoststm32\STM32F1\system\libmaple\include/libmaple/nvic.h:119:15: error: duplicate 'inline'
static inline __always_inline void nvic_globalirq_disable() {
^~~~~~~~~~~~~~~
C:\users\jimde\.platformio\packages\framework-arduinoststm32\STM32F1\cores\maple/ext_interrupts.h:109:15: error: duplicate 'inline'
static inline __always_inline void interrupts() {
^~~~~~~~~~~~~~~
C:\users\jimde\.platformio\packages\framework-arduinoststm32\STM32F1\cores\maple/ext_interrupts.h:123:15: error: duplicate 'inline'
static inline __always_inline void noInterrupts() {
^~~~~~~~~~~~~~~

Latest PlatformIO and Atom.
[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/3194546/Configuration.zip)

Bob-the-Kuhn, 2019-05-19:
I downloaded the PR #14041 code, added your config files and then did the inline fix.  It compiled!

Here's the files you need:
1) unzip & copy this file to .platformio\platforms\
[ststm32.zip](https://github.com/MarlinFirmware/Marlin/files/3195691/ststm32.zip)
2) unzip & copy this file to .platformio\packages
[framework-arduinoststm32.zip](https://github.com/MarlinFirmware/Marlin/files/3195693/framework-arduinoststm32.zip)
3) unzip & copy these files into your top level Marlin directory
[PR 14041.zip](https://github.com/MarlinFirmware/Marlin/files/3195694/PR.14041.zip)

I all cases it should first ask if you want to merge the new directories with the old and then it'll ask if you want to replace the old files with the new.  Click the "do this for all" box and then do the merge and then replace the files.

jmdearras, 2019-05-19:
@Bob-the-Kuhn close.
^~~~~~~~~~~~~~~~
Marlin\src\gcode\sdcard\M20-M30_M32-M34_M524_M928.cpp:152:5: note: suggested alternative: 'SERIAL_ECHOPGM'
SERIAL_ECHOPGM_P(port, "Current file: ");
^~~~~~~~~~~~~~~~
SERIAL_ECHOPGM
Marlin\src\gcode\sdcard\M20-M30_M32-M34_M524_M928.cpp:170:5: error: no matching function for call to 'CardReader::report_status(const int16_t&)'

);
^
In file included from Marlin\src\gcode\sdcard\M20-M30_M32-M34_M524_M928.cpp:28:
Marlin\src\gcode\sdcard\../../sd/cardreader.h:73:15: note: candidate: 'static void CardReader::report_status()'
static void report_status();
^~~~~~~~~~~~~
Marlin\src\gcode\sdcard\../../sd/cardreader.h:73:15: note:   candidate expects 0 arguments, 1 provided
Compiling .pioenvs\BIGTREE_SKR_MINI\src\src\gcode\sdcard\M27.cpp.o
*** [.pioenvs\BIGTREE_SKR_MINI\src\src\gcode\sdcard\M20-M30_M32-M34_M524_M928.cpp.o] Error 1
 [ERROR] Took 32.63 seconds

If I disable SD support in the config,  I get

from C:\users\jimde\.platformio\lib\U8glib_ID7\src\clib\u8g_com_arduino_parallel.c:63:
c:\users\jimde\.platformio\packages\toolchain-gccarmnoneeabi\arm-none-eabi\include\stdlib.h:255:6: note: previous declaration of 'random' was here
long random (void);
^~~~~~
compilation terminated due to -fmax-errors=5.
*** [.pioenvs\BIGTREE_SKR_MINI\lib9c4\U8glib_ID7\clib\u8g_com_arduino_parallel.c.o] Error 1

 [ERROR] Took 51.08 seconds

jmdearras, 2019-05-19:
from C:\users\jimde\.platformio\lib\U8glib_ID7\src\clib\u8g_com_arduino_parallel.c:63:
c:\users\jimde\.platformio\packages\toolchain-gccarmnoneeabi\arm-none-eabi\include\stdlib.h:255:6: note: previous declaration of 'random' was here
long random (void);
^~~~~~
compilation terminated due to -fmax-errors=5.
*** [.pioenvs\BIGTREE_SKR_MINI\lib9c4\U8glib_ID7\clib\u8g_com_arduino_parallel.c.o] Error 1

 [ERROR] Took 51.08 seconds

jmdearras, 2019-05-19:
@Bob-the-Kuhn I get:

Compiling .pioenvs\BIGTREE_SKR_MINI\src\src\gcode\sdcard\M27.cpp.o
Marlin\src\gcode\sdcard\M20-M30_M32-M34_M524_M928.cpp: In static member function 'static void GcodeSuite::M20()':
Marlin\src\gcode\sdcard\M20-M30_M32-M34_M524_M928.cpp:53:3: error: 'SERIAL_ECHOLNPGM_P' was not declared in this scope
SERIAL_ECHOLNPGM_P(port, MSG_BEGIN_FILE_LIST);
^~~~~~~~~~~~~~~~~~
Marlin\src\gcode\sdcard\M20-M30_M32-M34_M524_M928.cpp:53:3: note: suggested alternative: 'SERIAL_ECHOLNPGM'

SERIAL_ECHOLNPGM_P(port, MSG_BEGIN_FILE_LIST);
^~~~~~~~~~~~~~~~~~
SERIAL_ECHOLNPGM
Marlin\src\gcode\sdcard\M20-M30_M32-M34_M524_M928.cpp:58:3: error: no matching function for call to 'CardReader::ls(const int16_t&)'
);
^
In file included from Marlin\src\gcode\sdcard\M20-M30_M32-M34_M524_M928.cpp:28:
Marlin\src\gcode\sdcard\../../sd/cardreader.h:86:15: note: candidate: 'static void CardReader::ls()'
static void ls();
^~
Marlin\src\gcode\sdcard\../../sd/cardreader.h:86:15: note:   candidate expects 0 arguments, 1 provided
Marlin\src\gcode\sdcard\M20-M30_M32-M34_M524_M928.cpp: In static member function 'static void GcodeSuite::M27()':
Marlin\src\gcode\sdcard\M20-M30_M32-M34_M524_M928.cpp:152:5: error: 'SERIAL_ECHOPGM_P' was not declared in this scope
SERIAL_ECHOPGM_P(port, "Current file: ");
^~~~~~~~~~~~~~~~
Marlin\src\gcode\sdcard\M20-M30_M32-M34_M524_M928.cpp:152:5: note: suggested alternative: 'SERIAL_ECHOPGM'
SERIAL_ECHOPGM_P(port, "Current file: ");
^~~~~~~~~~~~~~~~
SERIAL_ECHOPGM
Marlin\src\gcode\sdcard\M20-M30_M32-M34_M524_M928.cpp:170:5: error: no matching function for call to 'CardReader::report_status(const int16_t&)'
);
^
In file included from Marlin\src\gcode\sdcard\M20-M30_M32-M34_M524_M928.cpp:28:
Marlin\src\gcode\sdcard\../../sd/cardreader.h:73:15: note: candidate: 'static void CardReader::report_status()'
static void report_status();
^~~~~~~~~~~~~
Marlin\src\gcode\sdcard\../../sd/cardreader.h:73:15: note:   candidate expects 0 arguments, 1 provided
*** [.pioenvs\BIGTREE_SKR_MINI\src\src\gcode\sdcard\M20-M30_M32-M34_M524_M928.cpp.o] Error 1
 [ERROR] Took 41.39 seconds

jmdearras, 2019-05-20:
@Bob-the-Kuhn  if I remove SD support, I get a lot of:
n file included from C:\users\jimde\.platformio\packages\framework-arduinoststm32\STM32F1\cores\maple/wirish.h:58,
from C:\users\jimde\.platformio\packages\framework-arduinoststm32\STM32F1\cores\maple/Arduino.h:30,
from C:\users\jimde\.platformio\lib\U8glib_ID7\src\clib\u8g_com_arduino_no_en_parallel.c:69:
C:\users\jimde\.platformio\packages\framework-arduinoststm32\STM32F1\cores\maple/ext_interrupts.h:90:6: error: conflicting types for 'attachInterrupt'
void attachInterrupt(uint8 pin, voidArgumentFuncPtr handler, void *arg,

^~~~~~~~~~~~~~~
C:\users\jimde\.platformio\packages\framework-arduinoststm32\STM32F1\cores\maple/ext_interrupts.h:69:6: note: previous declaration of 'attachInterrupt' was h
ere
void attachInterrupt(uint8 pin, voidFuncPtr handler, ExtIntTriggerMode mode);
^~~~~~~~~~~~~~~
In file included from C:\users\jimde\.platformio\packages\framework-arduinoststm32\STM32F1\cores\maple/wirish.h:60,
from C:\users\jimde\.platformio\packages\framework-arduinoststm32\STM32F1\cores\maple/Arduino.h:30,
from C:\users\jimde\.platformio\lib\U8glib_ID7\src\clib\u8g_com_arduino_no_en_parallel.c:69:
C:\users\jimde\.platformio\packages\framework-arduinoststm32\STM32F1\cores\maple/wirish_math.h:50:6: error: conflicting types for 'random'
long random(long max);
^~~~~~
In file included from C:\users\jimde\.platformio\packages\framework-arduinoststm32\STM32F1\cores\maple/wirish.h:41,
from C:\users\jimde\.platformio\packages\framework-arduinoststm32\STM32F1\cores\maple/Arduino.h:30,
from C:\users\jimde\.platformio\lib\U8glib_ID7\src\clib\u8g_com_arduino_no_en_parallel.c:69:
c:\users\jimde\.platformio\packages\toolchain-gccarmnoneeabi\arm-none-eabi\include\stdlib.h:255:6: note: previous declaration of 'random' was here
long random (void);
^~~~~~
In file included from C:\users\jimde\.platformio\packages\framework-arduinoststm32\STM32F1\cores\maple/wirish.h:60,
from C:\users\jimde\.platformio\packages\framework-arduinoststm32\STM32F1\cores\maple/Arduino.h:30,
from C:\users\jimde\.platformio\lib\U8glib_ID7\src\clib\u8g_com_arduino_no_en_parallel.c:69:
C:\users\jimde\.platformio\packages\framework-arduinoststm32\STM32F1\cores\maple/wirish_math.h:59:6: error: conflicting types for 'random'
long random(long min, long max);
^~~~~~
In file included from C:\users\jimde\.platformio\packages\framework-arduinoststm32\STM32F1\cores\maple/wirish.h:41,
from C:\users\jimde\.platformio\packages\framework-arduinoststm32\STM32F1\cores\maple/Arduino.h:30,
from C:\users\jimde\.platformio\lib\U8glib_ID7\src\clib\u8g_com_arduino_no_en_parallel.c:69:
c:\users\jimde\.platformio\packages\toolchain-gccarmnoneeabi\arm-none-eabi\include\stdlib.h:255:6: note: previous declaration of 'random' was here
long random (void

type errors all in the maple directory

Bob-the-Kuhn, 2019-05-20:
**M20-M30_M32-M34_M524_M928.cpp** error?  That file isn't in the current 2.0.x.

 I think you started with an older Marlin and the compiler is complaining about a file that shouldn't be there.

Try this:
1) delete the Marlin directory (the one that is at the same level as platformio.ini).
2) copy the Marlin directory from the ZIP to where the old Marlin directory was.

jmdearras, 2019-05-20:
I used the BigTreeTech 7z of their build.  I will try again with new code
in the morning.  Thanks for all the help!

Bob-the-Kuhn, 2019-05-20:
That explains it.  Using BigTreeTech's fork of Marlin.

Best if you get a[ copy of 2.0.x ](https://github.com/MarlinFirmware/Marlin/tree/bugfix-2.0.x )directly from Marlin and then do the changes.

jmdearras, 2019-05-20:
Still no go.  :(

^~~~
In file included from C:\users\jimde\.platformio\packages\framework-arduinoststm32\STM32F1\cores\maple/wirish.h:54,
from C:\users\jimde\.platformio\packages\framework-arduinoststm32\STM32F1\cores\maple/Arduino.h:30,
from C:\users\jimde\.platformio\lib\U8glib_ID7\src\clib\u8g_com_arduino_common.c:46:
C:\users\jimde\.platformio\packages\framework-arduinoststm32\STM32F1\cores\maple/boards.h:111:1: error: unknown type name 'bool'; did you mean '_Bool'?

bool boardUsesPin(uint8 pin);
^~~~
_Bool
In file included from C:\users\jimde\.platformio\packages\framework-arduinoststm32\STM32F1\cores\maple/wirish.h:58,
from C:\users\jimde\.platformio\packages\framework-arduinoststm32\STM32F1\cores\maple/Arduino.h:30,
from C:\users\jimde\.platformio\lib\U8glib_ID7\src\clib\u8g_com_arduino_common.c:46:
C:\users\jimde\.platformio\packages\framework-arduinoststm32\STM32F1\cores\maple/ext_interrupts.h:90:6: error: conflicting types for 'attachInterrupt'
void attachInterrupt(uint8 pin, voidArgumentFuncPtr handler, void *arg,
^~~~~~~~~~~~~~~
C:\users\jimde\.platformio\packages\framework-arduinoststm32\STM32F1\cores\maple/ext_interrupts.h:69:6: note: previous declaration of 'attachInterrupt' was here
void attachInterrupt(uint8 pin, voidFuncPtr handler, ExtIntTriggerMode mode);
^~~~~~~~~~~~~~~
In file included from C:\users\jimde\.platformio\packages\framework-arduinoststm32\STM32F1\cores\maple/wirish.h:60,
from C:\users\jimde\.platformio\packages\framework-arduinoststm32\STM32F1\cores\maple/Arduino.h:30,
from C:\users\jimde\.platformio\lib\U8glib_ID7\src\clib\u8g_com_arduino_common.c:46:
C:\users\jimde\.platformio\packages\framework-arduinoststm32\STM32F1\cores\maple/wirish_math.h:50:6: error: conflicting types for 'random'
long random(long max);
^~~~~~
In file included from C:\users\jimde\.platformio\packages\framework-arduinoststm32\STM32F1\cores\maple/wirish.h:41,
from C:\users\jimde\.platformio\packages\framework-arduinoststm32\STM32F1\cores\maple/Arduino.h:30,
from C:\users\jimde\.platformio\lib\U8glib_ID7\src\clib\u8g_com_arduino_common.c:46:
c:\users\jimde\.platformio\packages\toolchain-gccarmnoneeabi\arm-none-eabi\include\stdlib.h:255:6: note: previous declaration of 'random' was here
long random (void);
^~~~~~
In file included from C:\users\jimde\.platformio\packages\framework-arduinoststm32\STM32F1\cores\maple/wirish.h:60,
from C:\users\jimde\.platformio\packages\framework-arduinoststm32\STM32F1\cores\maple/Arduino.h:30,
from C:\users\jimde\.platformio\lib\U8glib_ID7\src\clib\u8g_com_arduino_common.c:46:
C:\users\jimde\.platformio\packages\framework-arduinoststm32\STM32F1\cores\maple/wirish_math.h:59:6: error: conflicting types for 'random'
long random(long min, long max);
^~~~~~
In file included from C:\users\jimde\.platformio\packages\framework-arduinoststm32\STM32F1\cores\maple/wirish.h:41,
from C:\users\jimde\.platformio\packages\framework-arduinoststm32\STM32F1\cores\maple/Arduino.h:30,
from C:\users\jimde\.platformio\lib\U8glib_ID7\src\clib\u8g_com_arduino_common.c:46:
c:\users\jimde\.platformio\packages\toolchain-gccarmnoneeabi\arm-none-eabi\include\stdlib.h:255:6: note: previous declaration of 'random' was here
long random (void);
^~~~~~
compilation terminated due to -fmax-errors=5.
*** [.pioenvs\STM32F1\lib9c4\U8glib_ID7\clib\u8g_com_arduino_common.c.o] Error 1
In file included from C:\users\jimde\.platformio\packages\framework-arduinoststm32\STM32F1\cores\maple/boards.h:38,
from C:\users\jimde\.platformio\packages\framework-arduinoststm32\STM32F1\cores\maple/wirish.h:54,
from C:\users\jimde\.platformio\packages\framework-arduinoststm32\STM32F1\cores\maple/Arduino.h:30,
from C:\users\jimde\.platformio\lib\U8glib_ID7\src\clib\u8g_com_arduino_fast_parallel.c:69:
C:\users\jimde\.platformio\packages\framework-arduinoststm32\STM32F1\cores\maple/wirish_types.h:65:9: error: unknown type name 'bool'
typedef bool boolean;
^~~~
In file included from C:\users\jimde\.platformio\packages\framework-arduinoststm32\STM32F1\cores\maple/wirish.h:54,
from C:\users\jimde\.platformio\packages\framework-arduinoststm32\STM32F1\cores\maple/Arduino.h:30,
from C:\users\jimde\.platformio\lib\U8glib_ID7\src\clib\u8g_com_arduino_fast_parallel.c:69:
C:\users\jimde\.platformio\packages\framework-arduinoststm32\STM32F1\cores\maple/boards.h:111:1: error: unknown type name 'bool'; did you mean '_Bool'?
bool boardUsesPin(uint8 pin);
^~~~
_Bool
In file included from C:\users\jimde\.platformio\packages\framework-arduinoststm32\STM32F1\cores\maple/wirish.h:58,
from C:\users\jimde\.platformio\packages\framework-arduinoststm32\STM32F1\cores\maple/Arduino.h:30,
from C:\users\jimde\.platformio\lib\U8glib_ID7\src\clib\u8g_com_arduino_fast_parallel.c:69:
C:\users\jimde\.platformio\packages\framework-arduinoststm32\STM32F1\cores\maple/ext_interrupts.h:90:6: error: conflicting types for 'attachInterrupt'
void attachInterrupt(uint8 pin, voidArgumentFuncPtr handler, void *arg,
^~~~~~~~~~~~~~~
C:\users\jimde\.platformio\packages\framework-arduinoststm32\STM32F1\cores\maple/ext_interrupts.h:69:6: note: previous declaration of 'attachInterrupt' was here
void attachInterrupt(uint8 pin, voidFuncPtr handler, ExtIntTriggerMode mode);

jmz52, 2019-05-20:
This looks like a problem with outdated U8glib library.
Try to add U8glib to lib_ignore list and remove Marlin\.piolibdeps\U8glib-HAL_<whatever>.

jmdearras, 2019-05-20:
Yes, that worked!  Now to get the nerve to try the code on the board.

jmdearras, 2019-05-20:
> This looks like a problem with outdated U8glib library.
> Try to add U8glib to lib_ignore list and remove Marlin.piolibdeps\U8glib-HAL_< whatever >

  Yes, that worked!  Now to get the nerve to try the code on the board.

jmdearras, 2019-05-20:
the SDcard reader does not work (it works with the demo code), it sees the card inserted and removed, but the menu says "No SD Card".  I'll dig into it.

thisiskeithb, 2019-05-21:
> I downloaded the PR #14041 code, added your config files and then did the inline fix. It compiled!
>
> Here's the files you need:
>
> 1. unzip & copy this file to .platformio\platforms [...]

@Bob-the-Kuhn: So we have to patch our files locally in order to compile Marlin for this board?

Bob-the-Kuhn, 2019-05-21:
That's what I had to do.

jmdearras, 2019-05-21:
I have not convinced @BigTreeTech that this is a problem, yet. :(  I

tpruvot, 2019-05-21:
in all cases, these c/c++ bugs cant be fixed in Marlin repo...

jmz52, 2019-05-21:
I was able to compile the PR https://github.com/MarlinFirmware/Marlin/pull/14041 code in Atom + PlatformIO without any patches to stm32duino.

PlatformIO components
```
> pio update
Updating tool-scons                      @ 2.20501.7      [Up-to-date]
Updating tool-unity                      @ 1.20403.0      [Up-to-date]
Updating contrib-pysite                  @ 2.27.190418    [Up-to-date]
Updating contrib-piohome                 @ 2.0.2          [Up-to-date]
Updating tool-pioplus                    @ 2.2.1          [Up-to-date]

Platform Manager
================
Platform Atmel AVR
--------
Updating atmelavr                        @ 1.13.0         [Up-to-date]
Updating toolchain-atmelavr              @ 1.50400.0      [Up-to-date]
Updating framework-arduinoavr            @ 3.10623.190424 [Up-to-date]

Platform ST STM32
--------
Updating ststm32                         @ 5.4.1          [Up-to-date]
Updating framework-arduinoststm32-maple  @ 1.10000.190516 [Up-to-date]
Updating toolchain-gccarmnoneeabi        @ 1.70201.0      [Up-to-date]

Library Manager
===============
```

Configuration files from https://github.com/MarlinFirmware/Marlin/issues/14053#issuecomment-493796521

In Configuration_adv.h comment out the EVENT_GCODE_SD_STOP option
`//#define EVENT_GCODE_SD_STOP "G28XY"`

ststm32 version requirements removed from platformio.ini
```
#
# BIGTREE_SKR_MINI
#
[env:BIGTREE_SKR_MINI]
platform      = ststm32
framework     = arduino
board         = genericSTM32F103RC
extra_scripts = buildroot/share/PlatformIO/scripts/STM32F1.py
build_flags   = !python Marlin/src/HAL/HAL_STM32F1/STM32F1_flag_script.py
  ${common.build_flags}
  -g
lib_deps      = ${common.lib_deps}
lib_ignore    = U8glib-HAL
  c1921b4
  libf3c
  lib066
  Adafruit NeoPixel_ID28
  Adafruit NeoPixel
  libf3e
  TMC26XStepper
lib_ldf_mode  = 1
src_filter    = ${common.default_src_filter} +<src/HAL/HAL_STM32F1>
monitor_speed = 115200
upload_protocol = stlink
debug_tool = stlink
```

Compilation produced tons and of warnings because to disabled thermal protection and fake termal sensors, but the result was a success
```
Linking .pioenvs\BIGTREE_SKR_MINI\firmware.elf
Building .pioenvs\BIGTREE_SKR_MINI\firmware.bin
Checking size .pioenvs\BIGTREE_SKR_MINI\firmware.elf
Memory Usage -> http://bit.ly/pio-memory-usage
DATA:    [===       ]  27.1% (used 13344 bytes from 49152 bytes)
PROGRAM: [======    ]  57.3% (used 150300 bytes from 262144 bytes)
 [SUCCESS] Took 43.69 seconds
```

I would say it really is time to let go old versions of stmst32 platform and move on.

jmdearras, 2019-05-22:
@jmz52 Could you be more specific on which files?
> Configuration files from [#14053 (comment)](https://github.com/MarlinFirmware/Marlin/issues/14053#issuecomment-493796521)
Thanks,
Jim

jmz52, 2019-05-22:
@jmdearras
I've used configuration files attached to your comment https://github.com/MarlinFirmware/Marlin/issues/14053#issuecomment-493796521
[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/3194546/Configuration.zip)

jmdearras, 2019-05-22:
OK, I will try this.  @BigTreeTech is not much help.

jmdearras, 2019-05-22:
> @jmdearras
> I've used configuration files attached to your comment [#14053 (comment)](https://github.com/MarlinFirmware/Marlin/issues/14053#issuecomment-493796521)
> [Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/3194546/Configuration.zip)

One more clarification.  You used the current Marlin build?

jmz52, 2019-05-22:
No, current build does not have SKR-Mini code yet.
I've used code from open Pull Request https://github.com/MarlinFirmware/Marlin/pull/14041 (BigTreeTech SKR Mini v1.1) i.e. https://github.com/yangwenxiong/Marlin/tree/bugfix-2.0.x

jmdearras, 2019-05-22:
> No, current build does not have SKR-Mini code yet.
> I've used code from open Pull Request #14041 (BigTreeTech SKR Mini v1.1) i.e. https://github.com/yangwenxiong/Marlin/tree/bugfix-2.0.x

the configuration file you linked was a generic STM32?

jmz52, 2019-05-22:
I did not check the board's definition in Configuraiton.h.
It was for generic STM32 board with SKR Mini build environment.

With configuration files from the very first post (#define MOTHERBOARD BOARD_BIGTREE_SKR_MINI_V1_1) compilation was a success without any modifications in Configuration_and.h
```
Linking .pioenvs\BIGTREE_SKR_MINI\firmware.elf
Building .pioenvs\BIGTREE_SKR_MINI\firmware.bin
Checking size .pioenvs\BIGTREE_SKR_MINI\firmware.elf
Memory Usage -> http://bit.ly/pio-memory-usage
DATA:    [==        ]  24.2% (used 11872 bytes from 49152 bytes)
PROGRAM: [=======   ]  69.7% (used 182740 bytes from 262144 bytes)
 [SUCCESS] Took 29.43 seconds
```

jmdearras, 2019-05-22:
@jmz52 close, but it did not work on mine.  
n file included from C:\users\jimde\.platformio\packages\framework-arduinoststm32-maple\STM32F1\cores\maple/wirish.h:60:0,
                 from C:\users\jimde\.platformio\packages\framework-arduinoststm32-maple\STM32F1\cores\maple/Arduino.h:30,

                 from C:\users\jimde\.platformio\lib\U8glib_ID7\src\clib\u8g_com_arduino_fast_parallel.c:69:
C:\users\jimde\.platformio\packages\framework-arduinoststm32-maple\STM32F1\cores\maple/wirish_math.h:59:6: error: conflicting types for 'random'
 long random(long min, long max);

compilation terminated due to -fmax-errors=5.
In file included from C:\users\jimde\.platformio\packages\framework-arduinoststm32-maple\STM32F1\cores\maple/boards.h:38:0,
                 from C:\users\jimde\.platformio\packages\framework-arduinoststm32-maple\STM32F1\cores\maple/wirish.h:54,
                 from C:\users\jimde\.platformio\packages\framework-arduinoststm32-maple\STM32F1\cores\maple/Arduino.h:30,
                 from C:\users\jimde\.platformio\lib\U8glib_ID7\src\clib\u8g_com_arduino_fast_parallel.c:69:
C:\users\jimde\.platformio\packages\framework-arduinoststm32-maple\STM32F1\cores\maple/wirish_types.h:65:9: error: unknown type name 'bool'
 typedef bool boolean;
         ^~~~

and more, all in the maple stuff.

jmz52, 2019-05-22:
C vs. C++ conflicts caused by outdated U8glib.
Remove everything from .piolibs to download new version.
If you are using text screen you can add U8glib-HAL to lib_ignore.

jmdearras, 2019-05-22:
@jmz52 I tried that and got the same errors.  And U8glib-HAL is already in the lib_ignore.  I really appreciate your help.  Any other thoughts?

jmdearras, 2019-05-22:
> @jmz52 I tried that and got the same errors. And U8glib-HAL is already in the lib_ignore. I really appreciate your help. Any other thoughts?

OK, I deleted that library from my .platformio tree under my ID and it worked!  Thanks!

jmz52, 2019-05-22:
Remove locally installed U8glib `C:\users\jimde\.platformio\lib\U8glib_ID7\`
My system has no libraries  in .platformio\lib

jmz52, 2019-05-22:
Could you please check if you can compile without U8glib-HAL in lib_ignore?

jmdearras, 2019-05-22:
Yes, it compiles.  Unfortunately, it does not seem to run in the board.  :(

jmdearras, 2019-05-22:
I reverted back to stock code and the board is OK.

jmdearras, 2019-05-22:
using the BigTreeTech config files, it load and runs.  Somewhat.  USB com works, and LCD displays.  Responds to knob turns (frame rate changes,) but the first knob click crashes it.  But progress!

jmdearras, 2019-05-22:
undefining speaker fixed the knob presses.  The display does not have a speaker, anyway, just a beeper.   But this probably points to a timer issue.

jmdearras, 2019-05-23:
with the https://github.com/yangwenxiong/Marlin/tree/bugfix-2.0.x and the attached config files, it should compile and run,
[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/3209820/Configuration.zip)

AletheianAlex, 2019-06-06:
If it helps at all, I have been getting around the Maple core conflicts for the past few years by putting a band-aid on the lib headers as needed.  Tedious, but I tried a few times a few years back to get the patches into the maple lib and nobody seemed interested over there.  ( It comes up in non-marlin use as well.) Example from wirish.h:
[image]

I'll attach the patched headers that got the lib to play nice and compile with u8g a few years ago.  It has been a while, but if I recall, you have to include a dummy binary.h header as well (if that has not been patched).

[STM32_patched_maplelibb_for_u8g.zip](https://github.com/MarlinFirmware/Marlin/files/3263671/STM32_patched_maplelibb_for_u8g.zip)

jmdearras, 2019-09-26:
> @jmdearras <https://github.com/jmdearras> is the issue still there?

I have not pulled for a while, I will test.

boelle, 2019-10-12:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

closed 2019-10-12 by boelle